scottlamb/moonfire-nvr #8: Seg fault

comdata, 2017-05-08:
Hello,

I am getting a Segfault all the time.

It seems to happen right here:

I0508 232630.200 stream-Auffahrt moonfire_nvr::streamer] Auffahrt: Opening input: rtsp://*:redacted@192.168.1.29:554/unicast
rt_sigaction(SIGTERM, {0x54bde2f4, [], SA_RESTORER|SA_RESTART|SA_SIGINFO|SA_NOCLDSTOP, 0x75d031a0}, {SIG_DFL, [], 0}, 8) = 0
clock_gettime(CLOCK_REALTIME, {1494278790, 201676377}) = 0
clock_gettime(CLOCK_MONOTONIC, {634483, 68003909}) = 0
futex(0x749ff81c, FUTEX_WAKE_OP_PRIVATE, 1, 1, 0x749ff818, {FUTEX_OP_SET, 0, FUTEX_OP_CMP_GT, 1}) = 1
futex(0x749ff7fc, FUTEX_WAKE_PRIVATE, 1) = 1
I0508 232630.201 main moonfire_nvr::cmds::run] Ready to serve HTTP requests
clock_gettime(CLOCK_MONOTONIC, {634483, 68300106}) = 0
epoll_wait(9, {{EPOLLOUT, {u32=4, u64=4}}}, 1024, 0) = 1
clock_gettime(CLOCK_MONOTONIC, {634483, 68471043}) = 0
clock_gettime(CLOCK_MONOTONIC, {634483, 68547137}) = 0
epoll_wait(9, {}, 1024, 0)              = 0
clock_gettime(CLOCK_MONOTONIC, {634483, 68773334}) = 0
clock_gettime(CLOCK_MONOTONIC, {634483, 68842761}) = 0
epoll_wait(9, [rtsp @ 0x73e16000] Estimating duration from bitrate, this may be inaccurate
I0508 232634.707 stream-Auffahrt moonfire_nvr::stream] Discarding the first packet to work around https://trac.ffmpeg.org/ticket/5018
 <unfinished ...>
+++ killed by SIGSEGV +++
Segmentation fault

The camera is running the fang hacks on a Xiaomi Wifi Cam.

Any ideas what could me missing?

ffmpeg
ffmpeg version 2.8.11 Copyright (c) 2000-2017 the FFmpeg developers
  built with gcc 4.9.2 (Raspbian 4.9.2-10)
  configuration: --arch=armel --target-os=linux --enable-gpl --enable-libx264 --enable-nonfree
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100

Kind Regards,
Christoph

scottlamb, 2017-05-10:
Thanks for the report! Couple ideas to figure it out:

   * Your ffmpeg output looks good, but could you run `ldd path/to/moonfire-nvr` and `ldd ffmpeg` to make sure the two are using the same libraries? (My old C++ version used to print the compiled-against and linked-against ffmpeg versions on startup; between this and issue #7, I'm seeing that I really need to reintroduce that feature into the Rust version.)
   * Could you get a stack trace with `gdb`? Just `gdb path/to/moonfire-nvr`, then `run`, then `bt` will probably do.

comdata, 2017-05-16:
I did a rebuild of the package and this seems to have fixed any issue. So it seems the ffmpeg library was not correct during the initial compile.
Thanks

closed 2017-05-16 by comdata