scottlamb/moonfire-nvr #10: test failures with ffmpeg 3.x

scottlamb, 2017-10-01:
Since 857a66f29ce676d8a87e2c894cd16e628fd06d4a, Moonfire NVR compiles with ffmpeg 3.x and doesn't segfault, but tests are failing. It looks like the `.mp4` output has changed in some fashion (I've only looked at the hashed, not full diffs), and the interpretation of edit lists may have changed.

```
failures:

---- mp4::tests::test_round_trip_with_shorten stdout ----
	thread 'mp4::tests::test_round_trip_with_shorten' panicked at 'assertion failed: `(left == right)`
  left: `"e0d28ddf08e24575a82657b1ce0b2da73f32fd88"`,
 right: `"c061c644f9f2188be34248da4eda5752bc916001"`', src/mp4.rs:2067:8
note: Run with `RUST_BACKTRACE=1` for a backtrace.

---- mp4::tests::test_round_trip_with_edit_list stdout ----
	thread 'mp4::tests::test_round_trip_with_edit_list' panicked at 'assertion failed: `(left == right)`
  left: `0`,
 right: `1`', src/mp4.rs:1810:12

---- mp4::tests::test_round_trip stdout ----
	thread 'mp4::tests::test_round_trip' panicked at 'assertion failed: `(left == right)`
  left: `"1e5331e8371bd97ac3158b3a86494abc87cdc70e"`,
 right: `"2059a2d0b797cb1b0551c1c1fddb25c788fe13ab"`', src/mp4.rs:2007:8

---- mp4::tests::test_round_trip_with_subtitles stdout ----
	thread 'mp4::tests::test_round_trip_with_subtitles' panicked at 'assertion failed: `(left == right)`
  left: `"de382684a471f178e4e3a163762711b0653bfd83"`,
 right: `"dec3fffbdb453e420dfeb20d37816e54a3936c8b"`', src/mp4.rs:2027:8

failures:
    mp4::tests::test_round_trip
    mp4::tests::test_round_trip_with_edit_list
    mp4::tests::test_round_trip_with_shorten
    mp4::tests::test_round_trip_with_subtitles

test result: FAILED. 42 passed; 4 failed; 0 ignored; 0 measured; 0 filtered out
```

scottlamb, 2017-10-02:
The ffmpeg versions don't agree on duration of the final packet in `src/testdata/clip.mp4`. How strange. The tests use that file as the basis of the data they put into the database and pull out again, so it makes sense that this would cause these test failures.

```
$ ffprobe -show_packets src/testdata/clip.mp4 | egrep '^duration=' | tail -1
ffprobe version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2007-2017 the FFmpeg developers
...
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
...
duration=90000
```

vs

```
$ ffprobe -show_packets src/testdata/clip.mp4 | egrep '^duration=' | tail -1
ffprobe version 3.3.4 Copyright (c) 2007-2017 the FFmpeg developers
...
  libavutil      55. 58.100 / 55. 58.100
  libavcodec     57. 89.100 / 57. 89.100
  libavformat    57. 71.100 / 57. 71.100
  libavdevice    57.  6.100 / 57.  6.100
  libavfilter     6. 82.100 /  6. 82.100
  libavresample   3.  5.  0 /  3.  5.  0
  libswscale      4.  6.100 /  4.  6.100
  libswresample   2.  7.100 /  2.  7.100
  libpostproc    54.  5.100 / 54.  5.100
...
duration=90085
```

scottlamb, 2017-10-02:
Hmm, my test clip doesn't use a 1/90,000 timescale (as specified in `mvhd` or as specified in `tkhd`). I think the two ffmpeg versions are doing doing the timescale conversion differently, and there may not be a single correct answer. I should fix my input file to be unambiguous.

scottlamb, 2017-10-03:
9eff91f addressed most of the failures, but the `mp4::tests::test_round_trip_with_edit_list` one remains. This is another case of the ffmpeg versions interpreting the same `.mp4` differently. Haven't figured out why yet.

```
$ ffprobe -show_packets with-edit-list.mp4 | egrep '^pts='
ffprobe version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2007-2017 the FFmpeg developers
...
pts=-1
pts=90378
pts=180262
pts=270011
pts=359992
pts=450047
pts=540014
pts=630035
pts=719993
pts=810004
```

vs

```
$ ffprobe -show_packets with-edit-list-ffmpeg3.mp4 | egrep '^pts='
ffprobe version 3.3.4 Copyright (c) 2007-2017 the FFmpeg developers
...
pts=0
pts=0
pts=89884
pts=179633
pts=269614
pts=359669
pts=449636
pts=539657
pts=629615
pts=719626
```

scottlamb, 2017-10-03:
Hmm, ffmpeg3 behaves like ffmpeg2 if I specify `-advanced_editlist 0` on the commandline.

I saw something funny about my elsts—the rate fraction was 1 where it probably should be 0. But fixing that had no effect, so either something else is wrong with my `.mp4` or ffmpeg has a bug.

axwell, 2018-09-28:
hmm, i have the same issue, is this fixed ? or is there a workaround ?

scottlamb, 2019-01-14:
> hmm, i have the same issue, is this fixed ? or is there a workaround ?

My apologies; I missed your comment before. If you're still looking to fix this...

What ffmpeg version are you using? It should be present in the full log output. For example, when I run tests, I see this:

```
I0113 210626.336 mp4::tests::test_round_trip_with_edit_list moonfire_ffmpeg] Initialized ffmpeg. Versions:
avutil: running=56.14.100 compiled=56.14.100
avcodec: running=58.18.100 compiled=58.18.100
avformat: running=58.12.100 compiled=58.12.100
```

and it works fine. Presumably you have a different version, perhaps one that doesn't pay attention to the `advanced_editlist` option.

Similarly, what version of Moonfire NVR are you using? `git rev-parse --short HEAD` should tell you.

polyzen, 2019-07-10:
ffmpeg 4.1.3
https://paste.xinu.at/NFMf/

scottlamb, 2019-07-11:
Thanks! I think they actually fixed a bug here, so I can just relax the test a bit to make it pass. My code:

```rust
        if let Some((orig_dur, new_dur)) = final_durations {
            // One would normally expect the duration to be exactly the same, but when using an
            // edit list, ffmpeg appears to extend the last packet's duration by the amount skipped
            // at the beginning. I think this is a bug on their side.
            assert!(orig_dur - shorten + pts_offset == new_dur,
                    "orig_dur={} new_dur={} shorten={} pts_offset={}",
                    orig_dur, new_dur, shorten, pts_offset);
        }
```

vs the panic message:

```
thread 'mp4::tests::test_round_trip_with_edit_list' panicked at 'orig_dur=90000 new_dur=90000 shorten=0 pts_offset=1', src/mp4.rs:1884:13
```

closed 2019-07-11 by scottlamb

polyzen, 2019-07-11:
Thank you very much.

commit 856c0191 (2019-07-11): Update test to allow ffmpeg 4.x edit list behavior

Fixes #10