scottlamb/moonfire-nvr #11: cargo test fails with 4 errors

dolfs, 2017-10-22:
During building (8de7e391f8cd754cf205e737cf47b19ceb7012b4):

See below output:
```
$ RUST_TEST_THREADS=1 MOONFIRE_LOG=trace cargo test
    Finished dev [unoptimized + debuginfo] target(s) in 0.4 secs
     Running target/debug/deps/moonfire_nvr-ea606875e7c62cdf

running 48 tests
test cmds::config::tests::test_decode ... ok
test coding::tests::test_bad_varints ... ok
test coding::tests::test_correct_varints ... ok
test coding::tests::test_zigzag ... ok
test db::tests::test_adjust_days ... ok
test db::tests::test_day_bounds ... ok
test db::tests::test_drop_tx ... I1022 160352.363 db::tests::test_drop_tx moonfire_nvr::db] Loading video sample entries
I1022 160352.364 db::tests::test_drop_tx moonfire_nvr::db] Loaded 0 video sample entries
I1022 160352.365 db::tests::test_drop_tx moonfire_nvr::db] Loading cameras
I1022 160352.365 db::tests::test_drop_tx moonfire_nvr::db] Loaded 0 cameras
I1022 160352.368 db::tests::test_drop_tx moonfire_nvr::db] reserved d2b629c0-7662-4a8f-a20a-4300d08db8b8
I1022 160352.370 db::tests::test_drop_tx moonfire_nvr::db] reserved d8f0ad1c-1ef5-4a2c-becf-d5dbc71af2c0
ok
test db::tests::test_fresh_db ... I1022 160352.382 db::tests::test_fresh_db moonfire_nvr::db] Loading video sample entries
I1022 160352.383 db::tests::test_fresh_db moonfire_nvr::db] Loaded 0 video sample entries
I1022 160352.383 db::tests::test_fresh_db moonfire_nvr::db] Loading cameras
I1022 160352.384 db::tests::test_fresh_db moonfire_nvr::db] Loaded 0 cameras
ok
test db::tests::test_full_lifecycle ... I1022 160352.398 db::tests::test_full_lifecycle moonfire_nvr::db] Loading video sample entries
I1022 160352.399 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 0 video sample entries
I1022 160352.400 db::tests::test_full_lifecycle moonfire_nvr::db] Loading cameras
I1022 160352.401 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 1 cameras
I1022 160352.401 db::tests::test_full_lifecycle moonfire_nvr::db] Loading recordings for camera testcam
I1022 160352.402 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 0 recordings for camera testcam
I1022 160352.405 db::tests::test_full_lifecycle moonfire_nvr::db] reserved 15410fa9-e2c6-42e1-92ea-a24b8760cd4a
I1022 160352.405 db::tests::test_full_lifecycle moonfire_nvr::db] reserved 602ceb37-7fac-4bdb-9378-b0e21e06318c
I1022 160352.425 db::tests::test_full_lifecycle moonfire_nvr::db] Loading video sample entries
I1022 160352.427 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 1 video sample entries
I1022 160352.427 db::tests::test_full_lifecycle moonfire_nvr::db] Loading cameras
I1022 160352.428 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 1 cameras
I1022 160352.429 db::tests::test_full_lifecycle moonfire_nvr::db] Loading recordings for camera testcam
I1022 160352.430 db::tests::test_full_lifecycle moonfire_nvr::db] Loaded 1 recordings for camera testcam
ok
test db::tests::test_no_version ... ok
test db::tests::test_version_too_new ... ok
test db::tests::test_version_too_old ... ok
test dir::tests::adjust ... ok
test h264::tests::test_decode ... ok
test h264::tests::test_sample_entry_from_annex_b ... ok
test h264::tests::test_sample_entry_from_avc_decoder_config ... ok
test h264::tests::test_transform_sample_data ... ok
test mp4::tests::test_all_sync_frames ... I1022 160352.509 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loading video sample entries
I1022 160352.509 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loaded 0 video sample entries
I1022 160352.510 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loading cameras
I1022 160352.511 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loaded 1 cameras
I1022 160352.511 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loading recordings for camera test camera
I1022 160352.512 mp4::tests::test_all_sync_frames moonfire_nvr::db] Loaded 0 recordings for camera test camera
D1022 160352.513 mp4::tests::test_all_sync_frames moonfire_nvr::dir] test camera: have remaining quota of 1048576
T1022 160352.592 mp4::tests::test_all_sync_frames moonfire_nvr::recording] recording::Segment::new slow path, desired_range_90k=2..20, recording=ListRecordingsRow {
    start: Time(
        128700576000000
    ),
    video_sample_entry: VideoSampleEntry {
        data: [
            0,
            0,
            0,
            0,
            0,
            0,
            0,
[... 2651 lines cut ...]

---- mp4::tests::test_round_trip stdout ----
	thread 'mp4::tests::test_round_trip' panicked at 'unexpected input error: End of file', src/mp4.rs:1778:28
note: Run with `RUST_BACKTRACE=1` for a backtrace.

---- mp4::tests::test_round_trip_with_edit_list stdout ----
	thread 'mp4::tests::test_round_trip_with_edit_list' panicked at 'unexpected input error: End of file', src/mp4.rs:1778:28

---- mp4::tests::test_round_trip_with_shorten stdout ----
	thread 'mp4::tests::test_round_trip_with_shorten' panicked at 'unexpected input error: End of file', src/mp4.rs:1778:28

---- mp4::tests::test_round_trip_with_subtitles stdout ----
	thread 'mp4::tests::test_round_trip_with_subtitles' panicked at 'unexpected input error: End of file', src/mp4.rs:1778:28

failures:
    mp4::tests::test_round_trip
    mp4::tests::test_round_trip_with_edit_list
    mp4::tests::test_round_trip_with_shorten
    mp4::tests::test_round_trip_with_subtitles

test result: FAILED. 44 passed; 4 failed; 0 ignored; 0 measured; 0 filtered out

error: test failed, to rerun pass '--bin moonfire-nvr'
pi@moonfire:~/moonfire-nvr$
```

scottlamb, 2017-10-23:
I think this is a hint:

```
avcodec compiled=56.60.100 running=56.1.0
avformat compiled=56.40.101 running=56.1.0
avutil compiled=54.31.100 running=54.3.0
```

It's running against an older version than it was compiled with. Not too surprising that didn't work, although annoying that we don't see the actual error. I suspect a subprocess running the test crashed with a SIGSEGV it some such, and annoyingly, the test harness didn't say as much.

857a66f changed the way moonfire-nvr uses ffmpeg, and it's apparently not working in your case. What does this command say?

```
pkg-config --libs --cflags libavutil
```

dolfs, 2017-10-23:
This ffmpeg thing continues to sting. A while ago I could not get it running and I had to manually install a necessary version. So that is what I have now:

> $ pkg-config --libs --cflags libavutil
> -I/usr/local/include -L/usr/local/lib  -lavutil  

/usr/local/lib contains the 54.31.100 version (that I manually installed I suppose).
/usr/lib/arm-linux-gnueabihf contains the 54.3.0 version and that was installed when I installed the latest rust with rustup (running 1.21.0)

So it appears that some part of the build saw the first, some saw the latter.

> I think this is a hint:
>
> avcodec compiled=56.60.100 running=56.1.0
> avformat compiled=56.40.101 running=56.1.0
> avutil compiled=54.31.100 running=54.3.0
> It's running against an older version than it [...]

—dolf

scottlamb, 2017-10-23:
It seems that it's not respecting the `-L/usr/local/lib` output by `pkg-config`. I'm not sure why. It works on my Raspberry Pi 2, which is similar to yours—each library has a system-installed copy in `/usr/lib/arm-linux-gnueabihf` and my manually-installed copy in `/usr/local/lib`. I just tried it with Rust 1.21.0 in case that was the difference, but it still works for me.

What does this say?

```
$ cd path/to/moonfire-nvr/ffmpeg
$ cargo build -vv
```

On my pi2, it includes these lines:

```
cargo:rustc-link-search=native=/usr/local/lib
cargo:rustc-link-lib=avutil
cargo:rustc-link-search=native=/usr/local/lib
cargo:rustc-link-lib=avcodec
cargo:rustc-link-search=native=/usr/local/lib
cargo:rustc-link-lib=avformat
```

which correspond to the pkg-config output and should ensure that the binary is linked with the correct copy of the libraries. These affect the rustc command for the `moonfire_ffmpeg` crate, which is:

```
     Running `rustc --crate-name moonfire_ffmpeg lib.rs --crate-type lib --emit=dep-info,link -C deb
uginfo=2 -C metadata=4e94040a92250526 -C extra-filename=-4e94040a92250526 --out-dir /home/slamb/git/
moonfire-nvr/ffmpeg/target/debug/deps -L dependency=/home/slamb/git/moonfire-nvr/ffmpeg/target/debug
/deps --extern libc=/home/slamb/git/moonfire-nvr/ffmpeg/target/debug/deps/liblibc-a1a8fca944970461.r
lib --extern log=/home/slamb/git/moonfire-nvr/ffmpeg/target/debug/deps/liblog-0b500cc3f48efa00.rlib
-L native=/usr/local/lib -L native=/usr/local/lib -L native=/usr/local/lib -L native=/home/slamb/git
/moonfire-nvr/ffmpeg/target/debug/build/moonfire-ffmpeg-56998b07166d1b4b/out -l avutil -l avcodec -l
 avformat -l static=wrapper`
```

dolfs, 2017-10-23:
Oops, I cannot answer as I actually removed my personally installed copy to see if that would fix it. It does in some sense as I now see:

> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: avcodec compiled=56.1.0 running=56.1.0
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: avformat compiled=56.1.0 running=56.1.0
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: avutil compiled=54.3.0 running=54.3.0

So versions are now consistent, but…
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: main moonfire_nvr::cmds::run] Ready to serve HTTP requests
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Front Right moonfire_nvr::streamer] Front Right: Opening input: rtsp://admin:redacted@192.168.1.82/Streaming/Channels/1
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Side moonfire_nvr::streamer] Side: Opening input: rtsp://admin:redacted@192.168.1.83/Streaming/Channels/1
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Back moonfire_nvr::stream] While opening URL rtsp://admin:dolf----@192.168.1.84/Streaming/Channels/1, some options were not understood: user-agent=moonfire-nvr, stimeout=10000000
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Front Left moonfire_nvr::stream] While opening URL rtsp://admin:dolf-----@192.168.1.81/Streaming/Channels/1, some options were not understood: user-agent=moonfire-nvr, stimeout=10000000
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Side moonfire_nvr::stream] While opening URL rtsp://admin:dolf----@192.168.1.83/Streaming/Channels/1, some options were not understood: user-agent=moonfire-nvr, stimeout=10000000
> Oct 22 22:41:21 moonfire moonfire-nvr[2698]: stream-Front Right moonfire_nvr::stream] While opening URL rtsp://admin:dolf----@192.168.1.82/Streaming/Channels/1, some options were not understood: user-agent=moonfire-nvr, stimeout=10000000
> Oct 22 22:41:22 moonfire moonfire-nvr[2698]: [rtsp @ 0x72c19000] Estimating duration from bitrate, this may be inaccurate
> Oct 22 22:41:22 moonfire moonfire-nvr[2698]: stream-Side moonfire_nvr::streamer] Side: sleeping for Duration { secs: 1, nanos: 0 } after error: Error: ffmpeg: Unknown error -3
> Oct 22 22:41:22 moonfire moonfire-nvr[2698]: cause: Some(Error(3))
So, still the unrecognized options, even though moonfire does think we have the compatible (equal) versions of ffmpeg and it does not protest about the version being too old.
So, with the understanding I removed my custom ffmpeg, I now get:
> $ pkg-config --libs --cflags libavutil
> -lavutil  

> cargo:rustc-link-search=native=/usr/lib/arm-linux-gnueabihf
> cargo:rustc-link-lib=avutil
> cargo:rustc-link-search=native=/usr/lib/arm-linux-gnueabihf
> cargo:rustc-link-lib=avcodec
> cargo:rustc-link-search=native=/usr/lib/arm-linux-gnueabihf
> cargo:rustc-link-lib=avformat
> lib: avutil
> lib: avcodec
> lib: avformat

     Running `rustc --crate-name moonfire_ffmpeg lib.rs --crate-type lib --emit=dep-info,link -C debuginfo=2 -C metadata=4e94040a92250526 -C extra-filename=-4e94040a92250526 --out-dir /home/pi/moonfire-nvr/ffmpeg/target/debug/deps -L dependency=/home/pi/moonfire-nvr/ffmpeg/target/debug/deps --extern log=/home/pi/moonfire-nvr/ffmpeg/target/debug/deps/liblog-0b500cc3f48efa00.rlib --extern libc=/home/pi/moonfire-nvr/ffmpeg/target/debug/deps/liblibc-a1a8fca944970461.rlib -L native=/usr/lib/arm-linux-gnueabihf -L native=/usr/lib/arm-linux-gnueabihf -L native=/usr/lib/arm-linux-gnueabihf -L native=/home/pi/moonfire-nvr/ffmpeg/target/debug/build/moonfire-ffmpeg-56998b07166d1b4b/out -l avutil -l avcodec -l avformat -l static=wrapper`    Finished dev [unoptimized + debuginfo] target(s) in 53.47 
secs

So it all seems internally consistent, but it seems these versions of ffmpeg do not understand the options you are using. Next I am going to try and install a newer version of ffmpeg.

> It seems that it's not respecting the -L/usr/local/lib output by pkg-config. I'm not sure why. It works on my Raspberry Pi 2, which is similar to yours—each library has a system-installed copy in [...]

—dolf

scottlamb, 2017-10-23:
Yeah, I'd put your custom one back. 857a66f  made moonfire-nvr a lot more forgiving about libav* versions, but it's definitely still desirable to use a newer version of the libraries than what comes with Raspbian Jessie. Not having `stimeout` means not being able to recover from a camera restart without restarting moonfire-nvr as well.

dolfs, 2017-10-23:
Agreed, but I know I had 2.8.10 which does support it. But, when I had it, the wrapper would select the other installation, which does not. Now it still uses that version, and there is no conflict, but it does not support it. Since I think you said it should now also work with 3.x due to the wrapper, I’ll try that next.

> Yeah, I'd put your custom one back. 857a66f <https://github.com/scottlamb/moonfire-nvr/commit/857a66f29ce676d8a87e2c894cd16e628fd06d4a> made moonfire-nvr a lot more forgiving about libav* versions, [...]

—dolf

scottlamb, 2017-10-23:
Yes, 3.x should work. Of course, 2.8.10 should have worked too so we'll see. If you can get the bad behavior of linking against the wrong library version again, I'd very much like to figure it out.

The only thing I see that's sketchy about what the Rust pkg-config crate, cargo, and rustc are (collectively) doing is that they aren't respecting the original interleaving of the `-L` and `-l` arguments. They're effectively adding in all the `-L`s then the `-l`s. If there were a `-L/usr/lib/arm-linux-gnueabihf` afterward on the commandline, I think that'd cause this behavior of the libraries in that directory winning over the ones in `/usr/local/lib`. I don't know why that would have happened in practice though.

scottlamb, 2019-07-10:
Modern distributions come with acceptable ffmpegs, so I'm going to close this.

closed 2019-07-10 by scottlamb