scottlamb/moonfire-nvr #12: /etc/localtime not handled correctly

dolfs, 2017-10-23:
I tried to startup and received the following in the log file:
```
Oct 22 16:49:50 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading video sample entries
Oct 22 16:49:50 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 4 video sample entries
Oct 22 16:49:50 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading cameras
Oct 22 16:49:50 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 4 cameras
Oct 22 16:49:50 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading recordings for camera Front Left
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 12781 recordings for camera Front Left
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading recordings for camera Front Right
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 11659 recordings for camera Front Right
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading recordings for camera Side
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 11527 recordings for camera Side
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loading recordings for camera Back
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::db] Loaded 11908 recordings for camera Back
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: main moonfire_nvr::cmds::run] Database is loaded.
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: thread 'main' panicked at 'unable to read localtime symlink: Error { repr: Os { code: 22, message: "Invalid argument" } }', /checkout/src/libcore/result.rs:906:4
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: stack backtrace:
Oct 22 16:49:51 moonfire moonfire-nvr[27295]: 0: std::sys::imp::backtrace::tracing::imp::unwind_backtrace
Oct 22 16:49:52 moonfire moonfire-nvr[27295]: at ./checkout/src/libstd/sys/unix/backtrace/tracing/gcc_s.rs:49
Oct 22 16:49:52 moonfire moonfire-nvr[27295]: 1: std::sys_common::backtrace::_print
Oct 22 16:49:52 moonfire moonfire-nvr[27295]: at ./checkout/src/libstd/sys_common/backtrace.rs:71
```

It turns out that on my Raspbian Jesse system, `/etc/localtime` is a copy of the required file from `/usr/share/zoneinfo` and not a symlink to it. Seems to me that the code should be able to handle either scheme.
When I changed the situation, by hand, to be a symlink, startup became possible.

scottlamb, 2017-10-23:
Hmm. Not sure of the best way to fix this. Here's what I'm trying to do at a high level:

The server stores timestamps in 90,000ths of a second (excluding leap seconds) since 1970-01-01T00:00:00Z. Basically, Unix time * 90,000.

It summarizes these into "days" to speed up the UI. One of the JSON responses includes for each camera a summary of each calendar data, so the calendar on the widget can load really fast, and then the UI can get more detailed information for each day, more quickly than retrieving 90+ days of data.

Those days are computed using the server's local time zone. The rest of the UI should match that time zone. Otherwise, it'd be pretty confusing. For example, you might click on a day that's supposed to have data and see nothing.

So my current approach is for the server to tell the UI what time zone to use, and the UI to use [moment-timezone](https://www.npmjs.com/package/moment-timezone) to match that. (I'd rather give it the actual IANA zone data rather than have the UI have its own lookup table that's large and potentially out of sync, but moment-timezone doesn't seem to parse this format.)

So the server needs to get the timezone name. There doesn't seem to be an actual POSIX API for this. There's the `tzname` global variable, but it doesn't give the IANA time zone name, just `PST` and `PDT` which are pretty much useless.

My hacky approach was to see where the `/etc/localtime` symlink points. If `/etc/localtime` isn't a symlink, then I don't know what to do. It turns out the string `America/Los_Angeles` isn't even in the file at all, so even if I wanted to parse it, I still wouldn't get the zone name.

I could just require the user specify the zone name on the commandline, and set the `TZ` environmental variable to that prior to calling `tzset`, but then it's another thing you have to configure.

Hmm, `/etc/timezone` is a file that says `America/Los_Angeles` on the Linux systems I just tried. Not on OS X. Grr. I guess I can support both ways.

dolfs, 2017-10-23:
Yes, this is one of those GNU-ism (Gnu Is Not Unix), that also seems to extend to OSX and Linux. They are very similar, but not identical. Perhaps you can check of /etc/localtime is actually a symlink before trying to read it. If it is not a symlink and /etc/timezone exists, go with that. Perhaps a TZ env variable should override it all, if present.

> Hmm. Not sure of the best way to fix this. Here's what I'm trying to do at a high level:
>
> The server stores timestamps in 90,000ths of a second (excluding leap seconds) since 1970-01-01T00:00:00Z. [...]

—dolf