scottlamb/moonfire-nvr #16: moonifire-nvr config cannot operate when misconfigured

dolfs, 2018-03-02:
Just tried to use the config command when (right after building) the samples directory was missing:
```
/usr/local/bin/moonfire-nvr config
I0302 123829.587 main moonfire_nvr::db] Loading video sample entries
I0302 123829.587 main moonfire_nvr::db] Loaded 0 video sample entries
I0302 123829.587 main moonfire_nvr::db] Loading cameras
I0302 123829.587 main moonfire_nvr::db] Loaded 0 cameras
E0302 123829.588 main moonfire_nvr] Error: unable to open sample file dir /var/lib/moonfire-nvr/sample: No such file or directory (os error 2)
cause: None
```
This seems to in part defeat the purpose of the config command. I would expect, as long as moonfire has been correctly built, this command to always allow changing of settings.

scottlamb, 2018-03-02:
config is for editing configuration stored in the database.

In trunk, the sample file dir is a commandline argument, rather than configuration stored on the database. Use `moonfire-nvr config --sample-file-dir` if you don't want the default.

In the `new-schema` branch, it actually is part of the database, and there in fact are messes it can't clean up (yet).

dolfs, 2018-03-02:
OK, understood, but I still don’t see why using the config command should be impossible if the samples dir is not configured correctly. Since the config command without extra options only operates on the database, and if the database exists and appears in order, why not?
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

scottlamb, 2018-03-02:
It uses it in a couple ways:

*   checks the amount of free space in the filesystem corresponding to the sample file dir, so that you can see how close you are to it when configuring retention.
*   deletes recordings when you lower a camera's retention below what's actually stored (including to 0 when you delete a camera). If you delete a camera, nothing would ever clean up those files otherwise.

dolfs, 2018-03-05:
OK, I see, although I would not bail until you reach a point of using config that requires the sample directory to be in place. I guess at the moment it is coded to do normal startup first, and then handle UI.

scottlamb, 2018-03-05:
You'd reach that point pretty quickly—you could add a camera but not configure its retention, so it's pretty useless.

I'm going to close for now, although there may be similar issues with the new-schema branch's operation.

closed 2018-03-05 by scottlamb