scottlamb/moonfire-nvr #20: Calling API with invalid arg causes connection reset

dolfs, 2018-03-05:
I happened to be doing UI development using a proxy to a normally running moonfire instance. Due to coding error the API for recordings was called with start and end time set to null. Rather then producing an appropriate HTTP response (may be 500, or even 404) would have been expected.

Needless to say that the connection reset causes error messages that are unwanted, but it also means API calling code does not get its error path executed.

Sample output:
```
[HPM] Error occurred while trying to proxy request /api/cameras/1c844181-b807-4b80-83eb-579c8e194451/recordings?startTime90k=null&endTime90k=null&split90k=324000000&_=1520229643233 from localhost:3000 to http://192.168.10.232:8080 (ECONNRESET) (https://nodejs.org/api/errors.html#errors_common_system_errors)
```

dolfs, 2018-03-05:
This was against running code from commit 8de7e391f8cd754cf205e737cf47b19ceb7012b4 so it is possible this has already been fixed. No time at the moment to upgrade running version.

scottlamb, 2018-03-05:
It hasn't been fixed. At present, almost any problem will cause it to just drop the connection rather than return a proper HTTP error. It should at least log something to console that will likely tell you what went wrong.

commit 422cd2a7 (2018-11-27): preliminary web support for auth (#26)

Some caveats:

  * it doesn't record the peer IP yet, which makes it harder to verify
    sessions are valid. This is a little annoying to do in hyper now
    (see hyperium/hyper#1410). The direct peer might not be what we want
    right now anyway because there's no TLS support yet (see #27).  In
    the meantime, the sane way to expose Moonfire NVR to the Internet is
    via a proxy server, and recording the proxy's IP is not useful.
    Maybe better to interpret a RFC 7239 Forwarded header (and/or
    the older X-Forwarded-{For,Proto} headers).

  * i [... cut]
closed 2018-12-27 by scottlamb