scottlamb/moonfire-nvr #26: user authentication

scottlamb, 2018-03-09:
I want proper user authentication; this (along with some manner of https) is a precondition of exposing Moonfire NVR to the Internet.

As @dolfs said, a temporary measure could be to do basic HTTP with a apache/nginx frontend. But I think it shouldn't be too bad to do a proper built-in thing.

Here's my proposal.

https is a prerequisite. (either built-in or through a proxy.)

You can create/delete/modify users through `moonfire-nvr config`.

Creating a session could be simply password-based. Some things that I consider out of scope:

   * two-factor authentication (maybe later)
   * email-based password recovery (maybe later)
   * web-based password changes (later)
   * OAuth2 (probably never; I want it to work without an Internet connection)
   * protection against online brute force attacks: rate limiting and/or auto password disable after too many failed guesses (maybe later; you've gotta pick a good password for now)
   * even logging of auth failures (likewise; pick a really good password for now)
   * roles / permissions (for now, the web interface is read-only, and everyone can read everything)
   * user/device settings, such as UI preferences stored on the server side (maybe later)

Sessions could simply be a long random number, stored as a hash for the same reason people hash passwords: so a database leak doesn't trivially compromise the credentials. (They should need to _edit_ the database for to gain access, which may be harder to do than getting an old backup, and which leaves more evidence behind.)

Straw man schema:

```sql

create table users (
  id integer primary key,
  username unique not null,

  -- If set, a hash for password authentication, as generated by `libpasta::hash_password`.
  password_hash text,
  password_id integer not null default 0,  -- increasing with password sets/clears.
  password_failure_count integer not null,  -- updated lazily; reset when password_id incremented.

  -- If set, a Unix UID that is accepted for authentication when using HTTP over
  -- a Unix domain socket. (Additionally, the UID running Moonfire NVR can authenticate
  -- as anyone; there's no point in trying to do otherwise.) This might be an easy
  -- bootstrap method once configuration happens through a web UI rather than text UI.
  unix_uid integer
);

create table user_sessions (
  user_id integer references user (id),
  hashed_session_id blob unique not null,  -- hash function TBD
  creation_password_id integer not null, -- the id it was created from
  creation_peer_addr blob not null,  -- IPv4 or IPv6 address
  creation_time integer not null,  -- sec since epoch
  creation_user_agent text,
  revocation_time integer, -- likewise
  description text, -- probably a device name
  last_use_time integer,  -- updated lazily on flush
  use_count not null -- likewise
);
```

Typical sessions should probably be represented as cookies with the `HttpOnly` flag set (meaning not accessible to Javascript). If a hit on `/api/...` lacks the cookie, it should return a HTTP 401 which the Javascript knows means to redirect the browser to `/login`, a simple HTML login form which on success sets the cookie and redirects back.

The cookies should also be `secure` (meaning only sent over https, not http).

There should also be a way of getting a session id for use by some automated thing. Example: a program that dumps events into the database based on my Elk security system's zone status. Something you can paste into its config file. Better to have separate, high-entropy credentials where possible so you can see what was compromised and disable it independently of the others.

dolfs, 2018-03-13:
There should also be a solution for mobile devices using the API. The standard web based approach for that is cumbersome. Consider supporting oauth based approach. Tokens handed out should perhaps be revokable through the web interface, but should otherwise be fairly long term expiration, or we need to also implement the token refresh portion.

scottlamb, 2018-12-27:
I still don't have all the checkboxes ticked, but it works okay for me, and it shouldn't urgently require schema changes, so I merged it to master.