scottlamb/moonfire-nvr #29: on-camera motion detection with ONVIF

scottlamb, 2018-03-09:
Realistically speaking any decent NVR must have motion detection. On-camera motion detection is one of the two paths forward. Marking `milestone-1.0?` accordingly.

Needs the signals schema (#28). Additionally, needs to speak SOAP in some way:

   * pure Rust SOAP library? raventid/soap-rs is the only one I've found; it's not usable yet.
   * C/C++ SOAP library? xris-hu/gsoap-onvif or xsmart/onvifcpplib?
   * just do the little bit of SOAP we need by hand? I noticed there's some node onvif library that does this, so maybe it's not so bad.

I'd also considered using Hikvision's simple non-SOAP/ONVIF proprietary protocol, but then Dahua came out with the Starlight camera series, and I bought one...I've even got some code to watch Hikvision camera motion events but it's much less appealing to me now.

scottlamb, 2019-02-20:
I've been experimenting with this.

Good news: it's actually not so bad to hand-roll the SOAP/ONVIF bits I need in pure Rust.

Bad news: common vendors don't speak ONVIF very well. Dahua is particularly bad. See [this thread](https://ipcamtalk.com/threads/api-for-event-stream-such-as-motion-detection.36494/) for a summary.

scottlamb, 2019-06-24:
My playground for experimentation with this is here: https://github.com/scottlamb/moonfire-playground

I have toy code for Dahua and Hikvision's proprietary protocols. ONVIF PullMessages turned out to be kind of disappointing for my Dahua case as described in the thread linked in the previous comment. The Dahua supports the RTSP `vnd.onvif.metadata` stream with some setup (and the hand-rolled SOAP stuff for configuring it is bearable). The Hikvision doesn't. So there's no one approach that will work for all cameras, but maybe a handful of different ones should cover it.

srd424, 2020-07-01:
I'm really interested in something like this for Reolink cameras as well... https://github.com/fwestenberg/reolink/issues/7#issuecomment-546816166 suggests it's possible, but I've not had much luck as yet trying to talk ONVIF to the camera in python ..

scottlamb, 2020-07-02:
That screenshot looks promising. There are three ways to get ONVIF notifications:

*   The RTSP `vnd.onvif.metadata` stream, which you set up with some ONVIF (SOAP over HTTP) calls then get via RTSP.
*   ONVIF PullPoint notifications, which are buggy on Dahua cameras.
*   ONVIF Basic notifications, which are a pain firewall-wise, as described [here](http://community.geniusvision.net/platform/cprndr/ta/onvifimpfaq/entry-h.tt7i90tcrqic.html).

I can't tell from skimming that thread which they're using. But I'm trying to get the RTSP method to work.

There's a Rust onvif library [here](https://github.com/lumeohq/onvif-rs). You could try `cargo run --example camera enable-analytics --uri http://your/camera --username user --password pass` which may or may not work. That library's a work in progress. The code [here](https://github.com/lumeohq/onvif-rs/blob/4a654a42ce8a69bb76519486523fd51c8a0ff2d7/onvif-rs/examples/camera.rs#L275-L348) is supposed to:

1.  find the metadata service URL via `GetServices` on the device management service (which has a well-known URI)
2.  find a metadata configuration via `GetMetadataConfigurations` on the media service
3.  if that configuration doesn't have analytics and events turned on, toggle them via `SetMetadataConfigurations`
4.  list all the profiles via `GetProfiles`
5.  if any profiles don't have a metadata configuration, add this one via `AddMetadataConfiguration`

After that, the RTSP stream URIs returned by `cargo run --example camera get-stream-uris ...` should have include ONVIF metadata, so you can use an RTSP library to watch it. I hope to make a polished version of that here...

xromansx, 2025-03-12:
Can it somehow be combined with https://github.com/rostislavzz/motion-onvif-events