scottlamb/moonfire-nvr #32: fancy scrub bar UI

scottlamb, 2018-03-09:
I've written the server-side code to support segmented `.mp4` files for HTML5 Media Source Extensions. (design/api.md talks about this.) It may have some bugs but is essentially complete.

The goal was that to use this to create the classic NVR UI where there's a scrub bar at the bottom showing an entire day, it grows over time as new stuff is recorded, and none of this has to wait for huge `moov` segments to transfer.

I think this needs some significant work to be viable, though:

   * debug the current segmented `.mp4` files. I noticed some of them were working, some weren't. I'm not sure what was different. I did notice that there are some silly H.264 NALs in there (repeated sps/pps even though they're not needed in the bytestream); that might be related.
   * a bunch of Javascript UI work. probably an automated test setup for UI changes with headless Chrome/Firefox or something.
   * an "event stream" so the UI could know when recordings appear and disappear
   * some way of sending HTTP requests for video data, streaming the response back, and being able to cancel it when you seek to another part of the video.
        * XMLHTTPRequest: but apparently only Firefox supports streaming arraybuffers with this. The other browsers don't return any response data until the response is complete, so there'd have to essentially be an HTTP request per frame, which is annoying.
        * Fetch + Streams APIs: but this is Chrome-only last I checked and doesn't support cancelling requests, so if you seek around a lot your connection will get quite congested while the system catches up with you.
        * WebSockets: this supports both streaming and cancellation, so I think it's the best way to go. But it requires some additional server-side work. hyper annoyingly doesn't support sharing the HTTP port with a WebSockets implementation. Could use a separate port for now.

IronOxidizer, 2022-09-01:
Assuming we only start loading in the mp4 segments on scrubbar release, I think progressive scrubbing (display the scrub thumbnail in the video window) would also be a great feature.

Suxsem, 2024-11-10:
hi, any progress on this one?