scottlamb/moonfire-nvr #33: H.265 support

scottlamb, 2018-03-09:
I think this is fairly simple to add (very similar to `src/h264.rs`) but there are a couple caveats:

   * I haven't found a freely available spec. Old versions of most of the ISO/IEC 14496 standards are freely available but I haven't seen this. We could probably puzzle together what to do based on ffmpeg source code or something but that's not ideal.
   * cameras might always do B-frames along with H.265, which would require reworking the `video_index` format.

It'd be nice to have though. My Dahua camera supports H.265, and I understand it's a significant improvement in video quality per byte.

scottlamb, 2019-07-04:
Ahh, there's a draft of ISO/IEC 14496-15 [here](https://mpeg.chiariglione.org/tags/isoiec-14496-15).

jlpoolen, 2020-05-23:
Some points regarding High Efficiency Video Coding (HEVC), or spcifically H.264 vs. H.265:

- patented technology: moonfire would be decoding, don't know if license is needed to decode H.265 content.
- h.265 is subsequent to h.264 and HEVC promises a 50% storage reduction as its algorithm uses efficient coding by encoding video at the lowest possible bit rate while maintaining a high image quality level.
- HEVC/H.265 comes with the trade-off requiring almost **10x more computing power.** [**Boldface** added.]
    Source: https://medium.com/advanced-computer-vision/h-264-vs-h-265-a-technical-comparison-when-will-h-265-dominate-the-market-26659303171a

scottlamb, 2020-05-24:
Yeah, those might be good considerations to add to a future "choosing your codec" section. I definitely don't want to drop H.264 support so folks would be able to choose (stream-by-stream) which they use. They could for example do video analytics on H.264 sub streams and have H.265 main streams that Moonfire NVR never decodes, only passes along to the client.

IIRC the Raspberry Pi 4 has hardware decoding support and a matching license for both H.264 and H.265. So I think neither the patent nor the decoding speed are necessarily a problem in some common scenarios. I'm not a lawyer and haven't sought legal advice on this, though, so take that with a grain of salt.

Also, on-NVR video analytics will always be optional. They're not even fully implemented yet (#30) (though I really want to fix that). And we can support on-camera video analytics (#29) as a CPU-friendlier alternative.

scottlamb, 2021-03-10:
I'm demoting this from the `1.0?` milestone to `wishlist`. Neither Chrome nor Firefox supports H.265 (https://caniuse.com/hevc). That means Moonfire NVR couldn't just store H.265 from the cameras and stream it unmodified to those browsers as it does for H.264. Users would have to stick with Safari or use a (presumably quite slow) pure software Javascript/WebAssembly H.265 decoder.

xx4212, 2021-07-14:
1. Mozilla (and possibly google) will not support HEVC while it is encumbered by patents ([link](https://developer.mozilla.org/en-US/docs/Web/Media/Formats/Video_codecs#hevc_h.265))

2. Chips&Media Launches AV1 Video Encoder Hardware ([link](https://www.design-reuse.com/news/50175/chips-media-av1-video-encoder-hardware-ip-4k-uhd.html))

So it is possible that in a couple of years cheaper (since no royalties) cameras with AV1 support will appear :)

MikePfaff, 2022-05-23:
[According to Reolink](https://community.reolink.com/topic/2713/rlc-842a-h-264/7), their 8MP camera RLC-842A will automatically switch from H.265 to H.264, if the stream resolution is lower than 8MP:
> Yes, for the new version 8MP camera, if the resolution is higher than 8MP, the video format will switch to H.265. If it is less than 8MP, it will be H.264.

Maybe this "lower stream resolution to get H.264" trick also works on other Reolink 8MP camera models?

jlpoolen, 2023-04-17:
The Reolink Duo is really two cameras in one unit where the feeds are from two sensors positioned such that they cover 180 degrees with no gaps.  Their streams are stitched together into one stream which is 8MP and encoded using H.265.  The really neat thing about this camera is that you just place it some place where you want 180 degree coverage, like near a straight sidewalk, and its done, you do not have to install two cameras (and most cameras do you support 90 degree viewing angles).

So the nature of the 8MP beast has changed with this shift of stitching together two 4 MP streams to create a panorama view.  I have an extensive video surveillance system and one problem I have faced is what happens a bit further down the street when a person of interest goes beyond my camera's field of view.  The 180 degree coverage is an important advancement.

jlpoolen, 2023-04-17:
Here's an example of the day and night versions of the sub stream:
[image]

[image]

scottlamb, 2023-04-17:
Yeah, that stitching is nice for the situation. In terms of Moonfire support, there is another user interested in implementing H.265 in Retina, and from there it'd be pretty easy to support Moonfire *recording* H.265. But actually playing it back in a web UI is another matter. Browser support is still bad: https://caniuse.com/hevc

In theory we could also add support for realtime transcoding in Moonfire, e.g. decoding the H.265 and encoding as H.264, given adequate hardware. The Rockchip RK3588 should be able to do this for example, or some desktop-class GPUs, or add-in accelerator cards such as NETINTs. (Coincidentally, I was talking with them in person today. Their older "Logan" generation T432 cards are down to $300 iirc.)

jlpoolen, 2023-04-17:
Several weeks ago I spent some time with Perl using the API and if I recall, I was able to easily pull segments from Moonfire.  I'm probably the out-in-left-field use case, but having the H.265 files may be useful even though they cannot be played back on most web browsers at this time.  I still have to compare the quality of the "main" feed at H.265 vs. "sub" feed at H.264 to see if there is an appreciable difference.  With the 180 degree camera and the ability to see way beyond, having better resolution could be a real plus.  I'll update here with my findings and examples supporting.

jlpoolen, 2023-04-20:
Here's a study documenting the difference between the 4MP feed encoded with H.264 and the 8MP feed encoded with H.265:  https://salemdata.net/jlp/reolink/h265/reolink_duo_h265_h264_comparison.html

I spoke with a police officer on an unrelated matter and he mentioned that many people share videos with them, but the lack of resolution practically makes the videos useless.  So from law enforcement's perspective, the higher the resolution and ability to discern numbers and features, the better.  It would be nice to be able to use Moonfire-nvr to preserve the H.265 feeds even though they cannot be displayed in the project's web interface.  I'd use the existing H.264 content to review knowing that if I have a really important segment, I can retrieve a noticeably better quality stream to hand to law enforcement.  But I live in a pretty high crime area... that's the price I pay for cheaper real estate.

scottlamb, 2024-08-15:
The `h265` branch has a version of this. Feedback welcome, if anyone cares to give it a spin. There's no schema change necessary; the db format was already codec-agnostic.

Suxsem, 2024-11-16:
hi @scottlamb , any news on this? can you please sync the h265 branch with master and publish a docker package so I can test it? thank you!

nalonsor1, 2024-12-07:
Hello @scottlamb . I built from the h265 branch but  it doesn't seems to  save h265  streams:
 moonfire_nvr::streamer: opening input url=rtsp://192.168.151.187/tenda
moonfire_base::clock: opening rtsp://192.168.151.187/tenda took PT30.000814838S!
moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN

Thank you!!!

scottlamb, 2024-12-07:
@nalonsor1 That's puzzling. Do you see a line immediately after starting with `caused by:`, as in the following example? In a quick skim, I don't see any situation in which it'd log that line and not something more helpful right afterward.

```
2024-12-07T13:07:33.514630  WARN s-courtyard-sub streamer{stream="courtyard-sub"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN
caused by: Unable to connect to RTSP server: Network is unreachable (os error 51)
```

nalonsor1, 2024-12-09:
Hi:
I dont see it. 
Here is a more extended log from nvr run

2024-12-09T20:52:54.829156  INFO main moonfire_nvr::cmds: Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.40.1
2024-12-09T20:52:54.839281  INFO main moonfire_db::db: Loading video sample entries
2024-12-09T20:52:54.839521  INFO main moonfire_db::db: Loaded 1 video sample entries
2024-12-09T20:52:54.839684  INFO main moonfire_db::db: Loading sample file dirs
2024-12-09T20:52:54.839964  INFO main moonfire_db::db: Loaded 1 sample file dirs
2024-12-09T20:52:54.840100  INFO main moonfire_db::db: Loading cameras
2024-12-09T20:52:54.840312  INFO main moonfire_db::db: Loaded 2 cameras
2024-12-09T20:52:54.840453  INFO main moonfire_db::db: Loading streams
2024-12-09T20:52:54.840749  INFO main moonfire_db::db: Loaded 2 streams
2024-12-09T20:52:54.840908  INFO main moonfire_db::db: Loading recordings for camera CP3PRO stream Main
2024-12-09T20:52:54.841100  INFO main moonfire_db::db: Loaded 0 recordings for camera CP3PRO stream Main
2024-12-09T20:52:54.841233  INFO main moonfire_db::db: Loading recordings for camera test stream Main
2024-12-09T20:52:54.841459  INFO main moonfire_db::db: Loaded 1 recordings for camera test stream Main
2024-12-09T20:52:54.841606  INFO main moonfire_nvr::cmds::run: Database is loaded.
2024-12-09T20:52:54.842612  INFO main moonfire_nvr::cmds::run: Directories are opened.
2024-12-09T20:52:54.842766  INFO main moonfire_nvr::cmds::run: Resolved timezone: Europe/Madrid
2024-12-09T20:52:54.842913  INFO main moonfire_nvr::cmds::run: Starting syncer for path /media/nvr/sample
2024-12-09T20:52:54.843620  INFO main syncer{path=/media/nvr/sample}: moonfire_db::writer: initial rotation
2024-12-09T20:52:54.845240  INFO main syncer{path=/media/nvr/sample}:flush{flush_count=0 reason="synchronous deletion"}: moonfire_db::db: flush complete:
/media/nvr/sample: added 0B in 0 recordings (), deleted 607K 658B in 1 (2/107), GCed 0 recordings ().
2024-12-09T20:52:54.846864  INFO main syncer{path=/media/nvr/sample}:flush{flush_count=1 reason="synchronous garbage collection"}: moonfire_db::db: flush complete:
/media/nvr/sample: added 0B in 0 recordings (), deleted 0B in 0 (), GCed 3 recordings (2/106, 2/105, 2/107).
2024-12-09T20:52:54.847385  INFO main moonfire_nvr::cmds::run: no LISTEN_FDs
2024-12-09T20:52:54.851340  INFO s-test-main streamer{stream="test-main"}: moonfire_nvr::cmds::run: starting
2024-12-09T20:52:54.851556  INFO s-test-main streamer{stream="test-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.40.26:8554/profile1
2024-12-09T20:52:54.852087  INFO s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::cmds::run: starting
2024-12-09T20:52:54.852270  INFO s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.151.187/tenda
2024-12-09T20:52:54.852585  INFO        sync-1 syncer{path=/media/nvr/sample}: moonfire_db::writer: starting
2024-12-09T20:52:54.859634  INFO          main moonfire_nvr::cmds::run: Ready to serve HTTP requests
2024-12-09T20:52:54.969624  WARN tokio-runtime-worker streamer{stream="test-main"}: retina::codec::h264: Ignoring bad H.264 format-specific-params "profile-level-id=4DE028;packetization-mode=1;sprop-parameter-sets=AAAAAWdNABaNjUBQF/yzcBAQFAAALuAABX5AEA==,AAAAAWjuOIA=": only SPS and PPS expected in parameter sets
2024-12-09T20:52:55.089364  WARN tokio-runtime-worker streamer{stream="test-main"}: retina::client: Ignoring data on unassigned RTSP interleaved data channel 2. This is the first such message. Following messages will be logged at trace priority only.

conn: 192.168.40.124:46506(me)->192.168.40.26:8554@2024-12-09T20:52:54
msg: 21001@2024-12-09T20:52:55
data: Length: 332 (0x14c) bytes
0000:   80 88 00 00  f8 c3 82 98  f0 f0 bb 7c  d5 d5 d5 d5   ...........|....
0010:   d5 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5   ................
0020:   d5 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5   ................
0030:   d5 d5 55 55  55 55 d5 d5  d5 d5 55 55  d5 d5 d5 d5   ..UUUU....UU....
0040:   d5 d5 d5 55  55 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5   ...UU...........
0050:   d5 d5 d5 d5  d5 55 d5 d5  d5 d5 d5 d5  d5 d5 d5 d5   .....U..........
0060:   d5 d5 d5 d5  d5 d5 d5 d5  d5 d5 d5 55  55 55 d5 55   ...........UUU.U
0070:   55 55 55 55  d5 d5 d5 d5  d5 d5 d5 55  d5 55 d5 d5   UUUU.......U.U..
...204 (0xcc) bytes not shown...
2024-12-09T20:53:24.853386  WARN        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_base::clock: opening rtsp://192.168.151.187/tenda took PT30.000978861S!
2024-12-09T20:53:24.853572  WARN        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN
caused by: deadline has elapsed
2024-12-09T20:53:25.853791  INFO        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.151.187/tenda
2024-12-09T20:53:55.855099  WARN        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_base::clock: opening rtsp://192.168.151.187/tenda took PT30.001111804S!
2024-12-09T20:53:55.855256  WARN        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN
caused by: deadline has elapsed
2024-12-09T20:53:56.855483  INFO        s-CP3PRO-main streamer{stream="CP3PRO-main"}: moonfire_nvr::streamer: opening input url=rtsp://192.168.151.187/tenda
[... cut ...] ived SIGINT; shutting down gracefully. Send another SIGINT or SIGTERM to shut down immediately.
2024-12-09T20:53:58.989893  INFO                 main moonfire_nvr::cmds::run: Shutting down streamers and syncers.
2024-12-09T20:53:59.023788  INFO          s-test-main streamer{stream="test-main"}: moonfire_nvr::streamer: shutting down
^C2024-12-09T20:54:08.726849 ERROR                 main moonfire_nvr: exiting due to error err=CANCELLED: immediate shutdown due to second signal (SIGINT)
root@www:~#

scottlamb, 2024-12-11:
@nalonsor1 I opened a new issue #331 for interoperability with your camera. From those logs, the problem is unrelated to #33; the camera is using H.264 rather than H.265.

commit b4836f3a (2025-01-28): v0.7.18: support H.265

There's no transcoding, so if the browser/player doesn't support these
formats, they don't play. But it will record and allow downloads, and
it seems to be working with Chrome on macOS at least.

Fixes #33
closed 2025-01-28 by scottlamb