scottlamb/moonfire-nvr #35: web configuration UI

scottlamb, 2018-03-09:
The TUI `moonfire-nvr config` is a decent way of configuring the system for now, but long-term I'd like all configuration to be done through the web UI:

   * lots of Javascript work to do for this
   * new JSON interfaces to define
   * server refactoring: the `Streamer` and `Syncer` interfaces in the server don't have any teardown logic, aren't exposed to the web UI in any way, etc.
   * needs to be authenticated, have a notion of a config permission
   * need a way to bootstrap the authentication if not `moonfire-nvr config`. I'm thinking a simple CLI for user manipulation which authenticates to the server over a Unix domain socket protected via file permissions or `SO_PEERCRED`.

scottlamb, 2019-06-24:
Regarding the final bullet point about Unix domain sockets, see: https://github.com/seanmonstar/reqwest/issues/39

scottlamb, 2021-04-01:
There are a lot of usability problems in the existing config interface, as described in [this comment](https://github.com/scottlamb/moonfire-nvr/issues/84#issuecomment-656150961). I'd love to address those in a web based-UI, where there's more screen real estate for explanation.

clydebarrow, 2021-08-24:
I'm willing to tackle the UI side of this. However there is AFAIK currently no exposed JSON API to manipulate the config data, so this would be a prerequisite. Should that be split out as a separate issue? I have no Rust experience.

scottlamb, 2021-08-25:
Awesome. Very glad to have your help. I Filed #153 for the server-side work, and I'll start working through it soon.

scottlamb, 2023-07-04:
@sky1e continuing our conversation from #276:

> @scottlamb wrote: FYI, in case you have too many more TUI config improvements in mind, I should warn you: I'd ultimately like to switch everything to web-based config (https://github.com/scottlamb/moonfire-nvr/issues/35).

> @sky1e wrote: I was presuming that even with a decent web ui, the TUI wouldn't just be removed. There are advantages to having a UI that requires nothing but terminal access to the server, as an option.

I actually am imagining having it configured only over HTTP (never having to shut down the server), and dropping the TUI/`cursive` entirely. Ultimately Moonfire isn't that useful without HTTP access. I'd like to have a really nice, unified UI that has tooltips, previews of cameras as you're setting them up, etc. And just to have a super light-weight binary (I'm planning on dropping the Docker container stuff too).

Can you be more specific about the advantages you see of a terminal-based UI? I mostly see one: it can be a way of bootstrapping authentication. I've been planning to address that in a few ways:

* you can get an authenticated HTTP endpoint on localhost by Unix socket authentication + `ssh -L localhost:8080:/var/lib/moonfire-nvr/sock moonfire-nvr@nvr-host` (as mentioned in the [config reference](https://github.com/scottlamb/moonfire-nvr/blob/master/ref/config.md)).
* we could have a very basic CLI (not curses-like TUI, I'm thinking simpler) that adds/removes users via a local Unix auth socket on a running Moonfire.
* and/or it could use its local Unix auth socket privileged access to create a "configuration token": basically something you paste into the web interface to get access before any user accounts are even set up.