scottlamb/moonfire-nvr #37: pure Rust RTSP client library

scottlamb, 2018-03-09:
Drop ffmpeg/libav for RTSP (possibly dropping ffmpeg entirely; maybe we'd end up using it anyway for H.264 decoding). Use a pure Rust RTSP library instead. None seems to exist today, so probably write one.

This would make the install instructions easier, particularly if you want to have a ffmpeg version which supports timeouts properly (Raspbian Jessie's packaged libav doesn't) and/or works well when the camera has audio enabled (see #36).

It also could be a bit more efficient, although it's surely insignificant compared to on-NVR motion detection.

Maybe in the future this could even lead to RTSP server/proxying support, so you can expose live video to the Internet without having to expose the cameras' RTSP ports directly. (I'm sure they're horribly insecure in every imaginable way.)

jlpoolen, 2021-03-14:
I'm coming to the mindset that ffmpeg when used as a client against my Reolink cameras has shortcomings.  That brought me back to this issue.

Since you use ffmpeg as an externally created rtsp client, what about the possibility of using an alternative?   Somewhere in my investigation I determined that Reolink is using Live555 code for their rtsp servers on the cameras I have.  Live555 is open source and they have a client (I may try to build the client and then see how it fares with long open connections to my Reolink cameras) and an integrated client library with less bells and whistles.

I'm just wondering if the integration of ffmpeg into Moonfire-nvr was a Herculean effort that you do not want to undertake for another outside client.  It looks like a pure rust version for rtsp V. 1 is a long way off as of 3/14/2021.  I'm thinking the ffmpeg client may be the weak link and wondering if the Live555 might be more robust, especially at handling network interruptions, disconnects or other problems long extended connections (days, weeks, months) present.

Have you tried the Live555 client?  If not, I'll do so.  I do not want to repeat steps you have already taken and determined to be unproductive.

scottlamb, 2021-03-15:
I haven't tried Live555. I don't know off-hand how difficult it'd be to use, but it'd be interesting anyway to see if Live555 is significantly more able than ffmpeg to maintain RTSP sessions with those cameras.

scottlamb, 2021-04-15:
btw, I also think pure-Rust RTSP support might be closer than it appears. There are several interrelated pieces to replace the ffmpeg RTSP usage:

*   extracting width and height from the H.264 SPS/PPS. Moonfire NVR already uses the [h264-reader](https://crates.io/crates/h264-reader) crate to get some other data from them; adding this part shouldn't be hard.
*   RTP, RTCP, and SDP: the recent webrtc.rs project has added crates for these that look pretty good. see [webrtc-rtp](https://crates.io/crates/webrtc-rtp), [webrtc-rtcp](https://crates.io/crates/webrtc-rtcp), and [webrtc-sdp](https://crates.io/crates/webrtc-sdp). They're young, and focused on WebRTC, but they might actually have everything we need.
*   RTSP itself. This is a relatively small piece.

scottlamb, 2021-04-29:
I played around a bit here: https://github.com/scottlamb/moonfire-playground/commit/b9ad194ab0d7920afce0ce0f7c8ba0d74829a660 It's indeed not too hard to do the basic stuff to get RTP data flowing.

scottlamb, 2021-05-06:
I'm trying to get some API feedback here for a new RTSP crate. https://users.rust-lang.org/t/rtsp-streaming-api-sanity-check/59464

jlpoolen, 2021-05-07:
yeah Team!  I'll try to spend some time this weekend... Scott has been 
busy and I need to catch up.

scottlamb, 2021-06-04:
> btw, I also think pure-Rust RTSP support might be closer than it appears.

Well, I was pretty far off with this. There was a lot of work left to do (eg neither RTP library really had the codec depacketization stuff in the way I needed). But I'm making a pure Rust RTSP library anyway! I moved it from the playground spot mentioned above to its own repository: https://github.com/scottlamb/retina Lots of unfinished stuff there but it basically works.

I haven't integrated it into Moonfire NVR yet. It expects to run on the tokio threads, and those sometimes stall in Moonfire NVR now due to #88 , so I might fix that first. And then I might make a flag for my pure Rust RTSP library vs ffmpeg rather than commit to it right away. And then perhaps on to the audio support...

jlpoolen, 2021-06-04:
Here's my build log and notes for those who do not want to think, but 
just do it:

    Quick Steps on How to Build and run the example client

    Note: requires rust version 1.52 or higher

    [from /usr/local/src where "git clone [URL]" is performed]
    cd retina
    cargo build
    [ There will be aproximately 169 crates.]

    To see help:
      cargo run --example client help

    To pull an MP4 file from a Reolink camera
      cargo run --example client mp4 test_output.mp4 --url
    rtsp://192.168.1.48:554/h264Preview_01_main --username [user]
    --password [secret]

    creates an outputfile of 17M after running for several seconds then
    Ctrl-C to terminate process.

    MP4 has AUDIO!!

Yeah Scott!
-- 
Email Rider

John Laurence Poole
1566 Court ST NE
Salem OR 97301-4241
707-812-1323 office

scottlamb, 2021-06-29:
Moonfire NVR 0.6.4 uses the new RTSP crate by default. I'll wait a while for bug reports before ripping out ffmpeg support. Once I fully commit to the new library, I'll likely rework the threading model (no more need for a thread per stream).

closed 2022-03-18 by scottlamb

jlpoolen, 2022-03-18:
Bravo.