scottlamb/moonfire-nvr #44: switch database to a simple key/value store for speed

scottlamb, 2018-03-09:
SQLite3 is a super-reliable database, was easy to get started with, and is easy to diagnose problems with, but it's not particularly fast. I've noticed in CPU profiles: its locking around column operations, its VM operations, and its prepare steps. The latency for some operations on the Pi2 isn't quite what I'd like it to be. Consider switching to a simpler key/value store, such as:

   * RocksDB
   * spacejam/sled (warning: currently very experimental)
   * lmdb

robotmay, 2018-03-24:
I've spent quite a bit of time recently trying to find a good embedded database for Rust, but there's not much about at the moment. Sled is probably the most interesting, albeit quite new, and it's under active development.

Is Redis an option? The Rust clients are quite well developed and used. Obviously it's not an embedded store, however.

dolfs, 2018-03-24:
It all depends on what you want from the alternative store. Originally you mentioned speed and that a simple key/value store would provide enough functionality. Assuming that to be true, REDIS can fulfill the role, but it will not be fully embedded. Redis is an in-memory store, but runs in a separate process (so it can be shared). As such you’ll have to run a client on the embedded side. Communication will be over (Unix) sockets which has some overhead compared to a true and fully embedded solution.

The other issue, not explicitly mentioned, is that of persistence. Sqllite is currently being used not only as a data store, but you are also taking advantage of the “implied” persistence of all data in it, which is updated after each write. REDIS does not have this model per so. There are different persistence models available (https://redis.io/topics/persistence <https://redis.io/topics/persistence>), each with various pro/cons.

The choice of a “separate” process data store may create more memory pressure than the current SQLite solution. This may be unwanted on systems with a smaller memory footprint such as rPi of various models.

I also seem to think that you currently use indexing capabilities to maintain and recycle sample storage on disk. Is this just as easy or efficient using a pure key/value store? Should you consider a mixed solution?

scottlamb, 2018-03-25:
> I've spent quite a bit of time recently trying to find a good embedded database for Rust, but there's not much about at the moment

[sanakirja](https://crates.io/crates/sanakirja) is another pure-Rust option, fwiw.

> Is Redis an option? The Rust clients are quite well developed and used. Obviously it's not an embedded store, however.

I'd like to stick with an embedded database; I want the system administration aspect to be trivial, and it's hard to beat embedded for that.

> The other issue, not explicitly mentioned, is that of persistence.

We definitely need atomic, durable operations. I think that's possible with any of the ones under consideration. I think consistency is reasonable to maintain at the application layer (my code mostly does on its own now by accident). Isolation is unneeded (as there's no concurrency anyway; all the database operations are done with from the one process with a mutex held to keep the specialized in-memory data structures in sync).

> I also seem to think that you currently use indexing capabilities to maintain and recycle sample storage on disk. Is this just as easy or efficient using a pure key/value store? Should you consider a mixed solution?

The current schema has really only one index of note (`recording_cover`), and maintaining that manually wouldn't be a problem. The code already keeps the `recording`, `recording_playback`, and `recording_integrity` tables in sync manually. If anything, that'd be a little easier in RocksDB due to its support for column families. (It'd also be nice to configure different compression options for the playback column family.)

---

I'm not in a hurry to switch. SQLite is not as fast as I want, but otherwise it's working well now, and obviously having a SQL interface has some advantages for debuggability. I can wait and see if sled becomes more stable by the time this reaches the top of the (extensive) todo list.

In the meantime, I'm going to slowly work toward separating the database layer in two: the part that just manipulates the on-disk data structures, and the part that layers on all the specialized in-memory caching and batching. Besides making it easier to switch, I think this will aid testability and help slow the previously unbounded growth of `db.rs`. I've already started this as part of the many changes on the `new-schema` branch (which I hope to merge in the next week, btw).

jlpoolen, 2020-05-26:
A consideration: I'd like to be able to tap into the database during its run from another application.  SQLite really does not support concurrent connections, so its really a captive database.  I'm wondering if the design could be such that postgreSQL could be substituted in for those who have a postgres database or willing to run one along with Moonfire.

What I am thinking of is running Moonfire and then using Perl to access the database to determine what is going on.  I haven't studied that portion of the code that creates MP4 on the fly, but my hope is that I could use Moonfire as an aggregator and capture of content and then using other applications to pull out streams and process and/or analyze.  The [Motion-Project ](https://github.com/Motion-Project/motion) looks very interesting and I am wondering if I can use the [v4l2loopback ](https://github.com/umlaeute/v4l2loopback) to feed extracts out of Moonfire.

Any rate, as you redesign your schema, you could have future options if the schema is design such that it can be ported.  Of course, perhaps there is a tight integration in your Rust code that cannot be achieved with PostgreSQL or MariaDB?

scottlamb, 2020-05-26:
I want it to be possible to read/stream video from another application. However, I'm intending for everything (web clients, mobile clients, programmatic clients) to use the HTTP API rather than talk directly to the database. I can think of at least three reasons to prefer this approach:

*   my general philosophy (in any distributed or complex system) of separating API from storage wherever possible. This means clients don't have to be as intimately familiar with the server's subtle implementation details and "clever" optimizations. Instead, they can deal with higher-level concepts. It means that it's possible to maintain compatibility across a much wider spread of client and server versions if desired. And it means the single owner of the database is responsible for maintaining invariants, so you know what to blame if the database gets corrupted.
*   caching. Moonfire NVR assumes it's the only _writer_ to the database and has in-memory indexes that will get stale if that assumption is violated. Eg, it maintains a mapping of calendar days to how much video was recorded in that day. (It's easier to keep this just in-memory. We can just load the time zone once on startup and not deal with it changing. If we persisted it to the database, we'd need invalidation when the administrator changes the time zone or Congress decides to change the Spring Forward / Fall Back dates again or whatever.) It also assumes it's the only _reader_ of the database. It defers flushing stuff to the database to minimize disk seeks and SSD write cycles. If you directly read the database, you won't be able to see some available but unflushed video segments. If you go through the HTTP API, it will combine the database contents with in-memory state to tell you about them immediately.
*   implementation and testing simplicity and flexibility. I've written software that supports multiple SQL backends but I'd rather avoid testing all those possibilities here. And supporting both SQL and not wouldn't be very fun; I'd likely get the worst of both worlds.

Any particular reason to avoid using the HTTP API? If it's missing a feature you need, I'm definitely open to fixing that. We can also serve it on a Unix domain socket (in addition to the TCP port) so that you don't have to deal with password/cookie authentication for local programmatic stuff. I want this anyway to bootstrap that authentication from the CLI while the server is running.

I've actually been thinking about dropping this issue. SQLite is very reliable and convenient, and as I start thinking about on-NVR analytics (which are likely to be relatively CPU-heavy), worrying about SQLite's small contribution to CPU and latency seems increasingly silly. Resource-wise it's not optimal (parsing, interpreter overhead, additional round trips for foreign key enforcement and such) but it's good enough. And other engines have their caveats. [This article](https://github.com/the-lean-crate/criner/issues/1) was interesting.

closed 2020-05-31 by scottlamb