scottlamb/moonfire-nvr #45: cross-compile to ARM

scottlamb, 2018-03-09:
Compiling Rust code on ARM is really really slow. I've seen cases where my Pi2 simply can't do it due to lack of RAM. Figure out how to cross-compile Moonfire NVR from x86-64 to ARM and write up instructions.

I've done some work / wrote down some notes in a Google doc but haven't gotten a fully working setup.

robotmay, 2018-03-24:
It can be a bit of a faff based on dependencies. One of the things you might run into is an OpenSSL issue, which I got around at some point by vendoring the library and setting some env vars:

`OPENSSL_LIB_DIR=$(realpath arm/openssl-1.0.1t) OPENSSL_INCLUDE_DIR=$(realpath arm/openssl-1.0.1t/include) cargo build --target armv7-unknown-linux-gnueabihf --release`

However https://github.com/japaric/cross has progressed since I had to do it, so that might be a good option!

jpumc, 2019-08-23:
Hi!

I managed to cross-compile Moonfire NVR to Pi1 from x86_64. I had to copy some files from the rpi to form a sysroot, download the toolchain from https://github.com/raspberrypi/tools/tree/master/arm-bcm2708/gcc-linaro-arm-linux-gnueabihf-raspbian-x64, create a .cargo/config and set some environment variables.

## sysroot
requires the c dependencies to be installed in the sysroot, then copy the raspbian installation to a dir on the build machine

## env vars, replace $TOOLCHAIN and $SYSROOT with their respective paths
```
PATH=$TOOLCHAIN/bin:$PATH
PKG_CONFIG_SYSROOT_DIR=$SYSROOT
PKG_CONFIG_ALLOW_CROSS=1
TARGET_CFLAGS=--sysroot=$SYSROOT
```

## `.cargo/config`, replace SYSROOT with its path
```
[target.arm-unknown-linux-gnueabihf]
linker = "arm-linux-gnueabihf-gcc"
rustflags = [
	"-C", "link-arg=--sysroot=SYSROOT",
]
[build]
target = "arm-unknown-linux-gnueabihf"
```

and then `cargo build` should work, if I didn't forget anything.

jlpoolen, 2020-05-29:
I used to cross-compile for ARM using Gentoo.  I haven't done it for years and the machine I had set up my cross compiler on is decommissioned, but I still have my various mini computers, e.g. Plug and BeagleBone.  One of the tricks I used was to use Gentoo's distributed cross-compiler, so I'd compile on a Plug, but something like compiling Firefox would take a very long time, so I'd send it off to an AMD powerhouse which would perform the compile in 20 minutes instead of the 24 hours on a Cortex processor.  What's the hardware you want to run the compiled version on?  A link to where I can buy the target computer would be helpful.  I worked with Plug, Efika and Beaglebone, never got into Raspberry Pi.

jlpoolen, 2020-05-29:
Yup, good ol' Neddy Seagon at Gentoo on a Raspberry Pi 4 entitle "Raspberry Pi4 64 Bit Install"
>**What works**
> Everything"

https://wiki.gentoo.org/wiki/User:NeddySeagoon/Raspberry_Pi4_64_Bit_Install

scottlamb, 2021-01-22:
I'm now using a Docker-based cross compilation setup. It's nice in that I can build amd64, arm64, arm32 with a single command. The target is also a Docker container. In theory we could target running on (say) a Raspberry Pi OS host as in jpumc's comment, but that requires downloading the specific distribution(s) in question and customizing the build/bundling steps more. I think the Docker container approach is pretty good, so I'll declare victory for now.

closed 2021-01-22 by scottlamb