scottlamb/moonfire-nvr #47: Moonfire cannot serve all static files present

dolfs, 2018-03-10:
NOTE: This is sideways related to #25 

I was doing some UI development and an attempt at producing an optimized bundle produces source maps in files separate from the bundle itself:
```
nvr.bundle.js
nvr.bundle.js.gz
nvr.bundle.js.map
```
#25 mentions the inability to serve the .gz version of the bundle when gzip is offered as acceptable. This would be a little special code in lieu of on-the-fly gzipping.

The issue addressed here is that therein bundle (nvr.bundle.js) attempts to load the source map in a separate request. The request fails with a 404 status. It seems Moonfire filters available static files, possibly by extension? May be not a good idea.

If a request in a url matches an existing file, it should probably be served up, but I can see how a complication would be to have a mapping of extensions to mime types. Perhaps it is as simple, at least for this case, to add something for .map.

scottlamb, 2018-03-10:
Yes, currently it has a simple mapping from extension to MIME type, and it skips extensions it doesn't know about. You can see the code in `fill_ui_files` in `src/web.rs`. What MIME type should `.map` use?

dolfs, 2018-03-10:
JavaScript

commit 5854b31b (2018-03-11): serve .map files as Javascript

This fixes #47 for now at least.
closed 2018-03-11 by scottlamb