scottlamb/moonfire-nvr #49: Consider containerizing Moonfire using Docker?

dolfs, 2018-03-12:
A benefit might be that it is easier to create the exact environment needed (dependent library versions etc.). A potential downside is that (a) docker would need to be installed (if not already) and (b) there might be some overhead that is not acceptable on the smallest of systems such as rPi.

scottlamb, 2018-03-12:
This would basically be a change to the recommended installation procedure, right?

dolfs, 2018-03-12:
If we decide to do the docker thing, then it would be a change, or we could decide to keep both methods.

I just wanted the question logged. I don't even know, at this point, whether the docker approach is viable on smaller systems. On the other hand, for anybody using another system that is multi-purpose it might be nice to not have to install packages globally etc.

Perhaps we just defer further research and decision until somebody expressed a true need. The work needed to enable this will roughly remain the same between now and then.

scottlamb, 2018-03-12:
I hope the dependent library question will be basically moot eventually. Rust and Javascript libraries are nicely auto-installed through `cargo` and `yarn`, respectively. As far as C++ libraries go, I'm only using:

   * OpenSSL. I'd be surprised if there weren't a suitable version installed already; I don't need any particularly new features, lots of software depends on this, and I think distributions and OSs are pretty serious about keeping fairly up-to-date for security reasons.
   * ffmpeg. This one is a huge pain now, but I want to get rid of it for the RTSP stuff at least. See #37.
   * sqlite. The sqlite authors have an "amalgamation" version (the whole library in one giant `.c` file) for ease of compiling in statically, and the Rust crate I'm using supports doing so if you do `cargo build --features=bundled`.
   * ncurses, for `moonfire-nvr config`. I'd to do a web config UI instead (#35).

dolfs, 2018-03-17:
I have a basic Dockerfile that, when I test it on my Mac, results in an Ubuntu instance with Moonfire completely built. I have not gone the stop of installing yet, and I have to solve the question of where to have the samples dir reside (probably need to use Docker Volumes), but it looks like this setup, when provided will allow Moonfire to be fired up under Docker.
I will also attempt to produce an image where a complete build can be done using cross compilation tools. If that succeeds it can be a vehicle to do various different cross-compile-builds allowing us to distribute pre-built packages.

scottlamb, 2018-03-21:
Wow, if Docker makes it possible to compile for Linux/arm from macOS/x86-64, I'd be very happy.

dolfs, 2018-03-21:
It should be possible. I have this docker running and doing a full build without problems. Now I need to make the next step and base the docker image on one with cross compilation build tools (I’ve found several). Along with that there will be a bit of a challenge because you cannot simply use the same commands, but have to use the x-compile binaries. In addition, I’ll have to install rust etc. So this is a work in progress. But, yet it would be very nice and if we get it going, it should also become possible to do this for several platforms and produce pre-compiled installable binaries, lowering the barrier to entry.
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

scottlamb, 2021-01-22:
This is now the recommended way to install the system. It even supports cross-compilation. Thanks for the suggestion!

closed 2021-01-22 by scottlamb