scottlamb/moonfire-nvr #50: favicon

scottlamb, 2018-03-12:
It's a small thing, but I keep lots of Chrome tabs open and have trouble finding the Moonfire one because there's no favicon. Make one. Probably red (to match the name moonfire) and some fitting imagery:

   * camera: profile view of a bullet style is probably most identifiable in few pixels. Maybe a head-on view of a bullet style with glowing red LEDs.
   * moon: probably a crescent to be identifiable as something other than just a circle.
   * tree: a [moonfire japanese maple](https://www.google.com/search?rlz=1C5CHFA_enUS705US705&biw=1519&bih=953&tbm=isch&sa=1&ei=8B2mWvrRH4Xi0gKQn6Fo&q=palmatum+moonfire&oq=palmatum+moonfire&gs_l=psy-ab.3..0i5i30k1.3951.5941.0.6029.17.15.0.1.1.0.208.1215.9j3j1.13.0....0...1c.1.64.psy-ab..3.14.1217...0j0i67k1j0i30k1j0i8i30k1.0.WgOVHCrdGag). might be hard to represent well in such a limited number of pixels. the leaf unfortunately is easily confused with marijuana.

It'd be nice to have a larger logo as well for display on the UI, the main page, maybe in the future an icon for phone apps. But better to have a just a favicon for now than nothing.

dolfs, 2018-03-12:
I just put in a fairly full setup, supporting all the various formats the web and mobile require. The image itself is a moon on fire coming over a forested mountain ridge, with a bullet cam super imposed. Not super artsy, but I suppose it will do for now.

scottlamb, 2018-03-13:
As mentioned on the PR #51, I'm afraid we can't use those images. :(

I just did some searching and found thenounproject.com has a decent collection of licensable icons. We could do CC-BY licensing (not exactly sure where best to put the attribution) or I'd be happy to just pay the $1.99 per icon. I kind of like [this one](https://thenounproject.com/search/?q=72530&i=72530), maybe in a red circle or with a red crescent behind it. These are simple enough icons that they scale down pretty well to Chrome's tab icon size.

dolfs, 2018-03-13:
I pushed an update with a reasonable image (for now), the is Creative Commons licensed for free use. It does also not use the camera.

For a favicon you need more than one size image, so licensing an icon is not going to cut it. Anyway, I am not an artist so I offer this as a reasonable placeholder that does not have the problems you originally commented on.
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

scottlamb, 2018-03-13:
Oh, while you were doing that I played with the icon I mentioned and came up with this (32x32 version):

[image]

Looks like realfavicongenerator.net can generate all the various formats.

dolfs, 2018-03-13:
Yes, that is what I used. I see no moon and no fire in your icon, but, granted, it has a camera.
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

scottlamb, 2018-03-13:
You're meant to imagine the circle as a moon, and the color as fire-like. /shruggie

I'm not an artist either; but IMHO at low resolution something icon-based with few elements is more identifiable than something photo-based.

I'm open to changing it up, especially if we get a real artist involved, but I at least have something with this that gets the icon to pass through webpack and web.rs with the proper mime type and such to work.

dolfs, 2018-03-13:
Well, passing through was accomplished with what I delivered too. Anyway. Enough said.
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

scottlamb, 2018-03-13:
Oh, so it did. I was all proud I made something work (remember, I really know less than you'd expect about making websites). I had to modify `web.rs` to make the `.ico` work and the webpack config to use the name `favicon.ico`, but I guess your version had Chrome load `.png`s (using the existing `web.rs` MIME mapping) and webpack's default hashed name substitution.

scottlamb, 2018-03-14:
I'll switch it to use your approach then.

dolfs, 2018-03-14:
How about I add your camera icon to the updated and CC moon on fire image I have in here? I am sure things could be improved later...
Dolf Starreveld                  	
email <mailto:registration@starreveld.com>

commit 320374c6 (2018-03-13): favicon: security camera in orange/red circle

This is derived from the following icon:
https://thenounproject.com/term/security-camera/72530
by iconsmind.com. I paid for the royalty-free license.

Fixes #50.
commit 45abeb22 (2020-05-30): overhaul HTTP serving and caching

* use content-hashed paths for static resources (except the top-level
  request), with immutable Cache-Control headers. This should improve
  cache behavior in both directions: avoid preventable HTTP requests and
  cause immediate refresh when needed. I had some staleness when
  browsing with my phone.

* set up the favicons properly while I'm at it (closes #50). I used the
  convenient favicons-webpack-plugin to build everything from a .svg.
  I've hit an error similar to lovell/sharp#1593 at least once though so
  I might change my mind about that part if i [... cut]
closed 2020-05-30 by scottlamb