scottlamb/moonfire-nvr #58: First Run Install Error

KristopherKubicki, 2018-04-30:
Hi,

I attempted a fresh install on a new Ubuntu 18.04 environment from the master branch.  script/build.sh went fine until this:

`./scripts/script-functions.sh: line 292: /home/kristopher/code/moonfire-nvr/src/schema.sql: No such file or directory`

and during startup: 

```
Apr 30 08:23:16 Sal systemd[1]: Started Moonfire NVR.
Apr 30 08:23:16 Sal moonfire-nvr[16201]: main moonfire_nvr] SqliteFailure(Error { code: CannotOpen, extended_code: 14 }, Some("unable to open database file"))
Apr 30 08:23:16 Sal moonfire-nvr[16201]: stack backtrace:
```

I saw that schema.sql was removed from src, so I ran this command:

`sudo -u moonfire-nvr sqlite3 ~moonfire-nvr/db/db < ~/code/moonfire-nvr/db/schema.sql`

After which, the syslog returned:

```
Apr 30 08:29:08 Sal systemd[1]: Started Moonfire NVR.
Apr 30 08:29:08 Sal moonfire-nvr[17050]: thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: QueryReturnedNoRows
Apr 30 08:29:08 Sal moonfire-nvr[17050]: stack backtrace:
Apr 30 08:29:08 Sal moonfire-nvr[17050]:    0:     0x55f82da4626e - backtrace::backtrace::libunwind::trace::hd98210360bf7305b

```

Attempting to run the config yeilds:

```
kristopher@Sal:~/code/moonfire-nvr$ sudo -u moonfire-nvr moonfire-nvr config
E0430 084330.569 main moonfire_nvr] QueryReturnedNoRows

```

Unfortunately I am stuck here and cannot get the NVR to load.  Please advise.  Thanks!

scottlamb, 2018-05-01:
Thanks for the report! The main problem is that the install scripts have bitrotted. I test the install procedure rarely if ever; I just upgrade my existing setups, and my main workstation is macOS rather than Linux anyway. A couple changes broke this:

  1. commit 168cd74 added schema versioning to the database. The error you're seeing is that it found no row in the `version` table. (And the error message is poor; my code has better handling for the table being entirely absent or for having a row with less than the current required version.) As of this commit, the install should be done via `moonfire-nvr init` rather than `sqlite3`.
  2. commit a5528e1 moved the schema file, so the `sqlite3` command obviously fails.

I just committed a quick fix to this script. As for preventing the scripts from bitrotting...I've been meaning to set up Travis CI; maybe I can make it actually use the install scripts.

KristopherKubicki, 2018-05-13:
OK - I got it up and running :) I had to delete the sqlite3 db file first.  

Looks great - excited to dive into it more

closed 2018-05-13 by KristopherKubicki