scottlamb/moonfire-nvr #59: Live stream of the video in Web UI.

p000, 2018-11-15:
Hello Scott, 
How to get the live stream of the video on the web browser form all the camera that are configured.

scottlamb, 2018-11-20:
Moonfire NVR doesn't support live streams yet. When I want a live stream of my cameras, I use an RTSP client (such as VLC) to access the cameras directly (or via a VPN if I'm not at home), bypassing Moonfire NVR.

It's definitely a feature I want, but it's not trivial, so it probably won't happen for the 1.0 release (which I intend to be the "minimum viable product"). Here's some notes below.

I think the best user experience would be to have one interface that can handle both recorded video and live monitoring with main or sub streams. When you open it up, it defaults to showing live. It also shows a scrub bar at the bottom. You can go to older stuff by interacting with the scrub bar. The scrub bar should show what times have video available, and that should reflect stuff newly recorded since you opened the page. There should be a button to jump to live and stay there.

I'm not sure how to achieve that. In a web browser, the closest thing that seems possible is HTML5 Media Source Extensions (MSE)—you can add segments to an existing video element as you go. In fact, I think this is basically the only way to do anything approaching live full-quality video in the browser at all; they don't support RTSP.

The MSE standard [specifies](https://www.w3.org/TR/mse-byte-stream-format-registry/) three video formats: webm, mp4, and mp2t.

Moonfire already knows how to generate .mp4, but I think it might add one key frame interval of latency. The complete index for a MSE segment has to come before any of the frames. I believe each MSE segment has to start with a key frame. So I think that means either Moonfire would have to wait for everything that depends on a given key frame before sending it (i.e., until the next key frame) or you'd have to re-add the same key frame whenever new P frames are added (either a ridiculous waste of bandwidth or duplicating Moonfire's logic for synthesizing .mp4s into the client side). It might also be prohibitively expensive client-side for the CPU to decode the key frame 30X as often; we'd have to experiment.

mp2t is better in this regard; it's designed for live streaming, so it just gives frames with their timestamps as they arrive. But only Safari and Microsoft Edge [support](https://developer.jwplayer.com/articles/html5-report/adaptive-streaming/mediasource.html) it. I suppose we could add support to Firefox and Chromium and wait for it to make it into releases people actually use but that's a lot of extra work/uncertainty. Even with mp2t I'm unsure if you can process a partial segment and continue to extend it or if the browser won't do anything with it until the full segment has arrived.

I think webm is similar to mp4.

Maybe duplicating the .mp4 logic into the client side is the most realistic option, although we could check first if it is possible to have a MSE segment that doesn't start with a key frame (instead being based on the previous segment), which would make it a lot easier.

scottlamb, 2019-02-07:
Interesting article and discussion:

https://mux.com/blog/the-low-latency-live-streaming-landscape-in-2019/
https://news.ycombinator.com/item?id=19096692

I hadn't considered the WebRTC approach. The backend support I added here (a `Transfer-Encoding: chunked` stream with single-GOP fragmented mp4 files) is the next best thing possible with Chrome/Firefox, so I feel pretty good about that.

The frontend piece is still unimplemented...

p000, 2019-02-19:
Mr. Scott,
Will you be able to help me write a package same as this project in golang ?

scottlamb, 2019-02-19:
I don't understand the question. Moonfire NVR is written in Rust (the backend portion that runs on your server) and JavaScript (the frontend portion that runs in your web browser). I'm not looking to switch to Go.

p000, 2019-02-25:
> I don't understand the question. Moonfire NVR is written in Rust (the backend portion that runs on your server) and JavaScript (the frontend portion that runs in your web browser). I'm not looking to switch to Go.

Ok. Thanks

urbaniak, 2019-12-29:
Maybe it's possible just keeping the last frame in memory for each camera and serve/recode it as a jpeg, so we can have live preview (possibly refresh that in 1-5s?)

scottlamb, 2019-12-29:
> Maybe it's possible just keeping the last frame in memory for each camera and serve/recode it as a jpeg, so we can have live preview and possibly refresh that in 1-5s?

The backend support is already implemented for a live H-264 `.mp4` stream (with roughly 1 second of lag). That should allow the same experience with less bandwidth and less CPU usage on the server side than transcoding to `.jpg`. What's needed is JavaScript work to take advantage of it. One of these days I'll get around to doing it myself, but frontend stuff is really not my area of expertise, so I'd be thrilled if someone else is interested in working on it.

gawmanarnar, 2020-03-01:
Is this currently being worked on? Just checking in, I’d like to take a shot at it.

scottlamb, 2020-03-01:
Awesome, and great timing! I just verified it works for me with an embarrassingly horrible Javascript prototype. And made it work properly with 6 or more streams by switching the protocol to WebSockets. (With the HTTP/1.1 hanging GET, I was hitting Chrome's limit of 6 active HTTP connections per host. But apparently the limit's higher with WebSockets.)

If you're interested in making a polished JavaScript implementation, I'd be very glad for the help.

Eventually: I want a unified live view / recorded view UI as mentioned in the opening comment of this issue. I think most of the server-side pieces for that are in place already; the exceptions are an event stream (#40) and motion detection / video analytics stuff.

For now: we can certainly start small, like a separate live view UI, and build from there.

I'm attaching my live.js and live.html prototype. The code is embarrassing to read, it doesn't integrate with the main UI (and just assumes the main UI has already taken care of logging you in), it has no error handling, it's unquestionably buggy, it hardcodes my camera names for placement into the grid, etc., but it at least shows how to make the thing work in some cases.

[live.js.txt](https://github.com/scottlamb/moonfire-nvr/files/4271413/live.js.txt)
[live.html.txt](https://github.com/scottlamb/moonfire-nvr/files/4271414/live.html.txt)
[image]

scottlamb, 2020-06-30:
I recently discovered [WebCodecs](https://github.com/WICG/web-codecs). I hope this will eventually mean we can do stuff like just sending an RTP stream frame-by-frame.

In the meantime, the `/live.m4s` endpoint that sends one GOP (typically 1- or 2-second chunks) works.

Server-side, the remaining part here is that the endpoint doesn't tell you when the stream is connected or not. Particularly when viewing multiple streams simultaneously, it'd be good to distinguish between "this camera has dropped out; let the others advance" and "there's a slight hiccup in your network connection; pause them all for a moment".

scottlamb, 2020-08-08:
The `/live.m4s` endpoint on the `new-schema` branch now is low latency. The screenshot below has two windows in it:

*   Firefox with a prototype Moonfire NVR live view
*   VLC accessing the camera directly with a `rtsp://` URL.

and the former actually has slightly less buffering/latency than the latter; I'd guess 1/10th of a second.

[image]

scottlamb, 2021-03-31:
There's an actual released live view now. It doesn't keep camera views in sync yet, but it's progress.

closed 2021-03-31 by scottlamb