scottlamb/moonfire-nvr #60: error: failed to run custom build command for `openssl v0.9.24`

polyzen, 2018-11-19:
Can't quite seem to narrow down what depends on this. Based on `Cargo.lock`, that's `reqwest`, but `reqwest` is a `dev-dependency`, and I've tried adding `--bins`.

scottlamb, 2018-11-19:
What's the actual build failure—does your machine not have that older version of openssl available? If so, upgrading to reqwest 0.9.x presumably would fix it. There was a similar issue in one of my other projects: scottlamb/http-serve#13.

polyzen, 2018-11-20:
```
...
   Compiling aho-corasick v0.6.6
   Compiling parking_lot_core v0.2.14                                                                                  
error: failed to run custom build command for `openssl v0.9.24`                                                        
process didn't exit successfully: `/build/moonfire-nvr-git/src/moonfire-nvr/target/release/build/openssl-4c84590159ae59
9e/build-script-build` (exit code: 101)                                                                                
--- stderr                                                                                                             
thread 'main' panicked at 'Unable to detect OpenSSL version', /build/.cargo/registry/src/github.com-1ecc6299db9ec823/op
enssl-0.9.24/build.rs:16:14                                                                                            
note: Run with `RUST_BACKTRACE=1` for a backtrace.                                                                     
                                                                                                                       
warning: build failed, waiting for other jobs to finish...                                                             
error: build failed                                                                                                    
```

I do have [`openssl-1.0`](https://www.archlinux.org/packages/core/x86_64/openssl-1.0/) installed, but as you say, this is a common issue.

closed 2018-11-20 by scottlamb

polyzen, 2018-11-20:
Thank you for resolving this so quickly.

commit 071be03c (2018-11-20): update most deps, notably including reqwest

Fixes #60

The reqwest dependency is significant because the old version required
an old version of openssl, complicating compilation on newer platforms.
reqwest also pulled in old/duplicate versions of hyper, tokio, etc.
Nice to drop a lot of that cruft.

I left rusqlite and uuid alone because they had breaking changes I
didn't want to mess with at the moment.

Bumped the minimum Rust version to 1.30.0, as required by the
new encoding_rs crate (and perhaps other things).