scottlamb/moonfire-nvr #69: update UI deps

scottlamb, 2019-07-10:
The UI dependencies are quite stale. github is giving me security warnings about them (although I think they're more relevant to node-js uses than to what Moonfire NVR is doing). Let's update them.

In theory this is just a `yarn update` but if I do that, following builds fail. I imagine the configs need to be reworked in some way, etc. See some previous mention at #62.

@dolfs 

```
$ yarn build
yarn run v1.12.3
$ webpack --mode production --config webpack/prod.config.js
/Users/slamb/git/moonfire-nvr/node_modules/webpack-cli/bin/config-yargs.js:89
				describe: optionsSchema.definitions.output.properties.path.description,
				                                           ^

TypeError: Cannot read property 'properties' of undefined
    at module.exports (/Users/slamb/git/moonfire-nvr/node_modules/webpack-cli/bin/config-yargs.js:89:48)
    at /Users/slamb/git/moonfire-nvr/node_modules/webpack-cli/bin/webpack.js:60:27
    at Object.<anonymous> (/Users/slamb/git/moonfire-nvr/node_modules/webpack-cli/bin/webpack.js:515:3)
    at Module._compile (internal/modules/cjs/loader.js:722:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:733:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:560:12)
    at Function.Module._load (internal/modules/cjs/loader.js:552:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:22:18)
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

dolfs, 2019-07-10:
I’m vacationing in MT until Aug so I won’t even have time to look into this until then. 

+1 (415) 613-7229 (US)

scottlamb, 2020-03-02:
Fixed with 92c532d.

closed 2020-03-02 by scottlamb