scottlamb/moonfire-nvr #70: shrink moonfire-nvr binary

scottlamb, 2019-07-20:
There's some significant bloat in the binary right now:

 - [ ] debug symbols. These are handy, but they're enormous.  This is the vast majority of the reason for the large binaries.  (Unstripped binary is 150M. Stripped is 9.2M.) When I release, it'd be nice to do split these to a separate file as Ubuntu does.
 - [ ] `protobuf`. It's surprisingly large; maybe there's some low-hanging fruit.
 - [ ] `regex`. This was a convenient way of doing my parsing, but it's not strictly necessary, and `cargo bloat` says it's pretty large.
- [ ] `cursive`. It's a lovely library, but long-term I want a web config UI (#35) instead.
- [ ] duplicate crypto libraries: `openssl` and `ring`. `libpasta` uses `ring` iirc. I could just stop using `openssl`, although that might not be a good choice when I actually have built-in https (#27)
- [ ] two arguments parsing libraries: `clap` and `docopt`. I use `docopt`. `prettydiff` is pulling in `clap`, which at first glance seems silly. I might convert to clap anyway though so maybe there's no point in messing with prettydiff.
- [ ] two versions of `parking_lot`. (tokio currently uses 0.7; I use 0.9 in moonfire-{nvr,db,base} and in mylog.) tokio is likely to update soon-ish, so I can probably just wait for this to be fixed.

These aren't causing material problems right now; the big binaries are just a bit embarrassing. So no particular size we need to get under or anything.

```
$ cargo bloat --release --crates
Compiling ...
Analyzing target/release/moonfire-nvr

File  .text     Size Crate
0.8%  21.0%   1.0MiB std
0.4%  12.0% 602.5KiB moonfire_db
0.4%  10.3% 521.0KiB protobuf
0.2%   5.7% 286.6KiB regex
0.2%   5.3% 268.5KiB moonfire_nvr
0.2%   4.8% 242.2KiB cursive
0.2%   4.3% 215.5KiB regex_syntax
0.1%   3.7% 187.3KiB h2
0.1%   3.0% 148.8KiB docopt
0.1%   2.4% 121.9KiB hyper
0.1%   2.3% 116.2KiB futures
...
```

scottlamb, 2020-03-21:
The split debug symbols (aka "fission") would currently break backtraces, unfortunately:

https://github.com/rust-lang/backtrace-rs/issues/287

IronOxidizer, 2020-09-07:
There are many (10+) duplicate packages with different versions. This is likely increasing the size significantly. These can be reduced by using dependencies with matching dependency versions.

Here's a quick list:
```
ansi_term v0.11.0
ansi_term v0.9.0
error-chain v0.11.0
error-chain v0.12.2
proc-macro2 v0.4.30
proc-macro2 v1.0.10
quote v0.6.13
quote v1.0.3
rand v0.3.23
rand v0.4.6
rand v0.5.6
rand v0.7.3
rand_core v0.3.1
rand_core v0.4.2
rand_core v0.5.1
strsim v0.8.0
strsim v0.9.3
structopt v0.2.18
structopt v0.3.13
structopt-derive v0.2.18
structopt-derive v0.4.6
syn v0.15.44
syn v1.0.17
time v0.1.43
time v0.2.10
unicode-xid v0.1.0
unicode-xid v0.2.0
```

For the depedency graph, use `cargo tree -d`

scottlamb, 2020-09-08:
Definitely not ideal, although I suspect none of the ones in that list are a major contribution to the binary size given the `cargo bloat` output.

scottlamb, 2022-01-05:
With 0406e09, the binary shrunk from 154 MiB to 70 MiB. I'm still hoping to reduce it further with split debug info. From [this comment](https://github.com/rust-lang/rust/issues/34651#issuecomment-922454870) and [this PR](https://github.com/rust-lang/rust/pull/89819#issuecomment-1005936137) it looks like work is actively being done on that.