scottlamb/moonfire-nvr #80: Instructions On Creating an MP4 Extract

jlpoolen, 2020-05-28:
I do not recall seeing instruction on how to download a video, so I played around with the interface.  Here's a screen shot that show how to extract and then download an MP4 based upon the row selected's start and finish time:

[image]

The video frame displayed was created by clicking on a row and the resultling video inherited the rows start and end time.  Note, however, I had values, 22:04:41 and 22:05:31, populated in the side Date & Time range dialog box and the resulting video had the start & end times of the row I selected, not the values of  22:04:41 and 22:05:31. 

How do I create a video using the Date & Time range?  Do I need to enter values in first and second box of the Date & Time range dialog and then click a row that has time periods that encompass those times?

scottlamb, 2020-05-28:
Huh, that looks broken. When you have "Trim Segment Start & End" checked (as in this screenshot), you shouldn't see any rows with start times before your selected start time; it will trim the video to start at that time. Similar for end times. If the camera was recording continuously (no disconnects) and you have "max video duration" set to infinity, you should only see one row per camera. If it's set to something else, you'll see additional rows if your selected timespan is greater than that limit.

I'd guess this is related to your other UI problem (#79).

jlpoolen, 2020-05-28:
I've been playing around and seeing the results of changing things and I think I'm getting understand how the Recordings Display box affects things, or should affect things.  I'll take a stab at some documentation on the use of this box as I interpreted the box and its values to not necessarily mean what you have intended.

scottlamb, 2020-05-28:
Thanks! I definitely appreciate help improving the documentation.

The way it's supposed to work is that all the rows you see are ones that overlap with the selected time range. If you have trim checked (the default), you shouldn't see anything outside the selected time range.