scottlamb/moonfire-nvr #83: PoE, PPPoE, multiple cameras & RaspPi

JasonKleban, 2020-06-07:
My off-the-shelf NVR has 4 PPPoE PoE ports, each on a dedicated network interface (as I understand it) because of bandwidth considerations of the cameras (3-12MP cameras).  I don’t know if that isolation is necessary based on network adapter performance and cat5 bandwidth.  If it is necessary, how does one connect a raspberry pi to multiple network interfaces like that?  Do we need a usb hub with multiple usb network adapters?  Or is that silly, and we just need a network hub typically?  I know I can inject PoE separately if necessary.  I can try to do the math, but if someone already has practical experience for all the components involved, that would be more reliable.  Thanks!

scottlamb, 2020-06-07:
That's new to me. I have a few questions:

   * Do you have the model numbers of your NVR and cameras? Maybe we can track down some info in their manuals.
   * Have you looked at the cameras' web interfaces? Is the PPPoE something you can turn on and off?
   * What's the max bandwidth of your cameras (typically measured in Mbit/sec)? Typically you can configure the bandwidth used in the camera's video settings, within some predefined range.

It's certainly common to have a bunch of PoE Ethernet ports on the back of a NVR. (I have a Lorex LNR280 aka Hikvision 7608 that's like this.) My understanding was that it's just convenient to have the switch built in to the device. In terms of bandwidth, I think it's rare for cameras to use more than 10 Mbit/sec for the main stream. They commonly have 100 Mbit/sec interfaces rather than 1 Gbit/sec because they just don't use that much bandwidth. In aggregate then 4 cameras uses a lot less than 1 Gbit/sec, so the NVR's single NIC to the switch is also fine.

I don't really know what using PPPoE accomplishes. I don't see how it saves bandwidth—it actually uses more due to extra per-packet overhead though not really significantly. My cameras (a relatively new Dahua IPC-HDW5231R-Z and some now-ancient Hikvision DS-2CD2032-Is) support PPPoE but I've never turned it on. I assumed it was for some sort of weird edge case where the camera is connecting directly to an ISP that requires PPPoE or something.

In any case, PPPoE is "just" a protocol layer. The physical setup is the same as regular Ethernet. You can still connect the cameras to a normal switch (either a PoE switch or a regular one with separate injectors as you mentioned), connect the Pi to it, configure PPPoE interfaces on the Pi, and everything should work. In theory; I've never actually tried that configuration...

But since you mentioned physical setup and multiple NICs: one thing I like to do is isolate the cameras to their own network segment. I do this for security reasons: these cheap cameras are notoriously insecure, so I don't want them to be able to access anything on my home network. So it's helpful to either have two physical NICs on the NVR (and yes, you can plug a USB NIC into a Raspberry Pi) or use a switch that supports IEEE 802.1q VLAN tagging.

JasonKleban, 2020-06-09:
Thank you for those explanations.  I think I was confused about the purpose of the PPPoE and I don’t think it is used or required.  I will check on the max bandwidth which is most certainly in the typical range and consider a dedicated usb.  Looking forward to trying this out.

closed 2020-06-09 by scottlamb