scottlamb/moonfire-nvr #85: webpack ... Killed error ... 137 (Out of memory)

JasonKleban, 2020-06-12:
I have a new Ubuntu 20 on rpi 3b+ all the way updated.  This might not be a moonfire-nvr related issue at all, but I'll log it anyway since I can't find any great matches elsewhere.  Following guide/install-manual.md, looks like an out-of-memory issue in the webpack step?  Possibly a bad loop?  `cargo` stuff works fine and is probably more aggressive, if I had to guess.

```
yarn build
yarn run v1.22.4
webpack --mode production --config webpack/prod.config.js
Killed
error Command failed with exit code 137.
```

I didn't try the scripted install yet.  I tried upgrading all dependencies to latest with yarn, no change.

scottlamb, 2020-06-12:
I've never seen that before but I haven't tried running `yarn build` on a 1 GiB machine in a while. Two ideas:

   * [use zram](https://www.cnx-software.com/2018/05/14/running-out-of-ram-in-ubuntu-enable-zram/)
   * just build it on another machine and copy it over. Unlike the `moonfire-nvr` binary, none of the JavaScript/HTML/etc. files in the `ui-dist` directory are architecture- or OS-specific. So "cross-compiling" is trivial; you just build it somewhere else and copy it over.

scottlamb, 2020-06-12:
I'm pleasantly surprised the `cargo` command completed. I've definitely hit errors with it before on a 1 GiB machine.

scottlamb, 2020-06-12:
I just tried it on my Raspberry Pi 2 (currently running Ubuntu 19.10) and hit the same problem. `yarn build` worked after I ran these commands:

```shell
sudo apt install zram-config
sudo systemctl start zram-config
```

YMMV.

I also noticed that (without `zram-config`) a build completed at revision 88fe6e5. I think the `favicons-webpack-plugin` thing in the next revision (45abeb2) is pretty intensive. I might get rid of it; it's not even producing the right output for me on my NUC setup for some reason. Some of the svg paths aren't scaled properly or something so I can't actually see the full camera shape in the circle.

JasonKleban, 2020-06-13:
With those two commands I was unsuccessful with the same error.  I will try to verify that zram is really enabled and/or try zswap ...

Update: 

I think I was mistaken before too about `cargo test` - rerunning, or rerunning after `cargo clean` it is failing after sitting here for a long time:
```
Compiling moonfire-nvr v0.1.0 (/home/ubuntu/repos/moonfire-nvr)
Building [=====================================================> ] 349/350: moonfire-nvr(bin)
````

Ultimate printout:
```
...
   Compiling moonfire-nvr v0.1.0 (/home/ubuntu/repos/moonfire-nvr)
error: linking with `cc` failed: exit code: 1
  |
  = note: "cc" "-Wl,--as-needed" "-Wl,-z,noexecstack" "-L" "/home/ubuntu/.rustup/toolchains/stable-aarch64-unknown-linux-gnu/lib/rustlib/aarch64-unknown-linux-gnu/lib" "/home/ubuntu/repos/moonfire-nvr/target/debug/deps/moonfire_nvr-d6b13864861fa9ed.11egqo4j2d0bob4p.rcgu.o
...
  = note: collect2: fatal error: ld terminated with signal 9 [Killed]
          compilation terminated.
```

JasonKleban, 2020-06-13:
`sudo apt install zram-config` didn't have the same outcome as the instructions based here: `https://github.com/StuartIanNaylor/zram-config` (for example, there was no /etc/ztab file from apt, among other differences).

With the /etc/ztab **updated away from the defaults** [as shown in the zram-config readme](https://github.com/StuartIanNaylor/zram-config#customize), `yarn build` finally worked on-board (284 seconds).  And `cargo test` (261 seconds) with all tests passing.  `cargo build --release` is taking significantly longer, but appears to be working.

closed 2020-06-13 by JasonKleban