scottlamb/moonfire-nvr #86: Missing /etc/systemd/system/moonfire-nvr.service?

JasonKleban, 2020-06-13:
When trying to follow https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install.md#dedicated-hard-drive-seutp.

> Edit the systemd configuration to do so
> 
> ```
> $ sudo vim /etc/systemd/system/moonfire-nvr.service
> $ sudo systemctl daemon-reload
> ```

Feels like that is missing a step?  First, I can't find the file.  And then it's not obvious to me (yet) what I'd be editing in that file.

```
$ ls /etc/systemd/system/
basic.target.wants                      getty.target.wants           sockets.target.wants
cloud-config.service.d                  graphical.target.wants       sshd.service
cloud-init-local.service.d              iscsi.service                sysinit.target.wants
dbus-fi.w1.wpa_supplicant1.service      mdmonitor.service.wants      syslog.service
dbus-org.freedesktop.Avahi.service      multi-user.target.wants      timers.target.wants
dbus-org.freedesktop.resolve1.service   multipath-tools.service      zram-config.service
dbus-org.freedesktop.timesync1.service  network-online.target.wants
default.target.wants                    paths.target.wants
```

scottlamb, 2020-06-14:
> I can't find [`/etc/systemd/system/moonfire-nvr.service`].

You should create it by following the "Building and installing from source" section, either the "scripted installation" or the "manual". Which did you use?

> And finally, nothing happens with [`sudo -u moonfire-nvr moonfire-nvr config 2>debug-log`]

What's in `debug-log`? Do the `moonfire-nvr` user and binary exist?

> And port 8080 isn't listening.

That's normal until you explicitly run `systemctl start moonfire-nvr` (after creating the service file and going through the `moonfire-nvr config` step). Some day it'll just be a web interface to configure (#35). Until then there's the separate `moonfire-nvr config` step.

JasonKleban, 2020-06-14:
Got it, thanks.  It's my fault for not following the instructions and/or reading carefully enough.  But let me share what I was thinking in case it helps you or others.

I was doing the manual installation but then I felt I didn't understand enough context when I got to [creating-the-user-and-database](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install-manual.md#creating-the-user-and-database) so I switched to the scripted install figuring it would do all the same stuff or skip it if already done or whatever.

Then, duh, I should have checked the debug log file but it didn't occur to me that something even ran yet because I figured there would be some standard stream still printing to the console (`2` is error, right?  having never seen the UI, I didn't know what to expect from the UI, now I see it is the reason for the ncurses dependency).

After successfully starting the UI, I believe it overwrites the file and I didn't look before then to see if it had any errors.  These are the contents now, in case anyone is curious what goes there:

```
I0614 113740.867 main moonfire_db::db] Loading video sample entries
I0614 113740.868 main moonfire_db::db] Loaded 0 video sample entries
I0614 113740.868 main moonfire_db::db] Loading sample file dirs
I0614 113740.869 main moonfire_db::db] Loaded 0 sample file dirs
I0614 113740.869 main moonfire_db::db] Loading cameras
I0614 113740.869 main moonfire_db::db] Loaded 0 cameras
I0614 113740.869 main moonfire_db::db] Loading streams
I0614 113740.870 main moonfire_db::db] Loaded 0 streams
I0614 114912.143 main moonfire_db::db] Flush 1 (why: drop): no recording changes
I0614 114912.150 main moonfire_nvr] Success.
```

closed 2020-06-14 by JasonKleban