scottlamb/moonfire-nvr #95: MPEG4 Support?

AndreVallestero, 2020-09-07:
Alot of older IP cameras using MPEG4 instead of h264. Does moonfire support pass-through and analysis of MPEG4 streams?

scottlamb, 2020-09-08:
MPEG-4 as in the older MPEG-4 Part 2 video codec? Interesting. No, it's not supported today. I haven't seen any cameras that support this format or any that don't support H.264. Example camera like this? I have seen MJPEG (only for lower-resolution streams) but haven't bothered with it since H.264 is superior. I've also seen H.265 on newer cameras but haven't implemented that yet.

Moonfire NVR's video codec-specific code comes down to:

*.  any transformation that have to be done on the frames from ffmpeg before storing them into the sample data files. In the case of H.264, this is converting from Annex B format to the AVC format. I'm not sure off-hand if any transformation is necessary for MPEG-4 Part 2.
*.  creating the `VideoSampleEntry` box to put into the `.mp4` files.
*.  extracting any other relevant metadata from it such as RFC 6383 codec parameter, max bitrate (to reserve the right amount of storage space for in-progress recordings), etc. The `new-schema` branch does a little more of this than the `master` branch right now.

Moonfire NVR doesn't actually do much analysis yet (there's some nascent code on the `new-schema` branch for it) but I plan to rely on ffmpeg's avcodec for decoding which surely already supports this codec as well as a bunch of others, so no problem there.

AndreVallestero, 2020-09-08:
Many older sercomm cameras (commonly used for home surveillance packaged with internet / tv deals, etc) use MPEG-4 Part 2 (mp4v). I just tested today and it manages to pick up the rtsp stream but outputs the following when testing.
```
Stream test failed
video stream has timebase 1/30000; expected 1/90000
```

Adding the camera regardless results in an empty frame in the web client.

scottlamb, 2021-03-10:
I'm marking this at milestone "wishlist" (may never happen). My reasoning is: I don't think any browser supports this video codec, so this would mean transcoding (and thus using quite a bit of CPU). I'm not sure getting a powerful enough NVR machine for that would make sense as opposed to upgrading to a more modern camera.

AndreVallestero, 2021-03-11:
Yeah, that makes sense. It might be worth just closing this issue as "transcoding abnormal source streams" seems like a separate issue and should be looked at in a broader scope than strictly MPEG4 part 2.