scottlamb/moonfire-nvr #106: Startup: E0205 173852.452 main moonfire_nvr] Sys(EACCES)

jlpoolen, 2021-02-06:
After implementing moonfire-nvr on a Raspberry Pi4 against an 8 TB Western Digital Purple with 4 Reolinks cameras feeding moonfire-nvr with 1920x1280 fps feed, I visited my web interface and found that I could only select the month of January, 2021.  Today is 2/5/2021; all of February was grayed.  I went into a shell and found my 8 TB allocation at 100% (with a little bit of wiggle room):

/dev/mapper/vgpurple-cameras 8521158564 8071073464   20571988 100% /mnt/purple

I tried killing moonfire-nvr, but I guess I started it as a system service.  Unfortunately, my wiki is down because my Xen hypervisor crashed so I'm without all my notes of the install.  Any rate, I ended up rebooted Pi and trying to start moonfire-nvr from where I built it.  Here's my console session:

`jlpoole@raspberrypi:/usr/local/src/moonfire-nvr $ target/release/moonfire-nvr run`
`E0205 173852.452 main moonfire_nvr] Sys(EACCES)`
`jlpoole@raspberrypi:/usr/local/src/moonfire-nvr $ `

I'm suspecting a storage threshold has been crossed and now I'm in the doldrums.  I searched this site for instructions, I recall something kind of tricky in order to make room... maybe going into SQLite and deleting some entries??  Whatever.  Any rate, I'm fish out of water (having my Xen - Dokuwiki instance unavilaible to me) and thus log this issue.  I'm sure somebody, perhaps me again, in the future will find themselves in the same situation... a solidly running instance stops and resuming it requires something more than just "start".

I checked the "Troubleshooting" section and did not find anything applicable to the predicament I'm in.

What do I do, or where should I look?

jlpoolen, 2021-02-06:
<slapping myself>  I did not have "sudo".  So let the above document what happens when you forget to run moonfire-nvr not as root.

Enhancement request: check if user is root and advise if not.

scottlamb, 2021-02-06:
I agree that error message is terrible. I'll improve it. Also, one trick you can use when you get an uninformative error message is to set the `RUST_BACKTRACE=1` environment variable.

I don't recommend running as root. It just needs permissions on the db and sample file dirs (and on the files it already created within them if you're fixing an existing install). I run it as a dedicated `moonfire-nvr` user, and the install instructions should help you get that set up.

scottlamb, 2021-02-06:
Oh, that's right: I did improve it, in d83bb1b. If you run a more current version, you should see something like this instead:

```
[slamb@nuc ~]$ sudo -u nobody moonfire-nvr run
E0205 191848.813 main moonfire_nvr] Sys(EACCES)

unable to open db dir /var/lib/moonfire-nvr/db: EACCES: Permission denied
```

Is that more helpful?

jlpoolen, 2021-02-06:
Yes, "Permission denied" would clue me in.  The benefit here is that I return to this with virtually an empty mind, so a fresh look is helpful.  After I get my Xen server going, I'll come back and sync up and rebuild.

It *is* nice to be able to start this and then just walk away from it for months simply enjoying the ability to download video snippets when needed.

scottlamb, 2021-02-06:
> After I get my Xen server going, I'll come back and sync up and rebuild.

You could also try the new pre-built Docker container.

jlpoolen, 2021-02-06:
I wasn't sure if Docker would work on a Raspberry Pi.  But if it will, I am game.

(btw.... I've lost about 2 weeks because I thought I'd be clever and use LVM for my entire disk and Xen environment, that was a terrible mistake. https://serverfault.com/questions/279571/lvm-dangers-and-caveats. I started having my nano session suddenly wiped clean -- coming from a Debian lvm 2.0.3 to Gentoo 1.78.x; I read the afore-referenced article and wow.  Back to basics with fdisk and good old partitions.)

scottlamb, 2021-02-06:
Docker works on my Raspberry Pi 4.