scottlamb/moonfire-nvr #108: FTL for streaming?

IronOxidizer, 2021-02-11:
Are there any plans to use the FTL protocol to potentially reduce latency even further? Maybe something along the lines of [Project Lightspeed](https://github.com/GRVYDEV/Project-Lightspeed)?

scottlamb, 2021-02-11:
Where would FTL fit in? Moonfire NVR talks directly with cameras (which support RTSP and perhaps RTMP, not FTL) and browsers (which support WebSocket and WebRTC, not FTL) today. I don't have any plans to add intermediaries.

Even the basic live view is still under development (#59); the server half is committed but the client half is not. It does seem to more or less work; I think the biggest contributors to latency right now are:

*   encoding delay in the camera. This isn't in my control, unfortunately.
*   buffering in the browser. Switching to WebRTC might help; I also am keeping an eye on [WebCodecs](https://github.com/WICG/web-codecs).

In total the delay with Moonfire NVR's live view on the browser seems pretty similar to using VLC to talk directly to the camera.

IronOxidizer, 2021-02-11:
I was thinking potential future clients other than browsers (desktop app, mobile app) could leverage FTL. Seeing as that's probably scope for now, I'll close this issue.

closed 2021-02-11 by IronOxidizer

scottlamb, 2021-02-12:
Ahh, yeah, that's certainly possible. I've thought about adding RTSP relay support but don't have any immediate plans for it, much less FTL. If you'd like to contribute support, though, I'd be happy to help you learn the codebase and fit it into the design.

IronOxidizer, 2021-02-12:
> buffering in the browser. Switching to WebRTC might help; I also am keeping an eye on WebCodecs.

Speaking of WebRTC, it might be worth investigating a serverless (TURN/STUN) implementation to reduce latency. I have a small demo of this here: https://github.com/IronOxidizer/serverless-webrtc-rust

scottlamb, 2021-02-16:
My current approach is to assume you can set up port forwarding to expose Moonfire NVR to the Internet. Docs [here](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/secure.md). So there's no intermediary causing extra latency.

It'd be nice to use NAT traversal to make installation easier (in general) and possible for folks behind Carrier Grade NAT (not sure how common that is but I know it exists), but no one's complained about it so far, and I have a lot of other improvements I want to make to Moonfire NVR.