scottlamb/moonfire-nvr #109: Cargo "subslice patterns are unstable" Causes Build To Fail

jlpoolen, 2021-02-12:
I upgraded my Raspberry Pi4.  Then "git pull" the work area I had compiled this project in around June 2020.

     pi@raspberrypi:/usr/local/src/moonfire-nvr/server $ uname -a
     Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
     pi@raspberrypi:/usr/local/src/moonfire-nvr/server $ cargo --version
     cargo 1.42.1

The build failed citing:

      error[E0658]: subslice patterns are unstable

Last  error repeats 10 previous ones for different parts of the code:

     error[E0658]: subslice patterns are unstable
        --> /home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/object-0.23.0/src/read/mod.rs:196:28
         |
     196 |             | [0x64, 0x86, ..] => FileKind::Coff,
         |                            ^^
         |
         = note: for more information, see https://github.com/rust-lang/rust/issues/62254

Log (6 month expiration) at: https://pastebin.com/Qy0UeR2D

**Background** 
My SD card was corrupted, but multiple attempts to boot somehow caused it to fix itself... I may have had interfence from my hard disk array and possibly Raspberry Pi was trying to boot from a disk thereon, I ended up turning off all the disks and then tried booting Raspberry Pi where I found initialization message waiting to find the hard disk array; I then turned on the array and the boot completed.  I then upgraded my system "apt upgrade", took 40 minutes.  Found cargo gone, i guess upgrades in Raspberry Land mean previously installation of packages get jettisoned.  Installed current available version of cargo and then attempted to build.

scottlamb, 2021-02-12:
I bump the minimum Rust version from time to time. It's at 1.45 now, so it's unsurprising you can't build with 1.42 anymore. `rustup update` should solve the problem.

closed 2021-02-12 by scottlamb

jlpoolen, 2021-08-19:
For the record:

on a rebuilt RaspberryPi4 as of today, I used the apt-cache to install rustc and it installed 1.41:

    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ rustc -V
    rustc 1.41.1
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ cargo --version
    cargo 1.42.1
    root@pi:/usr/local/src/moonfire-nvr/server $ 

I found this closed Issue, of course, after I ran into compile problems.  So, once again I ended up installing Rust directly from their web site:

    https://sh.rustup.rs -sSf | sh

scottlamb, 2021-08-19:
Yeah, Rust is changing rapidly enough that it's hopeless to use the distribution packages in Debian-based distributions. (Maybe you can get away with it in Arch.)

jlpoolen, 2021-08-19:
Let us not forget dear Gentoo which has 1.53 as the stable high watermark:

[I] dev-lang/rust-bin
     Available versions:  (stable) 1.52.1 1.53.0 ~1.54.0
       {clippy doc prefix rls rustfmt verify-sig ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32" CPU_FLAGS_X86="sse2"}
     Installed versions:  1.53.0(stable)(01:16:04 AM 07/31/2021)(-clippy -doc -prefix -rls -rustfmt -verify-sig ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32" CPU_FLAGS_X86="sse2")
     Homepage:            https://www.rust-lang.org/
     Description:         Systems programming language from Mozilla