scottlamb/moonfire-nvr #114: One of Four Cameras Is Failing

jlpoolen, 2021-03-11:
One of four cameras seems to go into a repeated loop stopping and being unable to accept streams and record same.  See screenshot below of the web UI where garage_west has an abnormal number of small files

Running moonfire-nvr updated 3/10/2021.  Have run previous version, including circa 6/1/2020 where I had several months of operation without this kind of problem.

**Desired result:** equal recording of camera "garage_west" as three other cameras and web interface offering files of similar size as other cameras

moonfire-nvr@raspberrypi:/usr/local/src/moonfire-nvr/server/target/release $ git describe
v0.6.0-40-ge66a88a
moonfire-nvr@raspberrypi:/usr/local/src/moonfire-nvr/server/target/release $ ./moonfire-nvr --version
moonfire-nvr 0.6.1
moonfire-nvr@raspberrypi:/usr/local/src/moonfire-nvr/server/target/release $

Ran with RUST_BACKTRACE=1

log ( 254MB; feel free to select snippets and post herein as I'll remove this large file after 1 week) for approximately 4 hour session this afternoon at:  

https://drive.google.com/file/d/1Ir0SZ3jlsiLO5IiyMELQnX4YkJp_Akxk/view?usp=sharing

Screenshot showing the Web interface and the garage_west files constantly restarting
[image]

As a check, I have the Reolink client software running and capturing events from garage_west and the three other cameras and there has been no noticeable difference for garage_west's camera.  I therefore think something on the receiving side, e.g. moonfire-nvr on my Raspberry Pi4 is causing the problem.

 - OS: Windows 7
 - Browser: Firefox
 - Version [e.g. 22]

scottlamb, 2021-03-11:
Well, looks like the panic in #113 is no longer happening, so that's an improvement at least.

Do you still have any logs from the older, working installation? I'd like to see the lines after `Initialized ffmpeg. Versions:` if so.

It's not happy with `garage_west` or `peck_west` from the beginning. Is there anything different between these cameras and the others? Different model, firmware, etc?

I see several kinds of errors here:

```
W20210310 13:06:38.690 s-garage_west-main moonfire_ffmpeg::rtsp] 0xaf300950: CSeq 8 expected, 0 received.

W20210310 13:17:09.060 s-garage_west-main moonfire_ffmpeg::rtsp] 0xaf300950: RTP H.264 NAL unit type 29
W20210310 13:17:09.060 s-garage_west-main moonfire_ffmpeg::rtsp] 0xaf300950:  is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.

W20210310 14:00:48.748 s-garage_west-main moonfire_nvr::streamer] garage_west-main: sleeping for Duration { secs: 1, nanos: 0 } after error: pts not monotonically increasing; got 26140140 then 26120517

E20210310 14:22:37.305 s-garage_west-main moonfire_ffmpeg::rtsp] 0xac5ac8d0: RTP: PT=60: bad cseq f304 expected=3882

E20210310 14:35:12.457 s-garage_west-main moonfire_ffmpeg::h264] 0xac6598f0: Invalid NAL unit 8, skipping.

E20210310 14:22:37.361 s-garage_west-main moonfire_ffmpeg::h264] 0xac518460: error while decoding MB 22 47, bytestream -9
I20210310 14:22:37.373 s-garage_west-main moonfire_ffmpeg::h264] 0xac518460: concealing 6907 DC, 6907 AC, 6907 MV errors in I frame
```

The `pts not monotonically increasing` is the direct cause of the disconnects, but I wonder if the other errors are due to the same root cause.

I wonder if you can reproduce with the `ffmpeg` commandline utility, without involving Moonfire NVR at all. Knowing that would narrow the problem down anyway. Maybe via a command like this:

```
$ ffmpeg -loglevel debug -fflags nobuffer -rtsp_transport tcp -i 'rtsp://admin:YOUR_PASSWORD_HERE@192.168.1.48:554/h264Preview_01_main' -c copy -f null -
```

That command will just open a stream and discard the received bytes until it gets dropped or you hit `q`. Note you'll have to manually redact your password from the logfile.

If that shows problems, it'd be interesting to see if dropping the `-fflags nobuffer` argument fixes it. Moonfire NVR started setting this option recently, in  75dce88.

We could also try getting a packet capture at the same time as a log; it'd be relatively easy then to see if ffmpeg is right about the `CSeq` being incorrect. But it'd be even more data to deal with and harder to redact the password in so maybe we'll hold onto that idea for later.

scottlamb, 2021-03-11:
Most of these complaints happen to a lesser extent on the other cameras too.

```
$ fgrep 'CSeq' logs.txt | cut -d' ' -f3 | sort | uniq -c
   1016 s-garage_east-main
  18265 s-garage_west-main
    716 s-peck_east-main
    243 s-peck_west-main

$ fgrep 'H.264 NAL unit' logs.txt | cut -d' ' -f3 | sort | uniq -c
      2 s-garage_west-main

$ fgrep 'sleeping for Duration' logs.txt | cut -d' ' -f3 | sort | uniq -c
      1 s-garage_east-main
    164 s-garage_west-main
     10 s-peck_east-main
      3 s-peck_west-main

$ fgrep 'bad cseq' logs.txt | cut -d' ' -f3 | sort | uniq -c
1986452 s-garage_west-main
   2895 s-peck_east-main

$ fgrep 'Invalid NAL' logs.txt | cut -d' ' -f3 | sort | uniq -c
     17 s-garage_west-main

$ fgrep 'error while decoding MB' logs.txt | cut -d' ' -f3 | sort | uniq -c
   1795 s-garage_west-main
     58 s-peck_east-main

$ fgrep 'concealing' logs.txt | cut -d' ' -f3 | sort | uniq -c
   5070 s-garage_west-main
    125 s-peck_east-main
```

scottlamb, 2021-03-11:
Further breakdown of the `sleeping for Duration` lines:

```
$ fgrep 'pts not monotonically increasing' sleeps.txt | cut -d' ' -f3 | sort | uniq -c
      1 s-garage_east-main
    110 s-garage_west-main
      4 s-peck_east-main
      3 s-peck_west-main
$ fgrep 'excessive pts jump' sleeps.txt | cut -d' ' -f3 | sort | uniq -c
      8 s-garage_west-main
      4 s-peck_east-main
$ fgrep 'exceeds maximum' sleeps.txt | cut -d' ' -f3 | sort | uniq -c
     28 s-garage_west-main
      2 s-peck_east-main
$ fgrep 'Connection refused' sleeps.txt | cut -d' ' -f3 | sort | uniq -c
     17 s-garage_west-main
$ fgrep 'Connection timed out' sleeps.txt | cut -d' ' -f3 | sort | uniq -c
      1 s-garage_west-main
```

jlpoolen, 2021-03-11:
I'll see if I have some old logs... I doubt it, though.
Also, I did upgrade the Raspberry Pi system, so I would think that I'm using a later version of ffmpeg.  It looks like an upgrade of ffmpeg may be the culprit here.  Nonetheless, I try to behave on the Raspberry Pi 4 server as if I were a consumer because that is the prospective user you may have for this project (if you are going to continue to suggest Raspberry Pi as a viable platform) and if they run into trouble doing an upgrade on the system, then it's good to flush the issue out with me.    It's going to take some time which I'll try to devote over the weekend.

scottlamb, 2021-03-11:
Good luck!

fwiw, I think a "consumer" would be more likely to follow the recommended path of using the Docker image, which comes with a fixed version of the ffmpeg libraries. But I don't know for sure yet if that would avoid this problem, or if it's a problem with ffmpeg at all. It's probably not the whole story; I have a pi4 with those same versions and that `ffmpeg` command doesn't seem to be having the same problems with my cameras (in the first few minutes at least).

scottlamb, 2021-03-11:
Hmm, in issue #84 you pasted a log including this:

```
I0710 190739.619 main moonfire_ffmpeg] Initialized ffmpeg. Versions:
avutil: running=56.22.100 compiled=56.22.100
avcodec: running=58.35.100 compiled=58.35.100
avformat: running=58.20.100 compiled=58.20.100
```

which is the same as the one from this bug:

```
I20210310 13:04:59.019 main moonfire_ffmpeg] Initialized ffmpeg. Versions:
avutil: running=56.22.100 compiled=56.22.100
avcodec: running=58.35.100 compiled=58.35.100
avformat: running=58.20.100 compiled=58.20.100
```

There could be some subtle difference not captured in the version number but it seems unlikely.

jlpoolen, 2021-03-11:
I found a log file that has:
    I0527 050128.310 main moonfire_nvr::cmds::run] Database is loaded.
    I0527 050128.373 main moonfire_nvr::cmds::run] Directories are opened.
    I0527 050128.373 main moonfire_nvr::cmds::run] Resolved timezone: America/Los_Angeles

    W0527 050128.373 main moonfire_nvr::web] UI directory file "jquery-ui.bundle.js.gz" has unknown extension; skipping
    W0527 050128.373 main moonfire_nvr::web] UI directory file "vendor.bundle.js.gz" has unknown extension; skipping
    W0527 050128.373 main moonfire_nvr::web] UI directory file "jquery.bundle.js.gz" has unknown extension; skipping
    W0527 050128.373 main moonfire_nvr::web] UI directory file "nvr.bundle.js.gz" has unknown extension; skipping
    I0527 050128.374 main moonfire_ffmpeg] Initialized ffmpeg. Versions:
    avutil: running=56.31.100 compiled=56.31.100
    avcodec: running=58.54.100 compiled=58.54.100
    avformat: running=58.29.100 compiled=58.29.100

Moreover, I cannot say for certain that in June or July I was using the same ffmpeg, but here's what is showing my Rasbertty Pi 4 now.  I am not familiar with the Raspberry Pi deployment of binaries to know whether the July 8th date is the date the binary was installed on my machine, or an artificial date create by a some package manager.

    jlpoole@raspberrypi:~ $ which ffmpeg
    /usr/bin/ffmpeg
    jlpoole@raspberrypi:~ $ ls -la /usr/bin/ffmpeg
    -rwxr-xr-x 1 root root 243304 Jul  8  2020 /usr/bin/ffmpeg
    jlpoole@raspberrypi:~ $ffmpeg -version
    ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
    built with gcc 8 (Raspbian 8.3.0-6+rpi1)
    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
    libavutil      56. 22.100 / 56. 22.100
    libavcodec     58. 35.100 / 58. 35.100
    libavformat    58. 20.100 / 58. 20.100
    libavdevice    58.  5.100 / 58.  5.100
    libavfilter     7. 40.101 /  7. 40.101
    libavresample   4.  0.  0 /  4.  0.  0
    libswscale      5.  3.100 /  5.  3.100
    libswresample   3.  3.100 /  3.  3.100
    libpostproc    55.  3.100 / 55.  3.100
    jlpoole@raspberrypi:~ $

I think what I need to do is just isolate the ffmpeg command for camera garage_west in its own session and see if I can recreate the error reports.

jlpoolen, 2021-03-12:
So I tried the following:

### A) 4 consoles on Rapsberry Pi
I connect to the Rapsberry Pi and with screen attempted to create four windows, each being a session using the following formula:

     ffmpeg -loglevel debug -fflags nobuffer -rtsp_transport tcp -i 'rtsp://admin:[redacted]@192.168.1.49:554/h264Preview_01_main' -c copy -f null - 

As can happen with screen, I became lost in its maze after trying to create a four panel window and unsuccessfully trying to give each a title.  (Will still try with tmux another time.)

So I connected to the Raspberry Pi with four ssh consoles via Putty on Windows and ran ffmpeg against each camera.  After about an hour, one then a second camera started to go into an infinite loop state with messages such as:

     frame=442457 fps= 12 q=-1.0 size=N/A time=10:01:36.15 bitrate=N/A speed=   1x

Again, I cannot say if my ffmpeg on the Raspberry Pi 4 was altered when I performed a system upgrade in the last few months.  But I suspected my disparity of results, e.g. June - January vs. recent few weeks, was being caused at the ffmpeg level.

### B) Parallel Console On Different Servers & Different ffmpeg
I then wanted to compare my Gentoo VM version of ffmpeg with the Rapsberry Pi high watermark.   I let the two consoles run overnight and 8 hours later found the Gentoo one in the endless loop "frame=...." above; the Raspberry Pi only had a handful of errors.

**Gentoo ffmpeg**

    ares /home/jlpoole # ffmpeg -version
    ffmpeg version 4.3.2 Copyright (c) 2000-2021 the FFmpeg developers
    built with gcc 9.3.0 (Gentoo 9.3.0-r2 p4)
    configuration: --prefix=/usr --libdir=/usr/lib64 --shlibdir=/usr/lib64 --docdir=/usr/share/doc/ffmpeg-4.3.2/html --mandir=/usr/share/man --enable-shared --cc=x86_64-pc-linux-gnu-gcc --cxx=x86_64-pc-linux-gnu-g++ --ar=x86_64-pc-linux-gnu-ar --nm=x86_64-pc-linux-gnu-nm --ranlib=x86_64-pc-linux-gnu-ranlib --pkg-config=x86_64-pc-linux-gnu-pkg-config --optflags='-march=native -O2 -pipe' --disable-static --enable-avfilter --enable-avresample --disable-stripping --disable-optimizations --disable-libcelt --enable-version3 --disable-indev=alsa --disable-indev=oss --disable-indev=jack --disable-indev=sndio --disable-outdev=alsa --disable-outdev=oss --disable-outdev=sndio --enable-bzlib --disable-runtime-cpudetect --disable-debug --disable-gcrypt --enable-gnutls --disable-gmp --enable-gpl --disable-hardcoded-tables --enable-iconv --disable-libtls --disable-libxml2 --disable-lzma --enable-network --disable-opencl --disable-openssl --enable-postproc --enable-libsmbclient --disable-ffplay --disable-sdl2 --disable-vaapi --disable-vdpau --disable-vulkan --enable-xlib --enable-libxcb --enable-libxcb-shm --enable-libxcb-xfixes --enable-zlib --disable-libcdio --disable-libiec61883 --disable-libdc1394 --disable-libcaca --disable-openal --enable-opengl --disable-libv4l2 --enable-libpulse --disable-libdrm --disable-libjack --disable-libopencore-amrwb --disable-libopencore-amrnb --disable-libcodec2 --enable-libdav1d --disable-libfdk-aac --disable-libopenjpeg --disable-libbluray --disable-libgme --disable-libgsm --disable-libaribb24 --disable-mmal --disable-libmodplug --disable-libopus --disable-libilbc --disable-librtmp --disable-libssh --disable-libspeex --disable-libsrt --enable-librsvg --disable-ffnvcodec --enable-libvorbis --disable-libvpx --disable-libzvbi --disable-appkit --disable-libbs2b --disable-chromaprint --disable-cuda-llvm --disable-libflite --disable-frei0r --disable-libfribidi --disable-fontconfig --disable-ladspa --disable-libass --disable-libtesseract --disable-lv2 --enable-libfreetype --disable-libvidstab --disable-librubberband --disable-libzmq --disable-libzimg --disable-libsoxr --enable-pthreads --disable-libvo-amrwbenc --enable-libmp3lame --disable-libkvazaar --disable-libaom --disable-libopenh264 --disable-librav1e --disable-libsnappy --disable-libtheora --disable-libtwolame --disable-libwebp --enable-libx264 --disable-libx265 --disable-libxvid --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-neon --disable-vfp --disable-vfpv3 --disable-armv8 --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-altivec --disable-vsx --disable-power8 --disable-amd3dnow --disable-amd3dnowext --disable-aesni --disable-avx --disable-avx2 --disable-fma3 --disable-fma4 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-xop --cpu=host --disable-doc --disable-htmlpages --enable-manpages
    libavutil      56. 51.100 / 56. 51.100
    libavcodec     58. 91.100 / 58. 91.100
    libavformat    58. 45.100 / 58. 45.100
    libavdevice    58. 10.100 / 58. 10.100
    libavfilter     7. 85.100 /  7. 85.100
    libavresample   4.  0.  0 /  4.  0.  0
    libswscale      5.  7.100 /  5.  7.100
    libswresample   3.  7.100 /  3.  7.100
    libpostproc    55.  7.100 / 55.  7.100
    ares /home/jlpoole #

**Raspberry Pi4 ffmpeg** 

    jlpoole@raspberrypi:~ $ ffmpeg -version
    ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
    built with gcc 8 (Raspbian 8.3.0-6+rpi1)
    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable- [... cut ...]
My next step is to "unmask" a more recent version of ffmpeg on my Gentoo VM and see what happens with that.  

This issue is moving away from Moonfire-nvr and looking to be an ffmpeg issue.  

Further thoughts: I suppose I should try to save what comes in via ffmeg's rtsp call to learn if the repeated message of "frame=442457 fps= 12 q=-1.0 size..." means nothing and affects what is processed, or whether it is white noise.  I'm going to have to learn more about ffmpeg which is a monstrous tome, nay operating system guised as a program.

scottlamb, 2021-03-12:
> the Raspberry Pi only had a handful of errors.

What were the errors? I don't see them here. I'm particularly interested to see if it says similar errors to as ones you saw when running through Moonfire NVR:

*    `CSeq 8 expected, 0 received.`
*   `RTP H.264 NAL unit type 29 is not implemented. Update your FFmpeg version to the newest one from Git. If the problem still occurs, it means that your file has a feature which has not been implemented.`
*   `RTP: PT=60: bad cseq f304 expected=3882`
*    `Invalid NAL unit 8, skipping.`
*    `error while decoding MB 22 47, bytestream -9`
*   `concealing 6907 DC, 6907 AC, 6907 MV errors in I frame`
*   and last but not least, any complaints about pts or dts. Moonfire generated those log messages about bad data coming in and decided to disconnect. ffmpeg has similar log messages (not exactly the same phrasing) and iirc doesn't treat this as a fatal condition.

I also don't know what you mean by "library warning". The `ffmpeg -version` output from both looks normal to me.

The repeated `frame=442457 fps= 12 q=-1.0 size...` message is normal; that's its way of showing progress. It shouldn't take an hour to start seeing this, though; I'd expect it to happen immediately.

jlpoolen, 2021-03-12:
Here's a copy from the Raspberry Pi console.  See on line 5 "WARNING: library configuration mismatch".

    jlpoole@raspberrypi:~ $ ffmpeg -loglevel debug -fflags nobuffer -rtsp_transport tcp -i 'rtsp://admin:testtest@192.168.1.49:554/h264Preview_01_main' -c copy -f null -
    ffmpeg version 4.1.6-1~deb10u1+rpt1 Copyright (c) 2000-2020 the FFmpeg developers
      built with gcc 8 (Raspbian 8.3.0-6+rpi1)
      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --libdir=/usr/lib/arm-linux-gnueabihf --cpu=arm1176jzf-s --arch=arm
      WARNING: library configuration mismatch
      avutil      configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
      avcodec     configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-omx-rpi --enable-mmal --enable-neon --enable-rpi --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp --cpu=cortex-a7 --arch=armv6t2 --disable-thumb --enable-shared --disable-doc --disable-programs
      avformat    configuration: --prefix=/usr --extra-version='1~deb10u1+rpt1' --toolchain=hardened --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal [... cut ...]
      Input stream #0:0 (video): 447500 packets read (36842824068 bytes);
      Input stream #0:1 (audio): 566060 packets read (140948879 bytes);
      Total: 1013560 packets (36983772947 bytes) demuxed
    Output file #0 (pipe:):
      Output stream #0:0 (video): 447480 packets muxed (36841282003 bytes);
      Output stream #0:1 (audio): 566060 packets muxed (140948879 bytes);
      Total: 1013540 packets (36982230882 bytes) muxed
    0 frames successfully decoded, 0 decoding errors
    Exiting normally, received signal 2.
    jlpoole@raspberrypi:~

scottlamb, 2021-03-12:
Okay, those `Application provided invalid, non monotonically increasing dts` errors look like the same problems that caused Moonfire NVR to close the stream. 

Do they happen on the Gentoo server? Do they happen if you remove the `-fflags nobuffer`?

The library warning seems to be saying that it compiled the binary and the libraries with this difference in options:

```diff
$ diff -u binary avutil
--- binary	2021-03-12 08:57:21.179545106 -0800
+++ avutil	2021-03-12 08:57:21.179363592 -0800
@@ -64,7 +64,10 @@
 --enable-chromaprint
 --enable-frei0r
 --enable-libx264
+--libdir=/usr/lib/arm-linux-gnueabihf/neon/vfp
+--cpu=cortex-a7
+--arch=armv6t2
+--disable-thumb
 --enable-shared
---libdir=/usr/lib/arm-linux-gnueabihf
---cpu=arm1176jzf-s
---arch=arm
+--disable-doc
+--disable-programs
```

I don't think it's causing any problems.

jlpoolen, 2021-03-14:
So the "frame=..." is a bottom line in a console streen that is continually updated to show progress.  I'm assuming this is an ncurses kind of feature to avoid scrolling caused by repetitive output.
[image]

So in my [comment earlier](https://github.com/scottlamb/moonfire-nvr/issues/114#issuecomment-797550466) above, it looks like something happened within the console (or ncurses?) that caused the update line to move down the screen rather than replace itself so the console looks like it has a constantly updated bottom line.  What I thought was an error with ffmpeg looks to have been something caused in my console where the refresh line no longer replaced, but moved down a line in the screen buffer causing the screen to scroll with each line of output.

I now have three consoles running against the same camera, each with a different version of ffmpeg.  I have on my taurus laptop the latest & greatest version of ffmpeg, [release 4.3.2](https://git.ffmpeg.org/gitweb/ffmpeg.git/tag/c872040562ef2988b107a01921dcc20c18b2deef).  I am doing this to see if there are consistent errors in all three session give a certain point of time.  So far, I'm seeing different results.  I'm now thinking network issues may be in play.  The garage cameras come from the garage through a Ubiquiti wireless link, the Peck comes via a long Cat6 running to the house next door, and the laptop has my current house's wireless.  So, a variety of network connections.

I eventually found all screens populated with red error messages and then hung.  No more updates.  I entered Control-C and the ffmpeg program terminated and I was returned to a standard prompt.  This proves that my ssh session was still alive and the hang up occurring on ffmpeg's side.

I'm starting to think that moonfire-nvr is faced with handling problems from ffmpeg and being able to recover gracefully when ffmpeg hangs as it appears to have done in several versions on several machines.  I'm going to do further testing just using various ffmpegs rtsp sessions, my approach has been haphazard and not very methodical and at this point we need to have some methodically produced results.

scottlamb, 2021-03-15:
The `frame=` line is continuously updated just by sending a carriage return, sending the cursor back to the start of the line. If more lines are printed in the meantime(like that `CSeq` log line), the display gets a little off as in your screenshot.

There's another option Moonfire NVR sets when it opens streams with ffmpeg, `stimeout 10000000` (socket timeout). If a RTSP connection hits that timeout (in microseconds, so 10,000,000 µs = 10 seconds), the connection's dropped, and then Moonfire NVR retries after a second. You can match that with this variation of the command:

```
ffmpeg -report -loglevel debug -fflags nobuffer -stimeout 10000000 -rtsp_transport tcp -allowed_media_types video -i 'rtsp://...' -c copy -f null -
```

If your network connection is unreliable, that might just be the best that can be done, with ffmpeg or any other RTSP client. Maybe what changed since you had a working setup on the Pi wasn't ffmpeg or Moonfire but the wireless network conditions.

scottlamb, 2021-04-30:
I've been playing a bit with a Reolink camera (a RLC-410, hardware version IPC_3816M, firmware v2.0.0.1441_19032101). I think it's actually sending interleaved RTP data over a RTSP connection which must be intended for a *different* RTSP connection:

*   in one connection that set up only one track and saw in the `PLAY` response `RTP-Info: ...ssrc=c517011f...`, it started receiving interleaved RTP packets on channel 0 with `SSRC=0xc517011f` (as expected) and `SSRC=0x5bF73F70` (wtf?). They both have packets coming in with independently incrementing sequence numbers: i.e, one was at 25580, 25581, ... while the other one was at 40477, 40478, ...
*   in one case, I got a packet capture where it sent interleaved RTP data over a RTSP connection before `PLAY`, which is illegal. Even worse, it was before authentication, so it was leaking private data over an unauthenticated session.

I imagine these problems are worse if you have multiple NVRs running, record both the `main` and `sub` streams, and/or have frequent connection drops for other reasons (eg poor wifi).

Once I move to a non-ffmpeg RTSP implementation, I might be able to work around this problem (by discarding data which doesn't have the expected `ssrc`) but wow, that's a bug that just shouldn't happen. I wonder if newer models are better (this one's apparently end-of-lifed) and if there's any way to get a bug report to an engineer who would actually fix it.

I also get warnings in the logs about long times to get the next frame if there's a dramatic change to the image. I have the thing in a cabinet with a door right now; if I open or close the door, I can see it in my logs every time. That really shouldn't happen either.

Finally I see cases where the RTP timestamp goes slightly backwards; Moonfire NVR drops the connection in this case. (iirc this also happens with my Hikvision and/or Dahua cameras sometimes.) This seems to happen when they do a time step with SNTP—it must be calculating offsets with `CLOCK_REALTIME` rather than `CLOCK_MONOTONIC` or some such. If the system clock goes backwards, so do the RTP timestamps. Notably, I currently have the Reolink camera to do a SNTP step every 60 minutes, and the timestamp problems happen almost exactly every two hours. I guess it goes forward half the time and back half the time. I also see a huge mismatch when I first set the clock, so I'm pretty sure this is the problem. I could probably work around this by say assuming an increment of the frame rate if the timestamp delta is unreasonable.

jlpoolen, 2021-08-14:
I think I've found something that might bear on the problems I encountered.  One of the facts that were different after I started having problems was that I had upgraded my RaspberryPi4 system.  This bug gets into the nitty gritty of kernels and sleep modes for USB attached hard drives, and some of the people posting noted their problem went away when the downgraded to a previous kernel.  I just wanted to add this URL because it may have bearing on what I experienced:
https://github.com/raspberrypi/linux/issues/3981
No need for comments at this time.  I'm also grappling with the issue of whether I should replace my SD HC to remove any question about media problems contributing to some dropped videos.  I've not been keeping diligent records, but what I am noticing is that my Windows Reolink client is capturing events, triggered by motion settings, at times when moonfire-nvr is not capturing.  I basically can use the Reolink software as a control:  if it saves files and I cannot find them through moonfire-nvr, then I know something has happened on the RapsberryPi end.  I believe my problems are hardware and/or system related and not related to moonfire-nvr.

I am kind of regretting having gone with RaspberryPi since it depends on an SD HC which has an admitted limited life.  Even Scott's suggestions tries to limit the cycles performed against the diskette.  I noticed some of the of the systems Scott recommended are out-of-stock at the linked vendor.  I'm thinking for 24/7 video, something not operating on an SD HC is preferable.  Just too may write cycles.    And the kernel issue in the bug of the URL herein is troubling.   I've also complicated things by using LVM.  I used to use LVM for my Xen hypervisor and then I discovered some shortcoming and well as LVM developer malaise that made me decide to go back to the tried-and-true partition approach.  

What tangled webs we weave.  But having video surveillance feeds coming to a generic browser is so worth it, I just wish I had more time.

scottlamb, 2021-08-14:
> I am kind of regretting having gone with RaspberryPi since it depends on an SD HC which has an admitted limited life.

From a previous email, I recall you have a 5-bay USB SATA enclosure. Have you considered buying a SATA SSD and sticking in there? The Pi4 supports USB booting, and a real SATA SSD would be much more reliable.

jlpoolen, 2021-08-14:
Good suggestion, but I'm shy about SSDs.  Any reason not to use a [Western Digital Blue](https://www.amazon.com/Drives-320GB-Cache-Notebook-WD3200LPVX/dp/B00C9TECUO/ref=sr_1_7?dchild=1&keywords=sata%2Bwestern%2Bdigital%2Bred&qid=1628981405&sr=8-7&th=1), at $20, it seems like a no-brainer?  I don't need the performance an SSD offers; the stability is more important to me.  I have two empty slots in my USB array.
Thank you.

scottlamb, 2021-08-14:
I don't think you need to feel shy about SSDs; they're in a different reliability class than SD cards, and they actually report SMART statistics so you can see how much wear they have on them. I'd expect them to be more reliable as a root volume than a refurbished ("renewed") HDD like that one.

But to answer your question: putting the root drive and SQLite database on a HDD is probably fine. Database accesses will be noticeably slower but probably bad enough so that Moonfire NVR will actually break.

scottlamb, 2021-08-15:
> Edit: changed the URL to points to 3.5" form, original link was to notebook which is a form factor I'm not sure is supported by my array.

I still see the 2.5" link. Another thing to check is if the 3.5" drive (or whatever you select) is using SMR; I try to avoid those drives.

Usually arrays have trays that let you put 2.5" drives in 3.5" slots. SSDs all come in 2.5" form so you'd need this to take my suggestion of putting in a SSD.

jlpoolen, 2021-08-18:
I did more reading about SSD and decided Scott's suggestion was well taken.  So I ordered the following:

-  [PNY CS900 120GB 3D NAND 2.5" SATA III Internal Solid State Drive (SSD) - (SSD7CS900-120-RB) ](https://www.amazon.com/dp/B0722XPTL6?psc=1&ref=ppx_yo2_dt_b_product_details) for $24.99
- [Sabrent 2.5” SSD & SATA Hard Drive to Desktop 3.5” SATA Bay Converter Mounting Kit (BK-PCBS)
](https://www.amazon.com/dp/B00UN550AC?psc=1&ref=ppx_yo2_dt_b_product_details) for $9.99
Note: have to be mindful about installing the assembled PNY and Sabrent carriage so that when pushed into the disk array, the inside mechanical lever is engaged and the pins align properly and seat.  You can't just shove the carriage in like you can with a regular hard disk.

Installed new Raspberry Pi minimal operating system on PNY.
Reconfigured my Pi to boot from USB first (now SD HC is removed).   I will be installing (have to buy a CR 1222 battery locally):

- [Adafruit PiRTC - Precise DS3231 Real Time Clock for Raspberry
Pi[ID:4282]](https://www.adafruit.com/product/4282?gclid=CjwKCAjw3_KIBhA2EiwAaAAlioXe3ingwFJL_eMfbmwnI1mZWb0hNm_FCEbWHkyX5Iu0K91q2giPoxoCXAEQAvD_BwE) $14.95 + $5 US Mail shipping.  Don't order CR 1222 battery as it will have to be sent by UPS which adds $10+ more for shipping.

The goal here is to isolate potential failures to more reliable systems than an SD HC card that has been in use for over 1 year.  Yesterday, on the SD HC-based system, something happened and I could not even reboot from a console, the system locked up.  /var/log/messages had some ext4 related errors which causes me to think there were disk problems.

    Aug 17 13:07:59 raspberrypi kernel: [55663.494808] Exception stack(0xc923dfa8 to 0xc923dff0)
    Aug 17 13:07:59 raspberrypi kernel: [55663.494828] dfa0:                   00000020 ad2f50b0 00000020 00000002 00000000 00000000
    Aug 17 13:07:59 raspberrypi kernel: [55663.494847] dfc0: 00000020 ad2f50b0 00000000 00000076 ad2f4598 ad2f4574 00000000 ad2f4568
    Aug 17 13:07:59 raspberrypi kernel: [55663.494865] dfe0: 00000076 ad2f43a0 b59e1805 b59e34d6
    Aug 17 13:07:59 raspberrypi kernel: [55663.494931] task:s-GarageEast-ma state:D stack:    0 pid: 1556 ppid:  1500 flags:0x00000000
    Aug 17 13:07:59 raspberrypi kernel: [55663.494962] Backtrace:
    Aug 17 13:07:59 raspberrypi kernel: [55663.494996] [<c0b8c820>] (__schedule) from [<c0b8d1f8>] (schedule+0x68/0xe4)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495016]  r10:00000001 r9:00000000 r8:00000000 r7:005e4859 r6:c33cc050 r5:c2ce3e00
    Aug 17 13:07:59 raspberrypi kernel: [55663.495032]  r4:ffffe000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495053] [<c0b8d190>] (schedule) from [<c057eb60>] (wait_transaction_locked+0xc0/0xf8)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495070]  r5:c1205048 r4:c33cc000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495092] [<c057eaa0>] (wait_transaction_locked) from [<c057f04c>] (start_this_handle+0x4b4/0x970)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495110]  r7:c3a08360 r6:c0e51834 r5:c33cc028 r4:c33cc000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495133] [<c057eb98>] (start_this_handle) from [<c057f610>] (jbd2__journal_start+0x108/0x1fc)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495153]  r10:00000001 r9:c13fb098 r8:00000000 r7:00000008 r6:00000002 r5:c33cc000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495168]  r4:c3a08360
    Aug 17 13:07:59 raspberrypi kernel: [55663.495194] [<c057f508>] (jbd2__journal_start) from [<c050d568>] (__ext4_journal_start_sb+0x84/0x100)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495214]  r10:c0c15df4 r9:c0529bd0 r8:c3ae24b0 r7:00000bb6 r6:00000002 r5:00000001
    Aug 17 13:07:59 raspberrypi kernel: [55663.495229]  r4:c5958400
    Aug 17 13:07:59 raspberrypi kernel: [55663.495255] [<c050d4e4>] (__ext4_journal_start_sb) from [<c0532614>] (ext4_da_write_begin+0x1cc/0x498)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495273]  r7:00000000 r6:01f40000 r5:c3ae23a0 r4:d99538f0
    Aug 17 13:07:59 raspberrypi kernel: [55663.495297] [<c0532448>] (ext4_da_write_begin) from [<c03af0c8>] (generic_perform_write+0x108/0x1dc)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495317]  r10:c0532448 r9:c0c15c24 r8:00000dcd r7:00000233 r6:c3ae24b0 r5:c90adee0
    Aug 17 13:07:59 raspberrypi kernel: [55663.495333]  r4:00000233
    Aug 17 13:07:59 raspberrypi kernel: [55663.495355] [<c03aefc0>] (generic_perform_write) from [<c051b1f8>] (ext4_buffered_write_iter+0xb4/0x184)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495375]  r10:c90adee0 r9:c1205048 r8:c90adee0 r7:c3ae2428 r6:c3ae23a0 r5:c2ce3e00
    Aug 17 13:07:59 raspberrypi kernel: [55663.495390]  r4:c90adef8
    Aug 17 13:07:59 raspberrypi kernel: [55663.495412] [<c051b144>] (ext4_buffered_write_iter) from [<c051b7a0>] (ext4_file_write_iter+0x120/0x988)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495431]  r8:c90adf58 r7:00000000 r6:c90adef8 r5:c3ae23a0 r4:00000000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495453] [<c051b680>] (ext4_file_write_iter) from [<c043dd64>] (vfs_write+0x234/0x464)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495473]  r10:00000000 r9:00000000 r8:c90adf58 r7:00000000 r6:c1205048 r5:00033ec2
    Aug 17 13:07:59 raspberrypi kernel: [55663.495488]  r4:cc96e9c0
    Aug 17 13:07:59 raspberrypi kernel: [55663.495509] [<c043db30>] (vfs_write) from [<c043e0f8>] (ksys_write+0x68/0xe0)
[... cut ...]
    Aug 17 13:07:59 raspberrypi kernel: [55663.495636] Exception stack(0xc90adfa8 to 0xc90adff0)
    Aug 17 13:07:59 raspberrypi kernel: [55663.495655] dfa0:                   00033ec2 aafaade8 0000002a aafaade8 00033ec2 00000000
    Aug 17 13:07:59 raspberrypi kernel: [55663.495675] dfc0: 00033ec2 aafaade8 accf20b0 00000004 00ba77d0 00033ec2 accf0bd8 aafaade8
    Aug 17 13:07:59 raspberrypi kernel: [55663.495693] dfe0: 00000004 accefa50 b59e13ad b59e34d6
    Aug 17 13:07:59 raspberrypi kernel: [55663.495757] task:s-PeckAlleyWest state:D stack:    0 pid: 1557 ppid:  1500 flags:0x00000000

scottlamb, 2021-08-31:
As this is now understood as an RTSP-level interop problem with these buggy cameras, let's continue discussion at scottlamb/retina#17 as necessary. Current best solution is to use the new --rtsp-transport=udp argument.

closed 2021-08-31 by scottlamb