scottlamb/moonfire-nvr #119: live view not working if no substream is configured

IronOxidizer, 2021-04-09:
Live view does not work for cameras with no sub stream configured. The result is a black screen and the following error on FireFox

```
Firefox can’t establish a connection to the server at ws://192.168.1.64:8080/api/cameras/7b2c7111-09c8-42a0-bdcd-ac1439133371/sub/live.m4s. LiveCamera.tsx:85:14
door: aborting due to ws close: 1006 
```

It's caused by this line:

https://github.com/scottlamb/moonfire-nvr/blob/8465b49cfab5561886c9ccea848cb38ef6907ba1/server/src/web.rs#L420

Potential solutions could be:

- allow the selection of main or sub stream for live view in the web UI
- fallback to main stream if sub stream is not available
- disable the selection of cameras with no sub streams and add notice during config that live view is only enabled for sub streams

I'm not sure if this is expected behavior, but it should at least be documented.

scottlamb, 2021-04-09:
Good point. I added a note to the troubleshooting doc for the moment. Long-term, I want it to support both streams; there's a bullet point about this at #115.

scottlamb, 2021-04-09:
I'm calling this a JS bug in that the Javascript has all the information to know in advance that the `sub` stream doesn't exist and the `main` stream does.

It'd be nice if the server reported an error in a way the client Javascript could actually see—maybe accepting the WebSocket, sending a text message with an error, then closing it—rather than a HTTP status that the browser [deliberately withholds](https://html.spec.whatwg.org/multipage/web-sockets.html#closeWebSocket) from the Javascript. But given that the error is avoidable (ignoring silly edge cases like the server being reconfigured to take away the sub stream between the loading `/api/` and loading `/api/cameras/.../sub/live.m4s`), I think that's a secondary concern.

khaidoan25, 2021-04-27:
> Good point. I added a note to the troubleshooting doc for the moment. Long-term, I want it to support both streams; there's a bullet point about this at #115.

Here I have an camera ip rtsp://username:pasword@<ip>. Have you supported to get all stream from rtsp camera ? Currently, the system has detected main-stream, but the sub-stream is detected dynamically (in the picture below). So, how can I configure to get sub-stream for live view.

[image]

Thanks in adavance.