scottlamb/moonfire-nvr #124: Connection timeout

khaidoan25, 2021-04-22:
I have tried to connect nvr with my camera, but it kept returning "Connection timeout". Below is my configurations, please show me if there is anything wrong.

My camera IP:
[image]

NVR configuration:
[image]

Thanks in advance.

closed 2021-04-27 by khaidoan25

scottlamb, 2021-04-27:
Did you get this figured out? I see the issue is closed.

I also saw got an email notification that you replied to a comment on #119, but I don't actually see it when I go to that issue, so I'm a little confused. I'll reply here:

> Here I have an camera ip rtsp://username:pasword@. Have you supported to get all stream from rtsp camera ? Currently, the system has detected main-stream, but the sub-stream is detected dynamically (in the picture below). So, how can I configure to get sub-stream for live view.

I just added a new [wiki page](https://github.com/scottlamb/moonfire-nvr/wiki/Configuring-cameras) that might help you find the right RTSP URL.

khaidoan25, 2021-04-28:
I have figured out how to configure the main-stream and sub-stream, that why I removed my comment.

Anyway, thank you for your response.