scottlamb/moonfire-nvr #128: Financial Support Option?

JasonKleban, 2021-05-18:
Is there a way to support the major contributors to this project financially?  I need this product (because the commercial options are so terrible) but I don't have the direct skills to help complete it.

scottlamb, 2021-06-04:
Thank you for the kind offer!

Sorry for the delay getting back to you—I've been working on RTSP support in other repositories (https://github.com/scottlamb/moonfire-playground then https://github.com/scottlamb/retina) and haven't been doing a good job of checking in here.

I'm the only major contributor right now. I've accepted one donation from another kind user but haven't been soliciting donations. The thing is that my work here can be pretty sporadic for several reasons:
*   time commitments. I don't realistically expect I could get enough donations to replace a full-time Silicon Valley job. I'm taking a sabbatical but will expect to return to full-time paid work before too long. Family care (I have two small kids), vacations, pandemic surprises, etc.
*   burnout. I'm taking that sabbatical because after a tough pandemic year I want to a break from deadlines and such and just a chance to recapture my love of writing software.
*   pursuing quality over speed. At work I'd listen to a (real or imagined/internalized) project manager telling me to prioritize user-visible features. Here I've mostly been ignoring that and instead building quality foundational pieces like a better RTSP library, even when it takes a lot longer than I initially expected (see recent comment on #37).

I'd gratefully accept donations as a thank you for what's already built. I don't want to accept donations if they come with an expectation that I'll deliver feature X by date Y or put in X hours/week or the like. If I set up GitHub Sponsors or the like, would I be setting expectations like that?

JasonKleban, 2021-08-11:
I assume you'd get to explain this in the setup.