scottlamb/moonfire-nvr #132: OS error 32 on attempt to Live View

Bobberty, 2021-07-21:
Running on a RPI4/4GiB with a 1TB USB Hard drive.
The cameras are on a vlan IPv4 camera network.  The public face is IPv6.
This system is not in a Docker nor is it proxied.  Compiled from source on ARM.

Moonfire-nvr works great until I attempt a live view from any camera.  I get the Infamous ws close 1006.

**Log File:**
`Jul 21 09:26:15 localhost moonfire-nvr[1383]: tokio-runtime-worker moonfire_nvr::web] Dropping WebSocket after error: IO error: Broken pipe (os error 32)

Jul 21 09:26:32 localhost moonfire-nvr[1383]: tokio-runtime-worker moonfire_nvr::web] Dropping WebSocket after error: IO error: Broken pipe (os error 32)

Jul 21 09:26:42 localhost moonfire-nvr[1383]: tokio-runtime-worker moonfire_nvr::web] Dropping WebSocket after error: IO error: Broken pipe (os error 32)

Jul 21 09:26:44 localhost moonfire-nvr[1383]: tokio-runtime-worker moonfire_nvr::web] Dropping WebSocket after error: IO error: Broken pipe (os error 32)

Jul 21 09:26:58 localhost moonfire-nvr[1383]: tokio-runtime-worker moonfire_nvr::web] Dropping WebSocket after error: IO error: Broken pipe (os error 32)
` 

**Firefox Request/Response Headers:**
HTTP/1.1 101 Switching Protocols
connection: Upgrade
upgrade: websocket
sec-websocket-accept: zQTauLoJPZXNH3DaPGYZbeNQV+o=
cache-control: private, no-cache
date: Wed, 21 Jul 2021 14:11:37 GMTGET /api/cameras/b2be3320-9bb6-4f24-ab31-6453db774c8a/sub/live.m4s HTTP/1.1

Host: [2001:db8::1]:8080
User-Agent: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:90.0) Gecko/20100101 Firefox/90.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
Sec-WebSocket-Version: 13
Origin: http://[2001:db8::1]:8080
Sec-WebSocket-Extensions: permessage-deflate
Sec-WebSocket-Key: grF6gcnNu9nLigFmxdmdXw==
Connection: keep-alive, Upgrade
Pragma: no-cache
Cache-Control: no-cache
Upgrade: websocket

**Other Notes:**
Looking at packet dumps, it appears that the js client is attempting to do a websocket ping to the server that never gets a response.

Bobberty, 2021-07-21:
Discovered that the sub must be set to record.  Would be a good idea to add this to documentation.

closed 2021-07-21 by Bobberty

scottlamb, 2021-07-21:
> Discovered that the sub must be set to record. 

Ahh, yes. This is #119, on my todo list to improve.

> Would be a good idea to add this to documentation.

It's in the troubleshooting guide [here](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#live-stream-always-fails-with-ws-close-1006); did you have somewhere else in mind?

Bobberty, 2021-07-21:
I think I read that line a few times.  Only fully digested after I set the SUB to record.
This project looks to scratch an itch I've had for years.  A simple NVR without crazy bells and whistles.
So, I'm a bit new to the workflow and project docs.
Perhaps a straight forward statement about recording.

Now that the primary itch is being scratched, I would like to do a test deployment that requires binding to IPv4 and IPv6.
I've noticed that Hyper takes a single address.  So, i'm looking at HAProxy as a possible solution.

scottlamb, 2021-07-21:
A proxy in front is also a good idea because there's no built-in TLS support yet (#27). But it's possible to have more than one bind address with hyper via more than one `hyper::Server` (they're cheap, iiuc). That's been on my mental wishlist anyway, so I opened #133.