scottlamb/moonfire-nvr #136: Stuck in a crash loop

cholcombe973, 2021-08-03:
**Describe the bug**
Pulling from the latest docker image it looks like i'm stuck in a crash loop. I did the normal setup and test and the test to the rstp url works from both the UI configurator and VLC. However when I start the docker container it seems to crash as soon as it tries to access the url. 

**To Reproduce**
Steps to reproduce the behavior:
1. I'm following the basic setup instructions. 
2. I changed the docker options to:
```
common_docker_run_args=(
        --mount=type=bind,source=/var/lib/moonfire-nvr,destination=/var/lib/moonfire-nvr
        --mount=type=bind,source=/mnt/raid/nvr,destination=/mnt/raid/nvr
        --user="$(id -u moonfire-nvr):$(id -g moonfire-nvr)"
        --security-opt=seccomp:unconfined
        --log-driver=journald
        --log-opt="tag=moonfire-nvr"
        --env=RUST_BACKTRACE=1
        --env=TZ=":${tz}"
)
```
and gave the moonfire-nvr user ownership of that directory.
3. I ran the configuration UI ncurses setup 
4. Started the docker container with `./nvr.sh run` and it crashes in a loop.

**Expected behavior**
I'm expecting it to connect to the camera. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Server (please complete the following information):**
  - If using Docker: `docker ps` + `docker images`
```
chris@Workstation:~$ docker ps
CONTAINER ID        IMAGE                                  COMMAND                  CREATED             STATUS                            PORTS               NAMES
4f93a54edd59        scottlamb/moonfire-nvr:latest          "/usr/local/bin/moon…"   3 minutes ago       Restarting (101) 15 seconds ago                       moonfire-nvr

chris@Workstation:~$ docker images
REPOSITORY                             TAG                 IMAGE ID            CREATED             SIZE
scottlamb/moonfire-nvr                 latest              bb6a1bc80ab1        4 weeks ago         735MB
```
  - If building from git: `git describe --dirty` + `moonfire-nvr --version`
  - Attach a [log file](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#viewing-moonfire-nvrs-logs). Run with the `RUST_BACKTRACE=1` environment variable set if possible.
```
I20210802 17:13:02.907 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.31.1
I20210802 17:13:02.910 main moonfire_db::db] Loading video sample entries
I20210802 17:13:02.910 main moonfire_db::db] Loaded 0 video sample entries
I20210802 17:13:02.910 main moonfire_db::db] Loading sample file dirs
I20210802 17:13:02.910 main moonfire_db::db] Loaded 2 sample file dirs
I20210802 17:13:02.910 main moonfire_db::db] Loading cameras
I20210802 17:13:02.910 main moonfire_db::db] Loaded 1 cameras
I20210802 17:13:02.910 main moonfire_db::db] Loading streams
I20210802 17:13:02.910 main moonfire_db::db] Loaded 2 streams
I20210802 17:13:02.911 main moonfire_db::db] Loading recordings for camera lorex_1 stream Main
I20210802 17:13:02.911 main moonfire_db::db] Loaded 0 recordings for camera lorex_1 stream Main
I20210802 17:13:02.911 main moonfire_db::db] Loading recordings for camera lorex_1 stream Sub
I20210802 17:13:02.911 main moonfire_db::db] Loaded 0 recordings for camera lorex_1 stream Sub
I20210802 17:13:02.911 main moonfire_nvr::cmds::run] Database is loaded.
I20210802 17:13:03.268 main moonfire_nvr::cmds::run] Directories are opened.
I20210802 17:13:03.268 main moonfire_nvr::cmds::run] Resolved timezone: America/Los_Angeles
I20210802 17:13:03.268 main moonfire_nvr::cmds::run] Starting syncer for path /mnt/raid/nvr
I20210802 17:13:03.270 main moonfire_db::db] Flush 1 (why: synchronous deletion): no recording changes
I20210802 17:13:03.270 main moonfire_nvr::cmds::run] Starting streamer for lorex_1-main
I20210802 17:13:03.270 main moonfire_nvr::cmds::run] Starting streamer for lorex_1-sub
I20210802 17:13:03.270 s-lorex_1-main moonfire_nvr::streamer] lorex_1-main: Opening input: rtsp://192.168.1.115:554/cam/realmonitor?channel=1&subtype=0
I20210802 17:13:03.270 s-lorex_1-sub moonfire_nvr::streamer] lorex_1-sub: Opening input: rtsp://192.168.1.115:554/cam/realmonitor?channel=1&subtype=1
E20210802 17:13:03.331 main moonfire_nvr] panic at '/var/lib/moonfire-nvr/.cargo/registry/src/github.com-1ecc6299db9ec823/hyper-0.14.7/src/server/server.rs:67:17'
Backtrace:
   0: failure::backtrace::internal::InternalBacktrace::new
             at .cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/internal.rs:46:44
   1: failure::backtrace::Backtrace::new
[... 65 lines cut ...]
      std::panic::catch_unwind
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/panic.rs:431:14
      std::rt::lang_start_internal
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/rt.rs:34:21
  20: main
  21: __libc_start_main
  22: _start
```
**Camera (please complete the following information):**
  - Camera manufacturer and model: Lorex LNE8964AB
  - Firmware version: Unknown
```
Device TypeLNE8964AB
System Version00022
Build Date2018-11-05
WEB VersionV3.2.1.582554
ONVIF Version16.12(V2.4.3.574996)
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - Browser: firefox
 - Version: 90.0.2

**Additional context**

scottlamb, 2021-08-03:
Thanks for the clear bug report with stack trace. From this line:

```
   8: hyper::server::server::Server<hyper::server::tcp::AddrIncoming,()>::bind::{{closure}}
```

I believe the problem is that it can't bind the HTTP server because the address is unavailable. It defaults to `0.0.0.0:8080`, and if you run eg `sudo netstat --listening --tcp --program` you'll find that some other program has claimed port 8080. You can stop the other thing or tell Moonfire NVR to use a different port via the `--http-addr` argument.

Moonfire NVR is giving a terrible error message. First of all, hyper is panicking with an error message:

```
panic!("error binding to {}: {}", addr, e);
```

and we appear to be swallowing that text. But anyway panics are supposed to be just for bugs, so we should be using `try_bind` instead of `bind`.

I'm marking this as a usability bug.

cholcombe973, 2021-08-03:
Thanks for digging into this! Yes that makes sense. I have a few other docker containers running on here. I'll remap the port. I was planning on digging into the code but you beat me to it 🙂. I'm looking forward to getting this working. This is a really useful project!

commit c55032df (2021-08-03): don't panic on bind failure

Fixes #136

Before:

```
E20210803 09:00:31.161 main moonfire_nvr] panic at '/Users/slamb/.cargo/registry/src/github.com-1ecc6299db9ec823/hyper-0.14.10/src/server/server.rs:68:17': error binding to 0.0.0.0:80: error creating server listener: Address already in use (os error 48)

(set environment variable RUST_BACKTRACE=1 to see backtraces)
...potentially unrelated log msgs from other threads before exiting...
```

After:

```
E20210803 09:06:02.633 main moonfire_nvr] Exiting due to error: unable to bind --http-addr=0.0.0.0:80
caused by: error creating server listen [... cut]
closed 2021-08-03 by scottlamb