scottlamb/moonfire-nvr #141: Docker Install on Raspberry Pi fails

jlpoolen, 2021-08-11:
I just rebuilt my Raspberry Pi.  I decided to try docker which I have never used on a Linux machine (I tried docker through a web interface years ago when they just started).

Here is the log of my session:  https://pastebin.com/KY2V9ySa

Suggestion:  add this line to docker instructions just before the docker command and after the git clone command.

    cd moonfire-nvr

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux

Used instructions at https://italypaleale.medium.com/yes-you-can-run-docker-on-raspbian-50722bf106dc to install docker.

Failed line at end of attempt:

     error: failed to solve: executor failed running [/bin/sh -c /dev-common.bash]: exit code: 100

jlpoolen, 2021-08-11:
I reviewed my docker install and learned I had not activated docker, so I did.  I then attempted the same docker command for Moonfire and it looks like the same result.  Log file (6 months) at:
https://pastebin.com/yySH8TGk

jlpoolen, 2021-08-11:
It seemed to me that at lines 27 & 28 in my 2nd [pastebin](https://pastebin.com/yySH8TGk), the failure concerned copying to a root level directory.  So I tested creating a directory tree at root using the "pi" account and was successful.  

    pi@raspberrypi:/usr/local/src/moonfire-nvr $ sudo mkdir -p /jlpFromMoonfireTest/test1/subtest1
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ ls -la /jlpFromMoonfireTest/test1/
    total 12
    drwxr-xr-x 3 root root 4096 Aug 11 07:21 .
    drwxr-xr-x 3 root root 4096 Aug 11 07:21 ..
    drwxr-xr-x 2 root root 4096 Aug 11 07:21 subtest1
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ ls -la /
    total 72
    drwxr-xr-x  19 root root  4096 Aug 11 07:21 .
    drwxr-xr-x  19 root root  4096 Aug 11 07:21 ..
    lrwxrwxrwx   1 root root     7 May  7 07:39 bin -> usr/bin
    drwxr-xr-x   3 root root  3584 Dec 31  1969 boot
    drwxr-xr-x  18 root root  4080 Aug 10 19:49 dev
    drwxr-xr-x 123 root root  4096 Aug 10 20:41 etc
    drwxr-xr-x   4 root root  4096 Aug 10 20:05 home
    drwxr-xr-x   3 root root  4096 Aug 11 07:21 jlpFromMoonfireTest
    lrwxrwxrwx   1 root root     7 May  7 07:39 lib -> usr/lib
    drwx------   2 root root 16384 May  7 08:21 lost+found
    drwxr-xr-x   3 root root  4096 May  7 08:24 media
    drwxr-xr-x   2 root root  4096 May  7 07:39 mnt
    drwxr-xr-x   7 root root  4096 Aug 10 20:41 opt
    dr-xr-xr-x 224 root root     0 Dec 31  1969 proc
    drwx------   6 root root  4096 Aug 10 20:43 root
    drwxr-xr-x  32 root root   980 Aug 11 06:25 run
    lrwxrwxrwx   1 root root     8 May  7 07:39 sbin -> usr/sbin
    drwxr-xr-x   2 root root  4096 May  7 07:39 srv
    dr-xr-xr-x  12 root root     0 Dec 31  1969 sys
    drwxrwxrwt  14 root root  4096 Aug 11 06:25 tmp
    drwxr-xr-x  11 root root  4096 May  7 07:48 usr
    drwxr-xr-x  11 root root  4096 May  7 08:24 var
    pi@raspberrypi:/usr/local/src/moonfire-nvr $

The docker script command reported out at line 27:

     COPY docker/deploy.bash /  

appears to me that the file deploy.bash should appear in the root listing; it was not there.  Perhaps docker works in a chroot and "/" within a docker execution is not "/" of the system?

jlpoolen, 2021-08-11:
Rereading Moonfire's instructions, I had not tested my docker installation.  So I did and passed.  https://pastebin.com/QrsxTRyk

Please put this bug "on hold" while I work through the [different] instructions at https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install.md

I had previously looked at the instructions at https://github.com/scottlamb/moonfire-nvr/blob/master/guide/build.md and thought it the Quick Install method.  Looks like I was reading an inappropriate instruction set.

jlpoolen, 2021-08-11:
Using the instructions set  https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install.md has me on my way.  (I'm now wrestling with adding my TB drives which were configured with LVM and LVM is not installed on Raspberry Pi by default.) 

I'm closing this bug as a result of my using an inappropriate instruction set.

closed 2021-08-11 by jlpoolen

scottlamb, 2021-08-11:
I'm glad you got it working. And I'll take your suggestion of adding the `cd` command to the instructions. Easy enough to do and might save someone else a double-take.

I'm a little puzzled why your second attempt failed. It looks like the `apt-get update` within the docker build container didn't like the signature given to it by `ports.ubuntu.com`:

```
#9 0.796 + apt-get update
#9 1.094 Get:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
#9 1.506 Get:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
#9 1.605 Get:3 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [101 kB]
#9 1.704 Get:4 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
#9 1.752 Err:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
#9 1.752   At least one invalid signature was encountered.
#9 1.943 Err:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
#9 1.943   At least one invalid signature was encountered.
#9 2.133 Err:3 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
#9 2.133   At least one invalid signature was encountered.
#9 2.348 Err:4 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
#9 2.348   At least one invalid signature was encountered.
#9 2.368 Reading package lists...
#9 2.425 W: GPG error: http://ports.ubuntu.com/ubuntu-ports focal InRelease: At least one invalid signature was encountered.
#9 2.426 E: The repository 'http://ports.ubuntu.com/ubuntu-ports focal InRelease' is not signed.
#9 2.427 W: GPG error: http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease: At least one invalid signature was encountered.
#9 2.427 E: The repository 'http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease' is not signed.
#9 2.427 W: GPG error: http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease: At least one invalid signature was encountered.
#9 2.428 E: The repository 'http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease' is not signed.
#9 2.429 W: GPG error: http://ports.ubuntu.com/ubuntu-ports focal-security InRelease: At least one invalid signature was encountered.
#9 2.429 E: The repository 'http://ports.ubuntu.com/ubuntu-ports focal-security InRelease' is not signed.
```

I wonder if the Ubuntu's servers were having an outage. It seems to be working for me now.

> Perhaps docker works in a chroot and "/" within a docker execution is not "/" of the system?

Yes. It uses a filesystem namespace, which is similar to a chroot. That's part of how you can (for example) build and run an Ubuntu image on an Arch host.

scottlamb, 2021-08-11:
Oh, it occurs to me that you were building on the Pi itself. I regularly build *for* arm machines (cross-compiling) but haven't done a build *on* arm machines. I think you hit the problem mentioned [here](https://stackoverflow.com/questions/64439278/gpg-invalid-signature-error-while-running-apt-update-inside-arm32v7-ubuntu20-04), and there are probably more problems. I'll add a note to the instructions that this is only tested on x86-64.

jlpoolen, 2021-08-11:
I just spent 4 hours to delete 8 TBs of previous videos files from Jan-Mar of this year before my Raspberry Pi crashed.  (More on that in an email discussion after I read your recommendations for disk preparation and contrast what I did.)

I appear to be up and running, at least I was able to run the configurator and I stopped there because I realized I had full disks.  After I confirmed my Docker installation using the Hello World example for Docker referenced in the 2nd set of instructions I found, I've been sailing forward.  

I'll configure this evening and update this ticket.  I am hopeful things will go smoothly.  I do have in my mind to note the sub-channel or whatever that possibly handles sound as I recall you had implemented sound, as well.  Lots to catch up on.

scottlamb, 2021-08-11:
Sorry to hear about the messy cleanup. I'll watch for your mail later.

>  I do have in my mind to note the sub-channel or whatever that possibly handles sound as I recall you had implemented sound, as well. Lots to catch up on.

I've got sound support working in the `retina` library but still haven't added it to Moonfire NVR. It's on my list, but it's a fair bit of coding to tweak Moonfire's custom storage schema, writing, and mp4 generation.