scottlamb/moonfire-nvr #142: Docker Permissions Help

jlpoolen, 2021-08-13:
re: https://github.com/scottlamb/moonfire-nvr/blob/master/guide/install.md
the above instructions have you:
1) create user mooonfire-nvr which you accomplish using sudo
2) create a script /usr/local/bin/nvr
3) run the script /usr/local/bin/nvr

When I do, I run into a permissions issue:

jlpoole@raspberrypi:~ $ nvr
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create": dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
jlpoole@raspberrypi:~ $

I tried changing to user moonfire-nvr and then running the nvr script:

    jlpoole@raspberrypi:~ $ su moonfire-nvr
    Password:
    moonfire-nvr@raspberrypi:/home/jlpoole $ nvr init
    docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create": dial unix /var/run/docker.sock: connect: permission denied.
    See 'docker run --help'.
    moonfire-nvr@raspberrypi:/home/jlpoole $

Here's my installation log:

    pi@raspberrypi:/usr/local/src/moonfire-nvr $  sudo useradd --user-group --create-home --home /var/lib/moonfire-nvr moonfire-nvr
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ sudo sh -c 'cat > /usr/local/bin/nvr' <<'EOF'
    > #!/bin/bash -e
    >
    > tz="America/Los_Angeles"
    > container_name="moonfire-nvr"
    > image_name="scottlamb/moonfire-nvr:latest"
    > common_docker_run_args=(
    >         --mount=type=bind,source=/var/lib/moonfire-nvr,destination=/var/lib/moonfire-nvr
    >         --user="$(id -u moonfire-nvr):$(id -g moonfire-nvr)"
    >         --security-opt=seccomp:unconfined
    >         --log-driver=journald
    >         --log-opt="tag=moonfire-nvr"
    >         --env=RUST_BACKTRACE=1
    >         --env=TZ=":${tz}"
    > )
    >
    > case "$1" in
    > run)
    >         shift
    >         exec docker run \
    >                 --detach=true \
    >                 --restart=unless-stopped \
    >                 "${common_docker_run_args[@]}" \
    >                 --network=host \
    >                 --name="${container_name}" \
    >                 "${image_name}" \
    >                 run \
    >                 --allow-unauthenticated-permissions='view_video: true' \
    >                 "$@"
    >         ;;
    > start|stop|logs|rm)
    >         exec docker "$@" "${container_name}"
    >         ;;
    > pull)
    >         exec docker pull "${image_name}"
    >         ;;
    > *)
    >         exec docker run \
    >                 --interactive=true \
    >                 --tty \
    >                 --rm \
    >                 "${common_docker_run_args[@]}" \
    >                 "${image_name}" \
    >                 "$@"
    >         ;;
    > esac
    > EOF
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ sudo chmod a+rx /usr/local/bin/nvr
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ nvr init
    docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post "http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create": dial unix /var/run/docker.sock: connect: permission denied.
    See 'docker run --help'.
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ su moonfire-nvr
    Password:
    su: Authentication failure
    pi@raspberrypi:/usr/local/src/moonfire-nvr $ sudo nvr init
    Unable to find image 'scottlamb/moonfire-nvr:latest' locally
    latest: Pulling from scottlamb/moonfire-nvr
    10ded72e3c7f: Pull complete
    55fc9400ad52: Pull complete
    d8c5e5e92f50: Pull complete
    8dc19d17717f: Pull complete
    94cc30e1ccdb: Pull complete
    36a358db4afb: Pull complete
    823a98033847: Pull complete
    5e9bf3458332: Pull complete
    2cee790b8fc7: Pull complete
    38cb4e356954: Pull complete
    Digest: sha256:c91d586ba480df737f47ac9acba5b27489350fd197ac3f4c8a6f435403fe1c1c
    Status: Downloaded newer image for scottlamb/moonfire-nvr:latest
    I20210811 07:40:23.312 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in Create mode with SQLite version 3.31.1
    I20210811 07:40:23.386 main moonfire_nvr::cmds::init] Database initialized.
    pi@raspberrypi:/usr/local/src/moonfire-nvr $
    
The SQLite database looks to have been installed with the correct user/group and permissions:

    jlpoole@raspberrypi:~ $ sudo ls -la /var/lib/moonfire-nvr
    total 28
    drwxr-xr-x  3 moonfire-nvr moonfire-nvr 4096 Aug 11 08:44 .
    drwxr-xr-x 51 root         root         4096 Aug 11 19:18 ..
    -rw-------  1 moonfire-nvr moonfire-nvr  944 Aug 12 17:28 .bash_history
    -rw-r--r--  1 moonfire-nvr moonfire-nvr  220 Apr 17  2019 .bash_logout
    -rw-r--r--  1 moonfire-nvr moonfire-nvr 3523 May  7 07:42 .bashrc
    drwx------  2 moonfire-nvr moonfire-nvr 4096 Aug 12 17:14 db
    -rw-r--r--  1 moonfire-nvr moonfire-nvr  807 Apr 17  2019 .profile
    jlpoole@raspberrypi:~ $ sudo ls -la /var/lib/moonfire-nvr/db
    total 472
    drwx------ 2 moonfire-nvr moonfire-nvr   4096 Aug 12 17:14 .
    drwxr-xr-x 3 moonfire-nvr moonfire-nvr   4096 Aug 11 08:44 ..
    -rw-r--r-- 1 moonfire-nvr moonfire-nvr 475136 Aug 12 17:14 db
    jlpoole@raspberrypi:~ $

So, something is awry vis-a-vis docker. I've not used docker before and do not know where it hides and what the appropriate permissions should be?  Should docker applications be installed as root, or as the user, i.e. moonfire-nvr?

jlpoolen, 2021-08-13:
Again, I'm unfamiliar with docker, but I thought it would be helpful to know what it's status is:

    jlpoole@raspberrypi:~ $ sudo /etc/init.d/docker status
    ? docker.service - Docker Application Container Engine
       Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: enabled)
       Active: active (running) since Wed 2021-08-11 19:08:16 PDT; 22h ago
         Docs: https://docs.docker.com
     Main PID: 625 (dockerd)
        Tasks: 10
       CGroup: /system.slice/docker.service
               +-625 /usr/bin/dockerd -H fd:// --containerd=/run/containerd/containerd.…

    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:24 raspberrypi moonfire-nvr[625]: [3B blob data]
    Aug 12 17:14:26 raspberrypi moonfire-nvr[625]: [654B blob data]
    Aug 12 17:14:26 raspberrypi dockerd[625]: time="2021-08-12T17:14:26.921525091-0…ete"
    Hint: Some lines were ellipsized, use -l to show in full.
    jlpoole@raspberrypi:~ $

scottlamb, 2021-08-13:
(typing on phone; please excuse terseness)

I think usually with docker people add themselves (the admin user) to the group `docker`, giving them permission to talk to the daemon. This is basically root access so it's similar to sing themselves to the `sudo` group security wise. If you haven't done that, you'll need to prefix your `nvr` commands with `sudo`.

This is something I left implicit in the instructions and should spell out.

commit 30cea5cf (2021-08-23): several documentation improvements

*   prefix docker/nvr commands with sudo (fixes #142).
    I was just going to link to the docker documentation on setting
    up non-root access, but that's kind of a personal preference.
    I included a `<details>` about it instead and made all the commands
    work with sudo.

*   take better advantage of github markdown's code block syntax
    highlighting. Use "console" for shell session stuff, put the
    "nvr" wrapper script in its own block with "bash".

*   add some comments to nvr wrapper script where people need to
    make changes and/or will be [... cut]
closed 2021-08-23 by scottlamb