scottlamb/moonfire-nvr #143: Configratuion - directory permissions

jlpoolen, 2021-08-13:
I'm trying to set my directories.  I have the following directory setup:

    jlpoole@raspberrypi:~ $ ls -la /
    total 84
    [redacted]
    drwxrwxrwx   9 moonfire-nvr moonfire-nvr  4096 Aug 12 17:04 videos
    jlpoole@raspberrypi:~
    jlpoole@raspberrypi:~ $ ls -la /videos
    total 52
    drwxrwxrwx  9 moonfire-nvr moonfire-nvr  4096 Aug 12 17:04 .
    drwxr-xr-x 20 root         root          4096 Aug 11 08:05 ..
    [redacted]
    drwxr-xr-x  2 moonfire-nvr moonfire-nvr  4096 Aug 12 17:02 garage_west
    drwx------  2 moonfire-nvr moonfire-nvr 16384 Jun 10  2020 lost+found
    [redacted]
    jlpoole@raspberrypi:~ $

When I try to specify the directory /videos/garage_west I get this error message:
[image]

When I try to specify the parent directory /videos, I get this error message:
[image]

Both of these directories are owned by user moonfire-nvr.  I launched the "nvr config" using:

    jlpoole@raspberrypi:~ $ sudo nvr config
    I20210812 17:49:55.271 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.31.1
    I20210812 17:49:55.290 main moonfire_db::db] Loading video sample entries
    I20210812 17:49:55.290 main moonfire_db::db] Loaded 0 video sample entries
    I20210812 17:49:55.290 main moonfire_db::db] Loading sample file dirs
    I20210812 17:49:55.291 main moonfire_db::db] Loaded 0 sample file dirs
    I20210812 17:49:55.291 main moonfire_db::db] Loading cameras
    I20210812 17:49:55.291 main moonfire_db::db] Loaded 1 cameras
    I20210812 17:49:55.292 main moonfire_db::db] Loading streams
    I20210812 17:49:55.292 main moonfire_db::db] Loaded 1 streams
    I20210812 17:49:55.292 main moonfire_db::db] Loading recordings for camera GarageWest stream Main
    I20210812 17:49:55.293 main moonfire_db::db] Loaded 0 recordings for camera GarageWest stream Main
    I20210812 17:51:22.406 main moonfire_db::db] Flush 1 (why: drop): no recording changes
    jlpoole@raspberrypi:~ $ 

The root-level directory /videos is a mounted drive to an LVM file system.  I had no problem creating the subdirectories, e.g. /videos/garage_west, in the console.

I'm scratching my head on this.  The SQLite database has appropriate permissions, see Issue #142,

jlpoolen, 2021-08-13:
I tried referencing some other directories, e.g. /tmp, /var/tmp/test, in the nvr config utility and I was successful. 
Therefore. something about the mounted directory to /videos is at play.

I was able to create a directory both as user jlpoole and user moonfire-nvr in a bash shell:

    jlpoole@raspberrypi:~ $ mkdir /videos/jlpooltmadethis
    jlpoole@raspberrypi:~ $ su moonfire-nvr
    Password:
    moonfire-nvr@raspberrypi:/home/jlpoole $ mkdir /videos/mfmadethis
    moonfire-nvr@raspberrypi:/home/jlpoole $ 

Here is my /etc/fstab entry:

     /dev/mapper/vgpurple-cameras /videos ext4 nofail,noatime,lazytime,data=writeback,journal_async_commit  0       2

scottlamb, 2021-08-13:
I'm typing on my phone so excuse my terseness.

Docker containers have their own filesystem view and need mount points explicitly defined for everything that should match the host filesystem. The script in the instructions has that for `/var/lib/moonfire-nvr`; you need to add matching lines for any sample file directories outside of that. (Suggestions welcome for making this more obvious in the instructions.)

jlpoolen, 2021-08-13:
Problem solved by adding this line after the "--mount" row in Scott's script:

      --mount=type=bind,source=/videos,destination=/videos

referenced: https://docs.docker.com/storage/volumes/

I'll take a stab at modifying the install document in my fork.
Thank you.

closed 2021-08-13 by jlpoolen