scottlamb/moonfire-nvr #145: LiveView: QuotaExceededError while appending Buffer

jlpoolen, 2021-08-17:
Upon return to my workstation this morning, I found my screen of Live View with these error messages: QuotaExceededError while appending buffer.  I have not seen this error message before so I thought I would document it here.

[image]

I tried clicking around all the pull downs and the primary larger window pull down and it seemed like everything was focused on the top right small window.  Even though I clicked the top left pull down, the right smaller view would change.  I could not break out, tried clicking all the view pull downs and the topmost striped bar and the Solo/Main Plus Five.  Here's  a video screenshot, unfortunately it did not capture my mouse.

https://user-images.githubusercontent.com/227313/129736482-a3b8298b-ab6d-401e-8ba6-4bc45bc7a0b3.mp4

I determined I had another Firefox tab opened pointing to the Live Screen page.  I closed down Firefox and then restarted and re-accessed the Live View and everything worked.

I am not expecting an answer here.  I wanted to document the kind of problem I encountered and the conditions, i.e. 2 tabs opened, and my attempt to break out failed: I had to restart Firefox.

I like that you have the Main Plus 5 values cached so if I can switch to Solo view and then switch back without having to select which feeds for each of the smaller views.  Of course, caching these values (5 plus 1) on the server side to persist as a user variable, accessible from multiple sessions would be great... but a lot of work.

jlpoolen, 2021-08-17:
I had a small view in the 5 plus 1 go to the state of "QuotaExceededError while appending buffer".  I clicked the pull down menu for that pane and selected none, then did the same selecting the feed that had been there and the feed resumed.

scottlamb, 2021-08-17:
I'm not too surprised you saw `QuotaExceededError`. That's part of why the live view is marked experimental. The browser has some (implementation-defined) limit for how much video can be in the buffer at once. The Javascript code for live view is still really basic. It makes some effort to remove video that's behind the current playback time but doesn't ensure it keeps up with real time. Likely it fell far enough behind that the upcoming buffer exceeded Firefox's limits. It could have slowly drifted behind if your camera's clock is faster than your desktop's, or maybe Firefox automatically paused when the browser window was hidden (or the display went to sleep) but Moonfire NVR kept streaming video until the buffer filled.