scottlamb/moonfire-nvr #146: Live view init segment error

cholcombe973, 2021-08-19:
**Describe the bug**
The latest version of the docker container seems to have broken live view for me. I'm seeing an error message in the firefox web console: `aborting due to init segment fetch status error`. I'm also seeing a websocket error it looks like: 
`The connection to ws://192.168.1.142:8082/api/cameras/3de51215-81d9-4365-aba5-5c7a98a99c34/sub/live.m4s was interrupted while the page was loading.`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'the web interface'
2. Click on 'live view'
3. Scroll down to 'select the camera you'd like to view'
4. See error

**Expected behavior**
I think the previous docker container that was tagged `0.6.4` worked ok. 

**Screenshots**
[image]

**Server (please complete the following information):**
  - If using Docker: `docker ps` + `docker images`
 ```
CONTAINER ID        IMAGE                                  COMMAND                  CREATED             STATUS              PORTS                    NAMES
bc9cc28bdd49        scottlamb/moonfire-nvr:latest          "/usr/local/bin/moon…"   41 minutes ago      Up 41 minutes       0.0.0.0:8082->8080/tcp   moonfire-nvr
chris@chris-Workstation:~$ docker images
REPOSITORY                             TAG                 IMAGE ID            CREATED             SIZE
scottlamb/moonfire-nvr                 latest              e1885469d692        5 days ago          752MB
scottlamb/moonfire-nvr                 <none>              bb6a1bc80ab1        7 weeks ago         735MB
```
  - Attach a [log file](https://github.com/scottlamb/moonfire-nvr/blob/master/guide/troubleshooting.md#viewing-moonfire-nvrs-logs). Run with the `RUST_BACKTRACE=1` environment variable set if possible.

**Camera (please complete the following information):**
  - Camera manufacturer and model: [e.g. Reolink RLC-410]
  - Firmware version: [e.g. `V2.0.0.1215_16091800`]

**Desktop (please complete the following information):**
 - OS: macOS bigsur
 - Browser firefox
 - Version 91

**Additional context**
Add any other context about the problem here.

scottlamb, 2021-08-19:
I have a theory. 27395ec added an `X-Aspect` header to init segments, and the client code now will return this error if it's absent. (I'll make the error more informative shortly.) I forgot to modify code that makes the etag (cache key). So if your browser cached the init segment for this camera before you upgraded and the server responded with status 304 "Not modified", the client code will see an init segment without the header and complain.

The only problem with this theory is the line that says `/api/init/1.mp4: 200` (rather than 304). But maybe the browser hides the 304 from the Javascript or something, just showing some post-cache status.

If my theory is correct, clearing your browser's cache for this domain should resolve the problem. Could you give it a try please?

cholcombe973, 2021-08-19:
Wow. Yup clearing the cache fixed it! Thanks

scottlamb, 2021-08-19:
Thanks for confirming! I should still bump my etag version number so reopening.

scottlamb, 2021-08-19:
Fixed at HEAD, except I should also do this:

>  (I'll make the error more informative shortly.)

scottlamb, 2021-08-20:
Hmm, still broken. :-( I'm seeing the problem myself on my phone for one of my cameras, even after updating my server past my fix attempt. I just got this packet capture of the exchange between my nginx proxy and Moonfire:

```
GET /api/init/68.mp4 HTTP/1.1
X-Forwarded-Proto: https
X-Real-IP: 192.168.1.136
Connection: close
Host: nvr.home.slamb.org
sec-ch-ua: "Chromium";v="92", " Not A;Brand";v="99", "Google Chrome";v="92"
sec-ch-ua-mobile: ?1
User-Agent: Mozilla/5.0 (Linux; Android 11; Pixel 4) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/92.0.4515.159 Mobile Safari/537.36
Accept: */*
Sec-Fetch-Site: same-origin
Sec-Fetch-Mode: cors
Sec-Fetch-Dest: empty
Referer: https://nvr.home.slamb.org/
Accept-Encoding: gzip, deflate, br
Accept-Language: en-US,en;q=0.9
Cookie: s=(redacted)
If-None-Match: "622ed7fcbc8beed5736e2c72291d9ed1798a96313958add5d49b2a06e700badf"
If-Modified-Since: Thu, 01 Jan 1970 00:00:00 GMT

HTTP/1.1 304 Not Modified
accept-ranges: bytes
date: Fri, 20 Aug 2021 03:44:14 GMT
last-modified: Thu, 01 Jan 1970 00:00:00 GMT
etag: "f665fccdd9a8d56ab97bdb5004544471f73c5a72fe6fd0d7741024e2ec4c3540"
cache-control: private, max-age=3600
```

The etags don't match! Moonfire NVR shouldn't have returned `304 Not Modified` in this case! Looks like a bug in my [`http-serve` crate](https://github.com/scottlamb/http-serve).  I'm surprised; I thought that was well-proven code.

scottlamb, 2021-08-20:
https://github.com/scottlamb/http-serve/issues/23

scottlamb, 2021-08-20:
Now it works on my phone, without my having cleared its cache.

commit 7e5bc426 (2021-08-31): improve init segment fetch error msg

Fixes #146, in combination with earlier commit 4892ed7.
closed 2021-08-31 by scottlamb