scottlamb/moonfire-nvr #147: nvr init failed

brian-liu1, 2021-08-19:
I have finished compiling the moonfir for armv7, but when i run nvr init , it failed with the log:
`I20100101 02:33:15.728 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in Create mode with SQLite version 3.7.13
E20100101 02:33:26.836 main moonfire_nvr] Exiting due to error: near "without": syntax error

Backtrace:
   0: backtrace::backtrace::libunwind::trace
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.60/src/backtrace/libunwind.rs:90:5
      backtrace::backtrace::trace_unsynchronized
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.60/src/backtrace/mod.rs:66:5
   1: backtrace::backtrace::trace
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.60/src/backtrace/mod.rs:53:14
   2: backtrace::capture::Backtrace::create
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.60/src/capture.rs:176:9
   3: backtrace::capture::Backtrace::new_unresolved
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/backtrace-0.3.60/src/capture.rs:170:9
   4: failure::backtrace::internal::InternalBacktrace::new
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/internal.rs:46:44
   5: failure::backtrace::Backtrace::new
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/backtrace/mod.rs:121:35
   6: <failure::error::error_impl::ErrorImpl as core::convert::From<F>>::from
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/error_impl.rs:19:17
   7: <failure::error::Error as core::convert::From<F>>::from
             at /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/failure-0.1.8/src/error/mod.rs:36:18
   8: moonfire_db::db::init
             at /home/brian/bpi/nvr/moonfire-nvr-master_n/moonfire-nvr-master/server/db/db.rs:2126:49
   9: moonfire_nvr::cmds::init::run
             at /home/brian/bpi/nvr/moonfire-nvr-master_n/moonfire-nvr-master/server/src/cmds/init.rs:44:5
  10: moonfire_nvr::Args::run
             at /home/brian/bpi/nvr/moonfire-nvr-master_n/moonfire-nvr-master/server/src/main.rs:66:34
  11: moonfire_nvr::main
             at /home/brian/bpi/nvr/moonfire-nvr-master_n/moonfire-nvr-master/server/src/main.rs:143:9
  12: core::ops::function::FnOnce::call_once
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/ops/function.rs:227:5
  13: std::sys_common::backtrace::__rust_begin_short_backtrace
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/sys_common/backtrace.rs:125:18
  14: std::rt::lang_start::{{closure}}
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/rt.rs:49:18
  15: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/ops/function.rs:259:13
      std::panicking::try::do_call
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/panicking.rs:379:40
      std::panicking::try
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/panicking.rs:343:19
      std::panic::catch_unwind
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/panic.rs:431:14
      std::rt::lang_start_internal
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/rt.rs:34:21
  16: std::rt::lang_start
             at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/rt.rs:48:5
  17: main
  18: __libc_start_main
             at /home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/arm-linux-gnueabihf/snapshots/glibc.git~release~2.25~master/csu/libc-start.c:295`

i am not well known about rust and i can not find way to finish the error.

scottlamb, 2021-08-19:
Your log mentions SQLite version 3.7.13. Moonfire NVR's SQL schema uses [`without rowid`](https://www.sqlite.org/vtab.html#worid), which was apparently added in version 3.14.0 (2016-08-08). I should add a version check to improve the error message, but the root cause is that your SQLite version is too old.

What Linux distribution are you using? I didn't think any were bundling versions that old anymore.

You can try compiling with `--features=bundled`, which will statically include a more recent SQLite version. Also, when you're compiling yourself (rather than using the Docker images), I recommend running the test suite so you can catch problems ahead of time.

```
cargo test --features=bundled
cargo build --release --features=bundled
```

brian-liu1, 2021-08-19:
My bananapi installed with linux3.10. I think it is very old but i have no choices.  and i compiled the sqlite by myself and i have no idea about the version that nvr is needed.

brian-liu1, 2021-08-19:
> Your log mentions SQLite version 3.7.13. Moonfire NVR's SQL schema uses [`without rowid`](https://www.sqlite.org/vtab.html#worid), which was apparently added in version 3.14.0 (2016-08-08). I should add a version check to improve the error message, but the root cause is that your SQLite version is too old.
> 
> What Linux distribution are you using? I didn't think any were using versions that old anymore.
> 
> You can try compiling with `--features=bundled`, which will statically link in a more recent SQLite version. Also, when you're compiling yourself (rather than using the Docker images), I recommend running the test suite so you can catch problems ahead of time.
> 
> ```
> cargo test --features=bundled
> cargo build --release --features=bundled
> ```

when i build the nvr ,i got an error 
`error: failed to run custom build command for `ncurses v5.101.0`

Caused by:
  process didn't exit successfully: `/home/brian/bpi/nvr/moonfire-nvr-master_/server/target/release/build/ncurses-297321b7ae671536/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=NCURSESW5_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_ALLOW_CROSS_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_ALLOW_CROSS_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_ALLOW_CROSS
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=NCURSESW5_STATIC
  cargo:rerun-if-env-changed=NCURSESW5_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=NCURSESW_NO_PKG_CONFIG
  cargo:rerun-if-env-changed=PKG_CONFIG_ALLOW_CROSS_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_ALLOW_CROSS_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_ALLOW_CROSS
  cargo:rerun-if-env-changed=PKG_CONFIG
  cargo:rerun-if-env-changed=NCURSESW_STATIC
  cargo:rerun-if-env-changed=NCURSESW_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_STATIC
  cargo:rerun-if-env-changed=PKG_CONFIG_ALL_DYNAMIC
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_PATH
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_arm-unknown-linux-gnueabihf
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_arm_unknown_linux_gnueabihf
  cargo:rerun-if-env-changed=TARGET_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR
  cargo:rustc-link-lib=ncursesw
  OPT_LEVEL = Some("3")
  TARGET = Some("arm-unknown-linux-gnueabihf")
  HOST = Some("x86_64-unknown-linux-gnu")
  CC_arm-unknown-linux-gnueabihf = None
  CC_arm_unknown_linux_gnueabihf = None
  TARGET_CC = None
  CC = None
  CROSS_COMPILE = None
  CFLAGS_arm-unknown-linux-gnueabihf = None
  CFLAGS_arm_unknown_linux_gnueabihf = None
  TARGET_CFLAGS = None
  CFLAGS = None
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some("true")
  CARGO_CFG_TARGET_FEATURE = None

  --- stderr
  thread 'main' panicked at '/home/brian/bpi/nvr/moonfire-nvr-master_/server/target/arm-unknown-linux-gnueabihf/release/build/ncurses-48b2e1943d8447a3/out/chtype_size failed: Os { code: 8, kind: Other, message: "Exec format error" }', /home/brian/.cargo/registry/src/github.com-1ecc6299db9ec823/ncurses-5.101.0/build.rs:107:10
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/std/src/panicking.rs:493:5
     1: core::panicking::panic_fmt
               at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/panicking.rs:92:14
     2: core::result::unwrap_failed
               at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/result.rs:1355:5
     3: core::result::Result<T,E>::expect
               at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/result.rs:997:23
     4: build_script_build::check_chtype_size
               at ./build.rs:106:20
     5: build_script_build::main
               at ./build.rs:59:5
     6: core::ops::function::FnOnce::call_once
               at /rustc/53cb7b09b00cbea8754ffb78e7e3cb521cb8af4b/library/core/src/ops/function.rs:227:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
error: build failed`
i just modify the build.rs to fix it in order to finish compile.

scottlamb, 2021-08-19:
It looks like you're hitting an error while cross-compiling ncurses. https://github.com/jeaye/ncurses-rs/issues/184 looks related. But you must have gotten a build to go through before (to get far enough for `nvr init` to fail) so hopefully you can get it to work again.

I'm about to close this issue because the reason `nvr init` failed is understood, and I'll shortly submit a change that makes that more clear. But feel free to open another issue if you need more help.

commit 4e77a264 (2021-08-19): document and check minimum SQLite version

Fixes #147
closed 2021-08-19 by scottlamb