scottlamb/moonfire-nvr #148: Raspberry Pi & Node.js -- What Version?

jlpoolen, 2021-08-20:
**Describe the bug**
Raspberry Pi & Node

**To Reproduce**
Raspberry Pi4 running  Linux pi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux

Building moonfire-nvr requires running node for the ui component.  The current high watermark for node available through the standard apt (default) is v10.24.0.   Some of the libraries/components for the UI state they need v12:
[image]

I was looking at alternative ways to install a later version of Node.js and found this page: 

Trying to build Node on a Raspberry Pi is probably a 2-4 day endeavor per https://gist.github.com/nwawrzyniak/58d087f9d0ad70c3e41584b077e1f13e.

[image] identifies alternative ways.

What is troubling is what looks to be Node downgrading support for the armv6 processor [Raspberry Pi] to experimental starting with v12.0.0
https://github.com/nodejs/nodejs.org/pull/2205

So, as if August 21, 2021, what version of Node should one install on a Raspberry Pi4?  And do you have a preferred method of installation?

scottlamb, 2021-08-20:
I'm trying to follow what node does so I can update packages as necessary for security fixes and such. node says 10 has reached end-of-life, so I'm not trying to support it anymore. It broke when I updated `gzipper` or some such.

Did you see these paragraphs I recently added to the install instructions?

> To build the UI, you'll need a [nodejs](https://nodejs.org/en/download/) release in "Maintenance LTS" or "Active LTS" status: currently v12 or v14.

...

> Ubuntu 20.04 (the latest LTS as of this writing) bundles node 10, which has reached end-of-life (see [node.js: releases](https://nodejs.org/en/about/releases/)). So rather than install the `nodejs` and `npm` packages from the built-in repository, see [Installing Node.js via package manager](https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions).

Do those answer your question?

> What is troubling is what looks to be Node downgrading support for the armv6 processor [Raspberry Pi] to experimental starting with v12.0.0 nodejs/nodejs.org#2205

The original Raspberry Pi is armv6 iirc, but Pi2 is armv7, and pi3 and pi4 are armv8. I think both 32-bit and 64-bit Raspberry Pi OSs should be supported with the official node packages.

scottlamb, 2021-08-20:
My own Pi4 (running the Raspberry Pi 64-bit OS beta) has node 14 installed via the `https://deb.nodesource.com/node_14.x`. I'm not regularly building on it anymore because I'm instead dogfooding the cross-compiled docker container. But I tried a build on it now, and it seems to work.

The continuous integration tests are running with node version 12, version 14, and version 16.

jlpoolen, 2021-08-20:
Thank you.  Installing v.14.17.5 now.

jlpoolen, 2021-08-20:
Installation worked.

    Download page: https://nodejs.org/en/download/
    downloaded ARMv7:  https://nodejs.org/dist/v14.17.5/node-v14.17.5-linux-armv7l.tar.xz

    VERSION=v14.17.5
    DISTRO=linux-armv71
    sudo tar -xJvf node-$VERSION-$DISTRO.tar.xz -C /usr/local/lib/nodejs 

Thank you.  Will be opening another issue... configuration re: ui directory ( I feel like I had this issue a year ago, but could not find reference to it.)

closed 2021-08-20 by jlpoolen