scottlamb/moonfire-nvr #149: nvr config error

brian-liu1, 2021-08-20:
nvr init finished ,when i run nvr config ,if failed with:
`[ttyS0]root@bpi-iot-ros-ai:/usr/target/release# nvr config            
I20100101 09:27:20.749 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in ReadWrite mode with SQLite version 3.32.2
I20100101 09:27:20.786 main moonfire_db::db] Loading video sample entries
I20100101 09:27:20.788 main moonfire_db::db] Loaded 0 video sample entries
I20100101 09:27:20.789 main moonfire_db::db] Loading sample file dirs
I20100101 09:27:20.791 main moonfire_db::db] Loaded 0 sample file dirs
I20100101 09:27:20.792 main moonfire_db::db] Loading cameras
I20100101 09:27:20.794 main moonfire_db::db] Loaded 0 cameras
I20100101 09:27:20.795 main moonfire_db::db] Loading streams
I20100101 09:27:20.798 main moonfire_db::db] Loaded 0 streams
Segmentation fault`

I am sure that nvr init is ok:
`[ttyS0]root@bpi-iot-ros-ai:/usr/target/release# nvr init
I20100101 09:24:43.833 main moonfire_nvr::cmds] Opening /var/lib/moonfire-nvr/db/db in Create mode with SQLite version 3.32.2
I20100101 09:24:43.880 main moonfire_nvr::cmds::init] Database is already initialized with schema version 6.`

i tried to use gdb to debug it but i failed for the app is run with scripts and gdb can not find how to run, i am look for help to fix this problem.

scottlamb, 2021-08-20:
I know from #147 that you're doing some custom cross-compiled ncurses build, and I suspect the problem relates to that. Moonfire NVR uses ncurses through a Rust library called Cursive. Could you see if the cursive's examples run correctly? when building on the host, you could do this:

```
$ git clone https://github.com/gyscos/cursive.git
$ cd cursive
$ cargo build --example hello_world
$ target/debug/hello_world
```

I'm not sure how your cross-compile setup works though.

brian-liu1, 2021-08-20:
> I know from #147 that you're doing some custom cross-compiled ncurses build, and I suspect the problem relates to that. Moonfire NVR uses ncurses through a Rust library called Cursive. Could you see [...]

I check the ncurses-5.101.0->buil.rs, follow the issue you have memtioned , i commented out the function check_chtype_size(&ncurses_lib), the process of compile goes smoothly。and laterly i will get the error when i run "nvr config". Now i want to know the function of "check_chtype_size", for my bpi is armv7 32bit. why the ncurses-5.101.0 will run chtype_size in cmd, beacause i cross-compile the moonfir-nvr on ubuntu and the cmd  can not run the chtype_size for that is an applicantion of armv7. that is the reason i get a build failed  error "Os { code: 8, kind: Other, message: "Exec format error"".

scottlamb, 2021-08-20:
From skimming the ncurses crate issue, it seems like to cross-compile ncurses reliably you need a way to test against the exact header version on the target to get a compatible ABI. Like having the build process ssh into it or something. Seems like a real pain to get working reliably.

Here's another idea. Cursive doesn't have to use ncurses. It supports other backends, including the termion one in pure rust. I think switching is a one-line change:

```diff
diff --git a/server/Cargo.toml b/server/Cargo.toml
index d1573b7..baafe37 100644
--- a/server/Cargo.toml
+++ b/server/Cargo.toml
@@ -26,7 +26,7 @@ bytes = "1"
 byteorder = "1.0"
 clap = { version = "2.33.3", default-features = false, features = ["color", "wrap_help"] }
 cstr = "0.2.5"
-cursive = "0.16.3"
+cursive = { version = "0.16.3", default-features = false, features = ["termion-backend"] }
 db = { package = "moonfire-db", path = "db" }
 failure = "0.1.1"
 ffmpeg = { package = "moonfire-ffmpeg", version = "0.0.1" }
```

brian-liu1, 2021-08-20:
> From skimming the ncurses crate issue, it seems like to cross-compile ncurses reliably you need a way to test against the exact header version on the target to get a compatible ABI. Like having the build process ssh into it or something. Seems like a real pain to get working reliably.
> 
> Here's another idea. Cursive doesn't have to use ncurses. It supports other backends, including the termion one in pure rust. (See [Cursive docs on backends](https://github.com/gyscos/cursive/wiki/Backends).) I think switching is a one-line change:
> 
> ```diff
> diff --git a/server/Cargo.toml b/server/Cargo.toml
> index d1573b7..baafe37 100644
> --- a/server/Cargo.toml
> +++ b/server/Cargo.toml
> @@ -26,7 +26,7 @@ bytes = "1"
>  byteorder = "1.0"
>  clap = { version = "2.33.3", default-features = false, features = ["color", "wrap_help"] }
>  cstr = "0.2.5"
> -cursive = "0.16.3"
> +cursive = { version = "0.16.3", default-features = false, features = ["termion-backend"] }
>  db = { package = "moonfire-db", path = "db" }
>  failure = "0.1.1"
>  ffmpeg = { package = "moonfire-ffmpeg", version = "0.0.1" }
> ```

Yes, it works.  Without ncurses everything seems ok.

scottlamb, 2021-09-01:
Glad you got it working. I think I can improve the cross-compilation story in #160.

closed 2021-09-01 by scottlamb