scottlamb/moonfire-nvr #153: JSON API for configuration

scottlamb, 2021-08-25:
Define and implement a JSON API that can do everything `moonfire-nvr config` currently does, so we can replace `moonfire-nvr config` with a web UI (#35):

*   [ ] add/remove sample file directories
*   [ ] configure retention, deleting old files as necessary to fit within lower limits.
*   [ ] test out RTSP URLs (at least reporting if they work and pixel dimensions)
*   [ ] add/edit/remove cameras
*   [ ] add/edit/remove users

One complication: Right now the code isn't really structured to allow changing these things while running. A few examples of what I mean:

*   the sample file directory syncer thread isn't really set up to dynamically change the limits mid-run. 
*   the code which lowers retention will lock the database as long as it feels like (maybe a minute or more)
*   we don't dynamically add/remove cameras

Maybe it's acceptable as a first pass to switch between "config mode" and "running mode" based on an API request; eventually we should just allow recording and reconfiguring at once.

@clydebarrow

clydebarrow, 2021-08-25:
Yes, I figured that there would probably be more work in reworking the internals to be able to config on the fly, rather than the API itself. My thoughts:

1. Database - perhaps make the locking finer grained - is that possible with SQLite?
2. Reconfiguring other stuff - I'd suggest having an API call that restarts the server processes, so that the reconfiguration can be done while running, but does not take effect until the UI says so - probably linked to an "Apply" button.

clydebarrow, 2021-08-28:
For user preferences there will be some data that can be saved in browser localstorage, but some, e.g. date formats, should be global for a user, and so should be saved in the server. So there should be an api call to save and retrieve user preferences. This could be simply one JSON-encoded string, so that it can be extended without database schema changes.

scottlamb, 2021-08-28:
I'll create an issue for that next schema version. That's an easy one to implement, so I'll include it for sure.

scottlamb, 2021-09-01:
User preferences are now on the `new-schema` branch. Let me know if the API works for you. I tried to come up with something that would be extensible to other user operations, although setting the current user's preferences is the only thing it's good for now.