scottlamb/moonfire-nvr #157: Live View Selection Starting Clockwise with Small Pane Hangs on Last Large Pane

jlpoolen, 2021-08-31:
**Describe the bug**
In Live View, specifying the feeds in five-plus-one view starting with top right small and going clockwise ends up with large pane not working.  After successfully specifying the 5 small panes, When I pull the menu down for the large pane and select the camera (as yet unspecified), the first pane changes to the one I selected for the large pane and the pull down menu freezes.  Then the entire interface locks up, although the videos continue to change.  The only way to get out of the quandry is to perform a refresh of the page and then select Live VIew and go through the sequence.  If I specify the large pane first, then rotatie around clockwise specifying the the small feeds, everything works.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main url, i.e: pi:8080
2. Click on Live View
3. Select "Solo" button and change to main-plus-five
4. Select top right small pane and select feed, work clockwise through the other four small panes, then try to select the feed for the large pane

**Expected behavior**
Should be able to select any pane and change the camera feed in any sequence.

**Screenshots**
[image]

**Server (please complete the following information):**
  - If using Docker: `docker ps` + `docker images`
  - If building from git: `git describe --dirty` + `moonfire-nvr --version`
     jlpoole@pi:/usr/local/src/moonfire-nvr $ git describe --dirty
     v0.6.5-6-g5c3c618
     jlpoole@pi:/usr/local/src/moonfire-nvr $ nvr --version
     moonfire-nvr 0.6.5
     jlpoole@pi:/usr/local/src/moonfire-nvr $ 

  - my screenshot was at 6:34, so I yanked from 6:28 - 6:38 from the server's log file:
[moonfire-nvr_Aug_31_0628-0638.log](https://github.com/scottlamb/moonfire-nvr/files/7084459/moonfire-nvr_Aug_31_0628-0638.log)

**Camera (please complete the following information):**
  - Camera manufacturer and model: [e.g. Reolink RLC-410]
  - Firmware version: [e.g. `V2.0.0.1215_16091800`]

**Desktop (please complete the following information):**
 - OS: Gentoo: 
 -      taurus /home/jlpoole/work # uname -a
        Linux taurus 5.10.27-gentoo #6 SMP Mon May 10 20:44:46 PDT 2021 x86_64 Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz    GenuineIntel GNU/Linux
        taurus /home/jlpoole/work #
 - 
 - Firefox 
 - Version 91.0.2 (64-bit)

jlpoolen, 2021-08-31:
I should clarify, when I am selecting the large pane, the choice I make is then reflected in the right top small pane and nothing shows up in the large pane.

jlpoolen, 2021-08-31:
Further, once the pull down menu for the large window displays and does not collapse, I can go through the choices and click them which causes the top right small to change to that selection (and if the selection was appearing in another small window, that corresponding window goes dark to "none", though I cannot select any other window's pull down menu).  I tried this 55 minutes later just to see if anything blocking might have timed out or whatever and the screen remains in a state where the main pull down can control the top right pane only.

scottlamb, 2021-08-31:
I'd never tried doing it in that order before. I see the same thing in Firefox and Chrome. No idea why yet.

jlpoolen, 2021-08-31:
Here's a Firefox JavaScript console log which doesn't seem to reveal anything:
[moonfire-nvr_Aug_31_FireFox_JavaScriptConsole_0740.log](https://github.com/scottlamb/moonfire-nvr/files/7084761/moonfire-nvr_Aug_31_FireFox_JavaScriptConsole_0740.log)

jlpoolen, 2021-08-31:
The Firefox log captures the event of refresh, select LiveView up to the point of freezing and then when I saw the smaller panes dutifully refreshing I marked that as the end point of the log to capture.

jlpoolen, 2021-08-31:
**Work-around** for this issue is to always specify the large pane first.

closed 2021-08-31 by scottlamb

jlpoolen, 2021-09-01:
I pulled the latest (and ended up rebuilding):

    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ git --describe dirty
    unknown option: --describe
    usage: git [--version] [--help] [-C <path>] [-c <name>=<value>]
               [--exec-path[=<path>]] [--html-path] [--man-path] [--info-path]
               [-p | --paginate | -P | --no-pager] [--no-replace-objects] [--bare]
               [--git-dir=<path>] [--work-tree=<path>] [--namespace=<name>]
               <command> [<args>]
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ git describe --dirty
    v0.6.5-22-gd2c7b62
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ git pull
    remote: Enumerating objects: 15, done.
    remote: Counting objects: 100% (15/15), done.
    remote: Compressing objects: 100% (2/2), done.
    remote: Total 9 (delta 7), reused 9 (delta 7), pack-reused 0
    Unpacking objects: 100% (9/9), done.
    From https://github.com/scottlamb/moonfire-nvr
       d2c7b62..981a91c  master     -> origin/master
    Updating d2c7b62..981a91c
    Fast-forward
     CHANGELOG.md               |  3 +++
     ui/src/Live/LiveCamera.tsx | 12 +++++++++---
     2 files changed, 12 insertions(+), 3 deletions(-)
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ cargo build --release
        Finished release [optimized + debuginfo] target(s) in 1.28s
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ nvr run --rtsp-transport=udp 2>/tmp/moonfire-nvr.log
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ git describe --dirty
    v0.6.5-24-g981a91c
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $which nvr
    /home/jlpoole/bin/nvr
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ ls -la /home/jlpoole/bin/nvr
    lrwxrwxrwx 1 jlpoole jlpoole 12 Aug 19 17:25 /home/jlpoole/bin/nvr -> moonfire-nvr
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $ ls -la /home/jlpoole/bin/moonfire-nvr
    lrwxrwxrwx 1 jlpoole jlpoole 62 Aug 19 17:10 /home/jlpoole/bin/moonfire-nvr -> /usr/local/src/moonfire-nvr/server/target/release/moonfire-nvr
    jlpoole@pi:/usr/local/src/moonfire-nvr/server $

I then started nvr and tried the selection of panes as described above starting with the smallest top right and ended up with the same problem at the large pane pull down menu: hung interface.

scottlamb, 2021-09-01:
For this change you have to rebuild/instead the JavaScript UI, not the Rust server.

jlpoolen, 2021-09-01:
Confirmed the fix works.  Thank you.

commit d2c7b627 (2021-08-31): fix #157: problem with live multi-camera selection

When you select the first camera in the drop-down on a grid square
that isn't the top left, then select a camera in the top-left, it
behaves strangely.

The root cause is that I had a dumb mistake in how I assigned React
keys. I used the camera index when a camera is selected, and the
inverse of the selected index when one is. But 0 == -0!