scottlamb/moonfire-nvr #162: New user interface

clydebarrow, 2021-09-03:
I have a first-pass version of a new web UI. Current functionality is close to the built-in UI, but it's a completely new project. You can find it here: https://github.com/clydebarrow/moonglasss.git

Most of the menu on the left does nothing yet. The Recordings page brings up a camera list, clicking on a stream will fetch and list available recordings. The camera icon on the right opens live view for the stream, the download buttons for each recording will save an MP4 file. The date/time selection works pretty much like the original.

[image]

scottlamb, 2021-09-03:
Thanks for sharing! I've never used Kotlin before and will be interested to look through it when I get more time.

It works on my backup NVR where I have disabled auth. The fit in the browser window is awkward though:

[image]

clydebarrow, 2021-09-03:
Which browser is that? The player layout will change, and needs optimisation.

scottlamb, 2021-09-03:
Chrome 93 on macOS on a 4k external display

clydebarrow, 2021-09-05:
The layout has been revised, and now works well in all shapes and sizes.

clydebarrow, 2021-09-07:
User authentication now working. I have tested the production build by running nginx on the nvr host.

Still some UI styling to finesse but will work on the live view now.