scottlamb/moonfire-nvr #170: ISO Standard LInk Update?

jlpoolen, 2021-09-27:
In the Overview of https://github.com/scottlamb/moonfire-nvr/blob/master/design/schema.md#detailed-design 
reference is made:

     (see ISO 14496-12)
     
with a link to:  https://www.iso.org/standard/68960.html The page that resolves at this link states that ISO/IEC 14496-12:2015 now has a status of "withdrawn"

The new revised standard is at https://www.iso.org/standard/74428.html

I just want to make sure that before I dive into the almost 200 page document that I'm reading the one that applies to this project.  Let me know if you would like me to update the link with a pull request.

jlpoolen, 2021-09-27:
Ouch!   It appears one has to purchase the standard which is now 245 pages I'm afraid that is too rich for my pocketbook.
[image]

scottlamb, 2021-09-27:
Probably we should update that link to [this wiki page](https://github.com/scottlamb/moonfire-nvr/wiki/Standards-and-specifications) which talks about where to get free legal copies of these standards, and we can expand that page. I'm in a hurry now, but some quick points:

*   I don't think it's reasonable for an individual to pay for ISO/IEC standards. Not only are they like CHF 198 apiece, but to do anything you need a long list of interrelated ones. We're talking new car money here.
*   the sometimes-free withdrawn ISO/IEC standards are almost as good as the current ones. New revisions don't break compatibility; they just make minor clarifications and add new stuff.
*   ISO/IEC sometimes removes standards from their publicly available standards page, but archive.org remembers.
*   Sometimes there's an equivalent standard from another organization that is free. ITU's T.REC H.264 is word-for-word identical to some ISO/IEC standard whose ridiculous number I've forgotten.
*    Sometimes you can find committee drafts of ISO/IEC standards which I'm told are effectively the same as the published standard, and I think (don't trust me though) can be redistributed. I've had trouble finding them though.
*   I've seen other complete copies come up in websearches, but I've avoided them because I imagine they're not being distributed legally.

closed 2021-11-23 by scottlamb

jlpoolen, 2021-11-23:
I ended up contacting our City of Salem library who determined that the only library in the State of Oregon was Oregon State University's library that had a subscription to IEEE.  I have been told I may come to their premises and use a public terminal and log into the IEEE portal and explore from there.  Others facing this should probably determine which libraries in their state have subscriptions and then see if they have public access.  Fortunately, the university's library is only a 40 minute drive away... but, still, what an inconvenience.

scottlamb, 2021-11-23:
Yeah, I agree these non-free standards are a real pain. I'll do my best to keep the new link up-to-date with ways to get access.

commit 46daed6c (2021-11-23): more helpful link for ISO/IEC 14496-12

Fixes #170