scottlamb/moonfire-nvr #173: Enhancement Request: Last-click row in Listing retains color to identify it was last selected

jlpoolen, 2021-09-30:
Enhancement Request

In the Web User Interface, when you have a long listing of rows in a table for a camera, when you click one, a video appears which overlaps the table of rows.  When you return to the table view after clicking outside of the video window, you have the same table.

I'm finding I do not remember which row I clicked.  When the rows are differentiated by a few minutes, instead of hours, it is difficult to remember which one you just clicked.  What would be a nice feature is that whatever you click sufficiently to launch a video, changes color and that color persists until a new click event occurs... causing the newly click row to have coloration thereby marking it the "active" row.

clydebarrow, 2021-09-30:
Something like this?

[image]

jlpoolen, 2021-09-30:
clydebarrow:  precisely.

clydebarrow, 2021-10-02:
Ok, so here's how you get the new ui:

Edit the `nvr` script created during installation of Moonfire-nvr. Locate the line like:

````
image_name="scottlamb/moonfire-nvr:v0.6.6"
````

Replace this with:

````
image_name="clydeps/moonglass:0.1.0"
````

This is a docker image built on top of `moonfire-nvr:v0.6.6` that replaces the UI html and javascript with the Moonglass ui.

Stop the nvr, remove the current container and restart:

````
nvr stop
docker rm moonfire-nvr
nvr run
````
Reload the UI from localhost:8080 (or wherever it is served from.) Enjoy.

Display mode (light/dark), time and date formats are configurable from the Preferences option in the drop down menu from the user icon. Most things should be self explanatory. Submit bugs and feature requests here:

https://github.com/clydebarrow/moonglass/issues

At some point this may become the default UI. Scott and I are cooperating on this but details are still being worked through.

Signals and a scrub bar are next on the TODO list.

jlpoolen, 2021-10-02:
Thanks, Clyde.  Um... I'm a build-from-source kinda guy and will try to devise a way to integrate/migrate/incorporate you modifications.  I tried Docker and got frustrated with my ignorance of all the intricacies of Docker and trying to read for a few minutes hoping I'd gain enough knowledge to accomplish what I wanted.  

I am anxious to try your modifications.

clydebarrow, 2021-10-02:
Sure. Building details are in the repository. The chief issue that due to CORS limitations the UI must be served from the same host as the API. I have details on setting up Nginx to accomplish that in the README.

Docker does have a steep learning curve :-)