scottlamb/moonfire-nvr #181: UI Build: api.ts Line 0:  Parsing error: Cannot read property 'map' of undefined

jlpoolen, 2021-11-07:
I upgraded my projects on a Raspberry Pi (GIT controlled tree) and on a Gentoo VM (Subversion controlled tree).  The Subversion controlled tree worked and I can run a server.  The GIT instance, however, will not build the server UI.  The moonfire-nvr is working (I hope) as I'm able to run it and the log on the screen suggests activity saving the video feeds.

I am attaching files relevant to each server that hopefully contain identifying information and attempt to work around the problem.

I'm running into problems on Raspberry Pi with the build of UI, here's the main problem.  

    jlpoole@pi:/usr/local/src/moonfire-nvr/ui $  SKIP_PREFLIGHT_CHECK=true npm run build

    > ui@0.1.0 build
    > react-scripts build && gzipper compress --exclude=png,woff2 --remove-larger ./build

    Creating an optimized production build...
    Failed to compile.

    ./src/api.ts
      Line 0:  Parsing error: Cannot read property 'map' of undefined

    jlpoole@pi:/usr/local/src/moonfire-nvr/ui $

I'm guessing there is a configuration issue with Node or npm that is tripping me up.  I'm tempting to start over on the Raspberry Pi and use Subversion to track the project as I don't seem to "get" git -- I'm always getting tripped up on its squirrely commands.  I'm logging this issue in case others are using GIT on a Raspberry PI and may encounter the problem I have in the UI build.

[ares_ui_build_log.txt](https://github.com/scottlamb/moonfire-nvr/files/7492725/ares_ui_build_log.txt)
[pi_ui_build_log.txt](https://github.com/scottlamb/moonfire-nvr/files/7492726/pi_ui_build_log.txt)

What do you recommend I do on the Raspberry PI to get past the blocker on the UI build?

scottlamb, 2021-11-23:
Did you ever get it to build on this machine?

One thing I see in the failing build log:

```
jlpoole@pi:/usr/local/src $ node --version
v14.17.5
jlpoole@pi:/usr/local/src $ npm --version
8.1.3
```

which seems strange to me because I think node.js and npm come bundled; node.js v14.17.5 pairs with npm 6.14.14 on their website; npm 8.1.3 isn't listed at all (yet). Maybe you installed a new npm that isn't compatible with the node.js release you're using.

Also, you shouldn't need `sudo` on your `npm install` step (which just installs things to `node_modules` within the current directory) or `npm run build` step.

jlpoolen, 2021-11-23:
Yes, by hook or by crook, I got it working.  I had forgotten I had filed this issue and I think I ended up doing a fresh install and compile.  I'm still on the same two versions:

	jlpoole@pi:~ $ date
	Tue 23 Nov 2021 11:14:52 AM PST
	jlpoole@pi:~ $ npm -v
	8.1.3
	jlpoole@pi:~ $ node -v
	v14.17.5
	jlpoole@pi:~ $ 
	`
I apologize, I should have updated this issue.

closed 2021-11-23 by jlpoolen